stlink-org/stlink #991: NUCLEO-L152RE: Cannot program the board with version 1.6.1

vincentdupaquis, 2020-06-19:
Thank you for giving feedback to the stlink project.

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: Nucleo-L152RE
- Programmer firmware version: I do not know I am using the one which is on the board
- Operating system and version: Ubuntu 20.04
- **Stlink tools version** and/or git commit hash: v1.6.1
- Stlink commandline tool name: st-util
- Target chip (and board if applicable): STM32L152RE

I am experiencing weird issues with stlink tools. Indeed, I am not able to write flash through gdb, although I can through the stlink-gui tool.

Commandline-Output:
First, I start st-util as a gdbserver : st-util -p 1234 -m -v99
Then I connect through gdb : 
target remote localhost:1234
restore build/target.bin binary 0x8000000
gdb replies by 
**Restoring binary file build/target.bin into memory (0x8000000 to 0x80040f4)
Writing to flash memory forbidden in this context**

Here is the full trace obtained from st-util :
vdupaquis@vdupaquis-Latitude-5491:~/tmp/stlink$ st-util -p 1234 -m -v99
st-util
2020-06-19T13:52:06 DEBUG common.c: *** looking up stlink version
2020-06-19T13:52:06 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-06-19T13:52:06 DEBUG common.c: stlink pid     = 0x374b
2020-06-19T13:52:06 DEBUG common.c: stlink version = 0x2
2020-06-19T13:52:06 DEBUG common.c: jtag version   = 0x1c
2020-06-19T13:52:06 DEBUG common.c: swim version   = 0x11
2020-06-19T13:52:06 DEBUG common.c: *** looking up stlink version
2020-06-19T13:52:06 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-06-19T13:52:06 DEBUG common.c: stlink pid     = 0x374b
2020-06-19T13:52:06 DEBUG common.c: stlink version = 0x2
2020-06-19T13:52:06 DEBUG common.c: jtag version   = 0x1c
2020-06-19T13:52:06 DEBUG common.c: swim version   = 0x11
2020-06-19T13:52:06 DEBUG common.c: stlink current mode: debug (jtag or swd)
2020-06-19T13:52:06 DEBUG usb.c: JTAG/SWD freq set to 0
2020-06-19T13:52:06 DEBUG common.c: *** set_swdclk ***
2020-06-19T13:52:06 DEBUG common.c: stlink current mode: debug (jtag or swd)
2020-06-19T13:52:06 DEBUG common.c: *** stlink_jtag_reset ***
2020-06-19T13:52:06 DEBUG common.c: *** stlink_reset ***
2020-06-19T13:52:06 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-06-19T13:52:06 DEBUG common.c: Loading device parameters....
2020-06-19T13:52:06 DEBUG common.c: *** stlink_core_id ***
2020-06-19T13:52:06 DEBUG common.c: core_id = 0x2ba01477
2020-06-19T13:52:06 DEBUG common.c: *** stlink_read_debug32 10086437 is 0xe0042000
2020-06-19T13:52:06 DEBUG common.c: *** stlink_read_debug32 200 is 0x1ff800cc
2020-06-19T13:52:06 INFO common.c: L152RE: 80 KiB SRAM, 512 KiB flash in at least 256 byte pages.
2020-06-19T13:52:06 DEBUG common.c: *** stlink_reset ***
2020-06-19T13:52:06 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-06-19T13:52:06 DEBUG gdb-server.c: Chip ID is 0x00000437, Core ID is 0x2ba01477.
2020-06-19T13:52:06 INFO gdb-server.c: Listening at *:1234...
2020-06-19T13:52:19 DEBUG common.c: *** stlink_force_debug_mode ***
2020-06-19T13:52:19 DEBUG common.c: *** stlink_reset ***
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 3 to 0xe0002000
2020-06-19T13:52:19 DEBUG common.c: *** stlink_read_debug32 261 is 0xe0002000
2020-06-19T13:52:19 INFO gdb-server.c: Found 6 hw breakpoint registers
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0002008
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 0 to 0xe000200c
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0002010
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0002014
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0002018
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 0 to 0xe000201c
2020-06-19T13:52:19 DEBUG gdb-server.c: init watchpoints
2020-06-19T13:52:19 DEBUG common.c: *** stlink_read_debug32 1000000 is 0xe000edfc
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 1000000 to 0xe000edfc
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0001028
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0001038
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0001048
2020-06-19T13:52:19 DEBUG common.c: *** stlink_write_debug32 0 to 0xe0001058
2020-06-19T13:52:19 INFO gdb-server.c: GDB connected.
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+
2020-06-19T13:52:19 DEBUG gdb-server.c: query: Supported;multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+
2020-06-19T13:52:19 DEBUG gdb-server.c: send: PacketSize=3fff;qXfer:memory-map:read+
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: vMustReplyEmpty
2020-06-19T13:52:19 DEBUG gdb-server.c: send: 
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: Hg0
2020-06-19T13:52:19 DEBUG gdb-server.c: send: 
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: qTStatus
2020-06-19T13:52:19 DEBUG gdb-server.c: query: TStatus;
2020-06-19T13:52:19 DEBUG gdb-server.c: send: 
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: ?
2020-06-19T13:52:19 DEBUG gdb-server.c: send: S05
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: qfThreadInfo
2020-06-19T13:52:19 DEBUG gdb-server.c: query: fThreadInfo;
2020-06-19T13:52:19 DEBUG gdb-server.c: send: 
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: qL1200000000000000000
2020-06-19T13:52:19 DEBUG gdb-server.c: send: 
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: Hc-1
2020-06-19T13:52:19 DEBUG gdb-server.c: send: 
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: qC
2020-06-19T13:52:19 DEBUG gdb-server.c: send: 
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: qAttached
2020-06-19T13:52:19 DEBUG gdb-server.c: query: Attached;
2020-06-19T13:52:19 DEBUG gdb-server.c: send: 
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: qL1200000000000000000
2020-06-19T13:52:19 DEBUG gdb-server.c: send: 
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: qXfer:memory-map:read::0,1000
2020-06-19T13:52:19 DEBUG gdb-server.c: query: Xfer;memory-map:read::0,1000
2020-06-19T13:52:19 DEBUG gdb-server.c: Xfer: type:memory-map;op:read;annex:;addr:0;length:4096
2020-06-19T13:52:19 DEBUG gdb-server.c: send: m<?xml version="1.0"?><!DOCTYPE memory-map PUBLIC "+//IDN gnu.org//DTD GDB Memory Map V1.0//EN"     "http://sourceware.org/gdb/gdb-memory-map.dtd"><memory-map>  <memory type="rom" start="0x00000000" length="0x80000"/>  <memory type="ram" start="0x20000000" length="0x14000"/>  <memory type="flash" start="0x08000000" length="0x80000">    <property name="blocksize">0x100</property>  </memory>  <memory type="ram" start="0x40000000" length="0x1fffffff"/>  <memory type="ram" start="0xe0000000" length="0x1fffffff"/>  <memory type="rom" start="0x1ff00000" length="0x1000"/>  <memory type="rom" start="0x1ffff800" length="0x10"/></memory-map>
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: qXfer:memory-map:read::27e,1000
2020-06-19T13:52:19 DEBUG gdb-server.c: query: Xfer;memory-map:read::27e,1000
2020-06-19T13:52:19 DEBUG gdb-server.c: Xfer: type:memory-map;op:read;annex:;addr:638;length:4096
2020-06-19T13:52:19 DEBUG gdb-server.c: send: l
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: m8000384,4
2020-06-19T13:52:19 DEBUG common.c: *** stlink_read_mem32 ***
2020-06-19T13:52:19 DEBUG gdb-server.c: send: 0c488546
2020-06-19T13:52:19 DEBUG gdb-server.c: recv: m8000385,4
2020-06-19T13:52:19 DEBUG common.c: *** stlink_read_mem32 ***
2020-06-19T13:52:19 DEBUG gdb-server.c: send: 4885460c

Best regards,
Vincent.

vincentdupaquis, 2020-06-19:
attempting the same with an elf fails the same way
restore build/target.elf
gdb$ restore build/target.elf
Restoring section .isr_vector (0x8000000 to 0x80000c0)
Writing to flash memory forbidden in this context

vincentdupaquis, 2020-06-19:
Something weird.
I've tried using **version v1.5.1-patch**
The restore command fails the same, but the "**load build/target.elf**" command works :/

Nightwalker-87, 2020-06-21:
@vincentdupaquis: ... and what about v1.6.0 ?

vincentdupaquis, 2020-09-08:
Just to let you know that 2 months later, after a full re-install of my system, the problem is still there.
Same issue, different project. I can run the "file" command to load a binary, I cannot run the "restore" command to load a binary file.
I even figured-out that I cannot even use "set variable" to program a flash constant, I get the same exact error message.
We tried running into a docker (using alpine as a distribution), and it works like a charm.

vincentdupaquis, 2020-09-08:
I will make a try with version 1.6.0

stappersg, 2020-09-08:
On Tue, Sep 08, 2020 at 08:57:04AM -0700, vincentdupaquis wrote:
> We tried running into a docker, and it works like a charm.

Interesting,
I didn't knew docker can access hardware as a STlink programmer.

Which docker to use "extra hardware"?

vincentdupaquis, 2020-09-09:
Yes, it can you have to set-it up correctly for accessing the usb, as
well as the port used by st-utils for the gdb server port.

Very convinient for running non-regressions using for instance jenkins,
although it needs precise setup.

Vincent.

vincentdupaquis, 2020-09-09:
The exact same issue appears with version 1.6.0 (same system, Ubuntu 20.04)

vincentdupaquis, 2020-09-09:
The exact same issue appears with version 1.5.1 (same system, Ubuntu 20.04)

vincentdupaquis, 2020-09-09:
Cannot try with version 1.5.0, it does not compiles (issue with a missing entry point in libusb)

stappersg, 2020-09-10:
On Wed, Sep 09, 2020 at 12:05:26AM -0700, vincentdupaquis wrote:
> Le 08/09/2020 à 21:45, Geert Stappers a écrit :
> > On Tue, Sep 08, 2020 at 08:57:04AM -0700, vincentdupaquis wrote:
> > > We tried running into a docker, and it works like a charm.
> >
> > Interesting,
> > I didn't knew docker can access hardware as a STlink programmer.
> >
> > Which docker to use "extra hardware"?

    Which docker command option to use "extra hardware"?

> >
> Yes, it can you have to set-it up correctly for accessing the usb, as
> well as the port used by st-utils for the gdb server port.
> 
> Very convinient for running non-regressions using for instance jenkins,
> although it needs precise setup.

With `docker run --help` is option '--device list" shown.
How to use it?

vincentdupaquis, 2020-09-11:
Everything is here :
https://stackoverflow.com/questions/24225647/docker-a-way-to-give-access-to-a-host-usb-or-serial-device

you just have to use the stlink package from (for instance debian or else).

Nightwalker-87, 2021-04-03:
@vincentdupaquis: This seems to be an issue in `gdb` itself, see
https://gitter.im/blacksphere/blackmagic?at=5b5e1843d4527523f63cb11f.

Nightwalker-87, 2021-04-07:
@Ant-ON Could this also be related to a write protected chip itself?

Ant-ON, 2021-04-07:
@Nightwalker-87 More like the command is not implemented in the `st-util`. Other debuggers behave in a similar way: https://ncrmnt.org/2019/04/14/note-to-self-how-to-flash-bin-files-via-raw-gdb/comment-page-1/

Nightwalker-87, 2021-04-11:
> Indeed, I am not able to write flash through gdb, although I can through the stlink-gui tool.

How is this possible then?

Ant-ON, 2021-04-12:
@Nightwalker-87 it can by use elf file and `load` command. This issue can rename to feature request. The 'restore ' command is not implemented

Nightwalker-87, 2023-09-04:
Closing this feature request due to inactivity.

closed 2023-09-04 by Nightwalker-87