stlink-org/stlink #994: Has the -run option been removed?

Hoel, 2020-06-22:
I am using st-flash inside an application and I was using -rst -run as last argument in order to run the target right after flashing, yesterday i updated to 1.6.1 due to a client having issues on his macOS with the older version of st-flash i was using (libUSB related). The current version 1.6.1 works perfectly fine except it seems that the option -run has vanished, i can still use --reset, but i cannot use --run and when checking the source i couldnt find any reference to it either.
Has the run option been removed completely? (It was convenient).

Nightwalker-87, 2020-06-24:
Good question, if so it was not on purpose...
I'd need to check.

Hoel, 2020-06-24:
OK thanks, this option is quite useful.

Nightwalker-87, 2020-06-24:
Did you use v1.6.0 before?

Hoel, 2020-06-24:
i was using an older version, previous to 1.6 i think. And yes of course i am not using the offical ST_link utility, i was using Texane st-flash, which has apparently recently renamed.
The options -rst -run (at the very end of the command line) did work(at least st-flash did not complain about them) maybe they were ignored but the target was indeed resetted and running after flashing, which is not the case now with last rev.

Nightwalker-87, 2020-06-24:
It must have been without any effect then with the observed and behaviour resulting from another, likely unintended influence.
I can't tell why this was the case by then. Additionally there has been some refactoring and cleanup in the toolset by several contributors in the meanwhile. Of what I am fairly sure by now is, that `-rst` and `-run` never appeared in the st-flash related source code what must have been the case, if we were talking about a (intended) feature.

I'm afraid to say that looking at this, it seems as if we can't really do anything about this regarding a "fix".
You may open a new feature request with this functionality to be coded from scratch, but that would require someone to implement it. As we are short of active developers currently we may not see a working solution anytime soon.

Nightwalker-87, 2020-06-24:
> ... but the target was indeed resetted and running after flashing, which is not the case now with last rev.

There are several reset-related issues which are still open. Please have a look.
They should be all tagged with the label `error/reset` by now.

Hoel, 2020-06-25:
ok i see, these options were just ignored, i will check the existing issues.

Nightwalker-87, 2020-06-25:
Yeah, that would be reasonable. They are even grouped a separate (temporary) milestone.
I hope that we can find a solution for the upcoming release to resolve most of these with an improved routine in the code.
Reset problems clearly are a perennial issue.

Nightwalker-87, 2020-06-27:
As there is no correlation between the reported previous behaviour and the mentioned option, this issue is now solved.

closed 2020-06-27 by Nightwalker-87