stlink-org/stlink #995: STM32L476 DISCOVERY: Fails to write with large (e.g. > 64K) file size

tabemann, 2020-06-23:
This is a repeat of a previous ticket, except with a larger file size; increasing the timeout fixed the other ticket, but it still is too short for writing the file I tried to write today, which is 96256 bytes in size. I created a new ticket because that other ticket was already closed.

- Programmer/board type: Stlink /v2-1
- Programmer firmware version: L4xx
- Operating system and version: Linux sirius 5.5.0-1-amd64 #1 SMP Debian 5.5.13-2 (2020-03-30) x86_64 GNU/Linux
- **Stlink tools version** and/or git commit hash: v1.6.0-390-geeb3190
- Stlink commandline tool name: `st-flash`
- Target chip (and board if applicable): STM32L476 DISCOVERY

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:

```
OUTPUT/ERROR of the commandline tool(s)tabemann@sirius:~/projects/zeptoforth$ st-flash write bin/0.6.0/stm32l476/zeptoforth_full-0.6.0.bin 0x08000000
st-flash 1.6.0-390-geeb3190
2020-06-22T22:35:33 INFO common.c: L4xx: 96 KiB SRAM, 1024 KiB flash in at least 2 KiB pages.
file bin/0.6.0/stm32l476/zeptoforth_full-0.6.0.bin md5 checksum: 94b6a895d25c308dfc8e355d5c71f57, stlink checksum: 0x008618d7
2020-06-22T22:35:33 INFO common.c: Attempting to write 96256 (0x17800) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Page:0x0 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08000000 erased
EraseFlash - Page:0x1 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08000800 erased
EraseFlash - Page:0x2 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08001000 erased
EraseFlash - Page:0x3 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08001800 erased
EraseFlash - Page:0x4 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08002000 erased
EraseFlash - Page:0x5 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08002800 erased
EraseFlash - Page:0x6 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08003000 erased
EraseFlash - Page:0x7 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08003800 erased
EraseFlash - Page:0x8 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08004000 erased
EraseFlash - Page:0x9 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08004800 erased
EraseFlash - Page:0xa Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08005000 erased
EraseFlash - Page:0xb Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08005800 erased
EraseFlash - Page:0xc Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08006000 erased
EraseFlash - Page:0xd Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08006800 erased
EraseFlash - Page:0xe Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08007000 erased
EraseFlash - Page:0xf Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08007800 erased
EraseFlash - Page:0x10 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08008000 erased
EraseFlash - Page:0x11 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08008800 erased
EraseFlash - Page:0x12 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08009000 erased
EraseFlash - Page:0x13 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08009800 erased
EraseFlash - Page:0x14 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800a000 erased
EraseFlash - Page:0x15 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800a800 erased
EraseFlash - Page:0x16 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800b000 erased
EraseFlash - Page:0x17 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800b800 erased
EraseFlash - Page:0x18 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800c000 erased
EraseFlash - Page:0x19 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800c800 erased
EraseFlash - Page:0x1a Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800d000 erased
EraseFlash - Page:0x1b Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800d800 erased
EraseFlash - Page:0x1c Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800e000 erased
EraseFlash - Page:0x1d Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800e800 erased
EraseFlash - Page:0x1e Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800f000 erased
EraseFlash - Page:0x1f Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x0800f800 erased
EraseFlash - Page:0x20 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08010000 erased
EraseFlash - Page:0x21 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08010800 erased
EraseFlash - Page:0x22 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08011000 erased
EraseFlash - Page:0x23 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08011800 erased
EraseFlash - Page:0x24 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08012000 erased
EraseFlash - Page:0x25 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08012800 erased
EraseFlash - Page:0x26 Size:0x800 2020-06-22T22:35:33 INFO common.c: Flash page at addr: 0x08013000 erased
EraseFlash - Page:0x27 Size:0x800 2020-06-22T22:35:34 INFO common.c: Flash page at addr: 0x08013800 erased
EraseFlash - Page:0x28 Size:0x800 2020-06-22T22:35:34 INFO common.c: Flash page at addr: 0x08014000 erased
EraseFlash - Page:0x29 Size:0x800 2020-06-22T22:35:34 INFO common.c: Flash page at addr: 0x08014800 erased
EraseFlash - Page:0x2a Size:0x800 2020-06-22T22:35:34 INFO common.c: Flash page at addr: 0x08015000 erased
EraseFlash - Page:0x2b Size:0x800 2020-06-22T22:35:34 INFO common.c: Flash page at addr: 0x08015800 erased
EraseFlash - Page:0x2c Size:0x800 2020-06-22T22:35:34 INFO common.c: Flash page at addr: 0x08016000 erased
EraseFlash - Page:0x2d Size:0x800 2020-06-22T22:35:34 INFO common.c: Flash page at addr: 0x08016800 erased
EraseFlash - Page:0x2e Size:0x800 2020-06-22T22:35:34 INFO common.c: Flash page at addr: 0x08017000 erased
2020-06-22T22:35:34 INFO common.c: Finished erasing 47 pages of 2048 (0x800) bytes
2020-06-22T22:35:34 INFO common.c: Starting Flash write for F2/F4/L4
2020-06-22T22:35:34 INFO flash_loader.c: Successfully loaded flash loader in sram
size: 32768
2020-06-22T22:35:34 ERROR flash_loader.c: flash loader run error
2020-06-22T22:35:34 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
```

Expected/description:

`st-flash` would successfully write zeptoforth_full-0.6.0.bin to the STM32L476 MCU.

Nightwalker-87, 2020-06-24:
Can you please also test the latest release (v1.6.1) and the current `develop` branch?

tabemann, 2020-06-27:
I just tried with the latest `develop` branch (which was nearly identical to the `develop` branch in the state I tested it with) and got the same result.

Nightwalker-87, 2020-06-27:
Thanks for the feedback.
I suppose we need an option in the code to allow for additional flash sizes as there appear to be several hardware variants.
Against the current workload with maintenance of this project I can't promise that I find time to look at this though...

tabemann, 2020-06-27:
In `./src/stlink-lib/flash_loader.c` I changed `WAIT_ROUNDS` to `#define WAIT_ROUNDS 1000` and now it works.

Nightwalker-87, 2020-06-27:
Uh wait - we need to evaluate that. :eyes:
This #define was reduced to 30 recently (#987) to fix a regression.
We need to ensure that all boards work with this setting.

Nightwalker-87, 2020-06-28:
@geoffreymbrown: What do you think about this?

geoffreymbrown, 2020-06-28:
I'm not sure about there being a reduction -- the change I made did
decrease the outer loop to 30, but it also increased the wait time in the
inner loop so it wasn't banging on the usb connection as frequently.    The
total wait time went up.    The only issue I see is that I don't have a
good understanding of the actual physical requirements.   Increasing the
overall wait time shouldn't affect the successful writes -- only the time
it takes to fail.  This isn't likely to be a problem.

I don't personally know what it is measuring -- I had assumed it was the
time to write a page, so I don't have a grasp on why writing large numbers
of pages is causing this problem.

tabemann, 2020-09-05:
I pulled code recently (in the last few days) from the develop branch and built it, hoping this issue would have been resolved for the STM32L476, but it still has not been:

```
tabemann@sirius:~/projects/zeptoforth$ st-flash write bin/0.10.2/stm32l476/zeptoforth_swdcom-0.10.2.bin 0x08000000
st-flash 1.6.1-93-g5c03678
2020-09-05T13:21:40 INFO common.c: L4xx: 96 KiB SRAM, 1024 KiB flash in at least 2 KiB pages.
file bin/0.10.2/stm32l476/zeptoforth_swdcom-0.10.2.bin md5 checksum: a84ce1626af3bdec519d9e1c2bd44df, stlink checksum: 0x00a8571e
2020-09-05T13:21:40 INFO common.c: Attempting to write 129024 (0x1f800) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Page:0x0 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08000000 erased
EraseFlash - Page:0x1 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08000800 erased
EraseFlash - Page:0x2 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08001000 erased
EraseFlash - Page:0x3 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08001800 erased
EraseFlash - Page:0x4 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08002000 erased
EraseFlash - Page:0x5 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08002800 erased
EraseFlash - Page:0x6 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08003000 erased
EraseFlash - Page:0x7 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08003800 erased
EraseFlash - Page:0x8 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08004000 erased
EraseFlash - Page:0x9 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08004800 erased
EraseFlash - Page:0xa Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08005000 erased
EraseFlash - Page:0xb Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08005800 erased
EraseFlash - Page:0xc Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08006000 erased
EraseFlash - Page:0xd Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08006800 erased
EraseFlash - Page:0xe Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08007000 erased
EraseFlash - Page:0xf Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08007800 erased
EraseFlash - Page:0x10 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08008000 erased
EraseFlash - Page:0x11 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08008800 erased
EraseFlash - Page:0x12 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08009000 erased
EraseFlash - Page:0x13 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x08009800 erased
EraseFlash - Page:0x14 Size:0x800 2020-09-05T13:21:40 INFO common.c: Flash page at addr: 0x0800a000 erased
EraseFlash - Page:0x15 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x0800a800 erased
EraseFlash - Page:0x16 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x0800b000 erased
EraseFlash - Page:0x17 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x0800b800 erased
EraseFlash - Page:0x18 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x0800c000 erased
EraseFlash - Page:0x19 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x0800c800 erased
EraseFlash - Page:0x1a Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x0800d000 erased
EraseFlash - Page:0x1b Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x0800d800 erased
EraseFlash - Page:0x1c Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x0800e000 erased
EraseFlash - Page:0x1d Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x0800e800 erased
EraseFlash - Page:0x1e Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x0800f000 erased
EraseFlash - Page:0x1f Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x0800f800 erased
EraseFlash - Page:0x20 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08010000 erased
EraseFlash - Page:0x21 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08010800 erased
EraseFlash - Page:0x22 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08011000 erased
EraseFlash - Page:0x23 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08011800 erased
EraseFlash - Page:0x24 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08012000 erased
EraseFlash - Page:0x25 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08012800 erased
EraseFlash - Page:0x26 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08013000 erased
EraseFlash - Page:0x27 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08013800 erased
EraseFlash - Page:0x28 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08014000 erased
EraseFlash - Page:0x29 Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08014800 erased
EraseFlash - Page:0x2a Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08015000 erased
EraseFlash - Page:0x2b Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08015800 erased
EraseFlash - Page:0x2c Size:0x800 2020-09-05T13:21:41 INFO common.c: Flash page at addr: 0x08016000 erased
[... cut ...]
EraseFlash - Page:0x3e Size:0x800 2020-09-05T13:21:42 INFO common.c: Flash page at addr: 0x0801f000 erased
2020-09-05T13:21:42 INFO common.c: Finished erasing 63 pages of 2048 (0x800) bytes
2020-09-05T13:21:42 INFO common.c: Starting Flash write for F2/F4/F7/L4
2020-09-05T13:21:42 INFO flash_loader.c: Successfully loaded flash loader in sram
size: 32768
2020-09-05T13:21:42 ERROR flash_loader.c: flash loader run error
2020-09-05T13:21:42 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
```

cmdrf, 2020-10-23:
I seem to have the same problem with an STM32F767. And since I'm busy bisecting anyway:

> There are only 'skip'ped commits left to test.
> The first bad commit could be any of:
> 54e5627a269dddcd00fd1476f7dee62c5b207a52
> ff263131cf0d00657c308402762cdd591d313d5e
> We cannot bisect more!

54e5627a269dddcd00fd1476f7dee62c5b207a52 doesn't build because of undefined `__FreeBSD__`.

cmdrf, 2020-10-23:
Another attempt, this time with `--reset` and two `st-flash` runs each time to be sure. It points to c1ea72d842702160173504f4e5869b55b8e986d7. Sorry for the spam.

Nightwalker-87, 2020-10-24:
Sry, but that does not seem to help any further...

Nightwalker-87, 2021-04-11:
@Ant-ON Fixed?

Ant-ON, 2021-04-12:
@Nightwalker-87 yes, it closes by #1124

PR #1124 (2021-04-13): Improvements and fixes of the flash loaders, unification of the reset function
STM32L4 flash loader: fixed waiting for busy
STM32F1 flash loader: fixed setting of the PG flag
STM32Lx flash loader: fixed size of the flash loader
Added disabling of the DMA channels and watchdog timers while flash loader are running.
All possible reset options are combined into one function. When using the JTAG reset, the use of the pulse command has been removed. Instead, separate pull-down and pull-up commands are used.
Added target change message to the gdb server.

(Closes #244, Closes #382, Closes #705, Clos [... cut]
closed 2021-04-13 by Nightwalker-87