stlink-org/stlink #1000: PROJECT NEWS & UPDATES

Nightwalker-87, 2020-06-27:
This issue is solely dedicated for current project activities and project updates of general interest.

Nightwalker-87, 2020-06-27:
**List of maintainers for distribution repositories:**

- macOS (MacPorts): **chenguokai**
- Debian: **bluca**
- Ubuntu: **bluca**
- Fedora: **Vascom**
- EPEL (CentOS/RHEL): **Vascom**
- openSUSE: _(TBD)_ --> _Volunteers are welcome._
- Arch Linux: _(TBD)_ --> _Volunteers are welcome._

Nightwalker-87, 2020-06-27:
**NOTE: In order to support the evolution and progress of this project we are looking for ACTIVE DEVELOPERS to help with implementing new features and addressing bug reports. If you are reading this and are interested in contributing, please have a look at the issues currently open. Any help is highly appreciated.**

Nightwalker-87, 2020-06-27:
**General Project Chronology:**
15.03.2020 - Review of old issues completed
24.03.2020 - Reviewed and updated missing features
27.04.2020 - Added Contribution Guidelines

Nightwalker-87, 2020-07-31:
**NOTE: Against the background of vanishing community interest in active development is currently halted.
To make best use of limited time resources, I need to limit myself to maintenance tasks until further notice.**

xor-gate, 2020-08-09:
Dear @Nightwalker-87 as you are the most active maintainer of the project why was the issue tracker disabled? I got a private mail on my personal domain about a concern of the issue tracker not available. You should not disable the issue tracker unless there is a serious reason for it. Then also the landing page README.md must be updated. Thanks in advanced. So I re-enabled it again. 

Please let me know.

Nightwalker-87, 2020-08-09:
@xor-gate: I did disable it temporary as we are seeing more and more ticket duplicates and some users who don't seem to follow any guidelines we set. However I did truly forget to add a note to our readme on this.

Leaving this unaddressed will likely result in the state we found when I started to maintain this project sooner than later.

I have also written to GitHub if there is the possibility to allow for more options to configure who can open issues and automatic handling. By now there has been no related response yet, apart from that the feature appears to be "useful".

However as you re-enabled it now, I may also consider not to tag issues any more that do not follow our guidelines. Another alternative could be to simply delete such. What´s your opinion on this?

xor-gate, 2020-08-09:
Endless stream of issues which are duplicate is unavoidable except for placing a notice in the issue template tickets should be searched. Best would be to leave the issue tracker open, close the ticket and mention the duplicate id.

Nightwalker-87, 2020-08-09:
> Endless stream of issues which are duplicate is unavoidable except for placing a notice in the issue template tickets should be searched. Best would be to leave the issue tracker open, close the ticket and mention the duplicate id.

In fact we already have such note in our pinned README on opening of new issues. Also I'm not talking about one or another duplicate, which can happen sometime - no question. That's why I say it appears to be annoying and refers to the second part: 

> some users who don't seem to follow any guidelines we set

I already prohibited recycling of solved issues which is an improvement for maintenance, but that does not help at this point.

xor-gate, 2020-08-09:
You are unable to enforce given guidelines as the issue tracker is straightforward to use and configure, as you know it is a job to keep the issue tracker tidy. Probably leave tickets untagged which are unprocessed, tickets which are processed are tagged and can be filtered per tag. When the ticket doesn't match the criteria it should be closed without any notice.

Nightwalker-87, 2020-08-09:
I have created a new milestone "Unreviewed issues" and will preset our issue-templates to it. Any other issues that override this or don't use the templates will NOT be processed at all. Together with updating the mentioned README anything else will be deleted, as closing would result in additional handling to avoid recycling and also stacking up another needed milestone collecting declined issues, which alltogether results in a waste of time - time I'd rather spend in issues that are helpful to the project.

xor-gate, 2020-08-09:
I agree, good work.

Nightwalker-87, 2020-10-11:
**NOTE: Within this project the assignment of tickets on closure, as soon as a notable solution has been found, is common practise. This is done for statistical reasons in order to be able to keep track of contributors to this project (licensing).
If you have contributed to this toolset and don't want to receive any further notifications on assigned tickets in the future, you may unsubscribe from these issues by disabling related notifications anytime.**

Nightwalker-87, 2020-12-05:
@xor-gate: It appears we silently lost Travis CI functionality due to an upgrade of the bindings and integration. I've tried to reinstall it but it does not seem to work any more. :disappointed:  Is there anything we can do here?

xor-gate, 2020-12-05:
Travis seems to work for most build targets but e.g this one is broken https://travis-ci.org/github/stlink-org/stlink/jobs/747797289#L1123

xor-gate, 2020-12-05:
And this one https://travis-ci.org/github/stlink-org/stlink/jobs/747797286#L1004
It seems to work for most targets https://travis-ci.org/github/stlink-org/stlink/builds/747797280

Nightwalker-87, 2020-12-05:
I was requested to authorise travis _again_ upon access when clicking on one of those links. Before, there was no access nor any build status in the branch or PR view von Github for quite a while, so there was an error indeed. However this now seems to bring back the known features an I am able to have an eye on builds again. Thanks for the help.

Nightwalker-87, 2020-12-07:
Just found that it still is not functional as it used to be: No build status is available on the branches an PRs.

Nightwalker-87, 2020-12-12:
I thought so, as I got notified on the approval for access of "Travis CI for Open Source" for this project:

> An organization owner has approved the following application to access private data in the stlink-org organization:
>
>"Travis CI for Open Source" from Travis CI
>
> Because you are a member of the stlink-org organization and have already granted this application access to your personal account, Travis CI for Open Source can now access stlink-org organization resources on your behalf.

Obviously there was a mistake within the setup, though I granted access to the organisation as well from my personal account previously. I can confirm it is now working again (see last PR and the merge into `develop`). Thx for taking action here. :1st_place_medal:

Nightwalker-87, 2020-12-25:
Release v1.6.2 is now planned for Q1/2021.
Please help to resolve and close remaining issues with contributions and testing!

Nightwalker-87, 2021-03-26:
The next release **v1.6.2** is now planned for **Apr 2021**.

If you find this toolset useful and are willing to contribute, please check the remaining issues.
Note that contribution is not limited to coding - pre-release testing and feedback are just as helpful and always welcome.
Thank you for your support.

Nightwalker-87, 2021-05-21:
Note: The next release v1.7.1 is currently projected for Autumn 2021.

Nightwalker-87, 2021-05-24:
**List of maintainers for distribution repositories:**

- macOS (MacPorts): **chenguokai**
- Debian: _self-hosted_
- Ubuntu: _self-hosted_
- Fedora: **Vascom**
- EPEL (CentOS/RHEL): **Vascom**
- Arch Linux: _(TBD)_ --> _Volunteers are welcome._
- openSUSE: _(TBD)_ --> _Volunteers are welcome._

Nightwalker-87, 2021-05-24:
**General Project History:**

19.02.2020 - New Project Maintainer: **Nightwalker-87**
20.02.2020 - Release v1.6.0
15.03.2020 - Review of old issues completed
24.03.2020 - Reviewed and updated missing features
20.04.2020 - Project hosting transferred to new Git-Organisation **stlink-org**
27.04.2020 - Added Contribution Guidelines
01.06.2020 - Release v1.6.1
20.03.2021 - Surpassed the mark of **500** closed & resolved issues :tada: 
02.04.2021 - Added GitHub Actions: Automated CI & CodeQL Analysis
25.04.2021 - Release v1.7.0
24.05.2021 - Migrated project on Travis CI to travis-ci.com

Nightwalker-87, 2021-05-24:
**GENERAL NOTES:**
Within this project the assignment of tickets on closure, as soon as a notable solution has been found, is common practise. This is done for statistical reasons in order to be able to keep track of contributors to this project (licensing). If you have contributed to this toolset and don't want to receive any further notifications on assigned tickets in the future, you may unsubscribe from these issues by disabling related notifications anytime.

In order to support the evolution and progress of this project we are looking for ACTIVE DEVELOPERS to help with implementing new features and addressing bug reports. If you are reading this and are interested in contributing, please have a look at the issues currently open. Any help is highly appreciated.**

Nightwalker-87, 2021-07-14:
**NOTE:**
Open issues for the  next release (v1.7.1) can be found under Issues --> Milestones --> v1.7.1.

Interested users and developers are encouraged to check open tickets :pushpin: and help with implementing new features and addressing bug reports. We are looking forward to review old tickets especially to further reduce the current backlog :recycle:. Any help is highly appreciated, but please also consider the necessary time when doing so to avoid "dead" and incompletely resolved issues.

Nightwalker-87, 2021-09-26:
**NOTE:**

**Since September 2021 active development has ceased for this project.**

In the recent 1,5 years I have spent quite al lot of efforts to maintain and move forward with this project. When I stepped in, the remnants of many years without active maintenance had to be dealt with and worked up. Now with a deceasing amount of time available, I can only keep up some general maintenance tasks related to general project organisation and may push some single edits once in a while.

In order to move on with new releases we would clearly need developers contributing regularly, likely on a weekly basis and thus are willing to spend the time necessary to address and resolve open issues and bug reports. Looking at the codebase there is still a lot of work to do - on the internal structure as well as regarding to the user experience on the frontend, appearance and usability.

Otherwise this project will very likely remain in the current state. This would mean that there may only be some single bugfixes now an then, mainly dependant on frequent input from the community, but no functional updates and/or new features. Also new releases can not be planned against this background. Such a scenario could even more lead to frustrated users who are looking for alternatives, as we have seen it in the past. So the future evolution of this toolset is mainly up to you...

Nightwalker-87, 2022-02-07:
We now have a new project tab for discussions.
This section of the project is open for questions and comments related to the project itself as well as to the source code without the need to open a separate issue for this purpose. I intend to keep up with moderation and will try to answer open requests on a best effort basis.

Along with the completed refactoring and restructuring work in the codebase, let me announce that all temporary restrictions on external contribution have now expired.

Nightwalker-87, 2022-02-07:
This topic has moved to the section "Discussions".