stlink-org/stlink #1002: STM32F215: SP as LF and LF as SP

sidju, 2020-06-30:
- Programmer/board type: Stlink v2
- Programmer firmware version: How do I get this?
- Operating system and version: Arch Linux, updated yesterday
- **Stlink tools version** and/or git commit hash: Master release at 11/6 ( https://www.archlinux.org/packages/community/x86_64/stlink/ )
- Stlink commandline tool name: st-util
- Target chip (and board if applicable): STM32F215

**Problem:**
- When using gdb the "lr" register is shown as "sp" and vice versa.

**Expected/description:**
- I would expect the labels in gdb to correspond to how the CPU treats the registers.

Nightwalker-87, 2020-06-30:
>     * Programmer firmware version: How do I get this?

As you are under linux, you may use the ST-Link firmware upgrade tool based on Java to read out the current version:
https://www.st.com/en/development-tools/stsw-link007.html

As I could convince myself previously this also works with non-genuine ST programmers and boards.
Also these can safely be updated to the latest available version.
It even works with CKS32F1xxx clone ICs as can be found on some Bluepill-Boards and a few cheap STM32 programmers.

sidju, 2020-07-01:
Added programmer firmware version.

sidju, 2020-07-01:
I have now tested some more and found some other issues, such as breakpoints not working. Do you want them here or separately?
( I am currently refraining from upgrading my st-link, to be able to reproduce the errors. )

Nightwalker-87, 2020-07-01:
It depends - do you think it could be related? If so you may extend this ticket of course, else a new ticket would be appropriate.
Acutally you can update the firmware. Should this have any positive effect (for whatever reason) we will not look at this any further, because we don't maintain nor offer support for the programmer firmware. Of interest are only problems related to our `stlink` toolset.

sidju, 2020-07-09:
Updating the firmware definitely changes the issue. However I cannot tell what is what now, so I don't know how to proceed. I'll just use openocd instead and let someone with more of a clue and more normal hardware debug this.

Nightwalker-87, 2020-07-09:
@sidju: This ticket is NOT solved.
Closing of tickets in this project is done either by administrators or by automated ticket maintenance and not by anyone else.
Please unsubscribe from this issue instead.

Ant-ON, 2020-11-11:
@Nightwalker-87 this is a typical problem solved by #1027. I think we can close this issue

closed 2020-11-11 by Nightwalker-87