stlink-org/stlink #1003: st-flash --debug formatting problem

thatch, 2020-07-01:
When building from master, I get messages formatted like this from `stlink_print_data`:

Commandline-Output:

```
$ st-flash --debug --reset write foo.bin 0x8000000
...
2020-07-01T13:39:39 DEBUG common.c: data_len = 2 0x2
2020-07-01T13:39:39 DEBUG common.c:  082020-07-01T13:39:39 DEBUG common.c:  002020-07-01T13:39:39 DEBUG common.c: 

```

I believe this is due to `DLOG` outputting the timestamp on each call; this should probably prepare a buffer and call `DLOG` once.  I'll try to get to this.

Nightwalker-87, 2020-07-01:
Invalid ticket opening. Please fix via #905 or ticket will be deleted.

thatch, 2020-07-02:
I don't understand what 905 has to do with this. If I wasn't clear, the multiple timestamps on the same line is what I was noticing, not the received bytes.

Nightwalker-87, 2020-07-02:
Sry, there was a typo on the numbering, which has now been corrected - it should make sense now.

Nightwalker-87, 2020-07-05:
Closing this due to lack of response by the original issuer.

closed 2020-07-05 by Nightwalker-87