stlink-org/stlink #1005: STM32L562E-DK development board: read_debug32 returns only 0

ndusek, 2020-07-08:
- Programmer/board type: STM32L562E-DK development board
- Programmer firmware version:
  - Board firmware: 1.0.1
  - STLink version: 3
  - JTAG version: 5
- Operating system and version: Linux 5.6.0
- **Stlink tools version** and/or git commit hash: git hash commit be0157f63329ee48f272fd1a49e709a5ddb6183b
- Stlink commandline tool name: test-usb
- Target chip (and board if applicable): STM32L5

We are having an issue reading debug space on this board. `read_debug32` won't return anything but 0, regardless of what address is specified.

The board is advertising STLink V3SET device (VID=0x0483:PID=0x374f).

`test-usb` utility seems to indicate that some basic functionality works.

Core_ID of 0xbe12477 being returned by `test-usb` utility.  ST documentation seems to indicate that chip_id should be 0x477, which seems to be consistent with existing chip_id enumerations.

With `st-info --probe` test, word one of two word response contains 0x80 or 0x18.  Second word is always 0 which defeats the ability for chip probe to determine the chip_id.

ST documentation indicates that L5 board is a feature enhancement of
L4/L4+ boards but architecture/topology is largely the same.  Coded up
test chip identification structure based on STLINK_CHIPID_STM32_L46X
definition with debug code added to cause
`./src/common:stlink_chip_id()` to a chip_id value of 0x477.

ST documentation indicates that chip has dual bank support with
SRAM1=192K and SRAM2=64K.  SRAM2 is mapped at 0x0a030000 with second
mapping at 0x20030000 to be contiguous with SRAM1.

Flash size data register read for 0x1fff75e0 returns 0 with first word
of read being 0x18.  Additional work would appear to be problematic
without proper functionality of the `read_debug32` function.

Would appreciate any suggestions the group may have for further
investigation.

Nightwalker-87, 2020-07-08:
This board (L5 series) is not yet supported.

ndusek, 2020-07-09:
Thanks for the quick reply. We'll pull down the #999 feature set and see if that solves our problems. Looking forward to getting support for the L5 boards merged.

Ant-ON, 2020-07-09:
This board use Stlink-v3, but v3 protocol don't fully (with bugs) implemented. See #820 feature

Nightwalker-87, 2020-08-09:
Correct, referring to user feedback, both are currently the _main_ tasks that need to be addressed on the feature-request side.

Nightwalker-87, 2020-11-06:
What is the current state here?

Ant-ON, 2020-11-07:
There may be a problem with the reset as with STM32H7. If possible, @ndusek you can try this branch https://github.com/Ant-ON/stlink/tree/try_h7_debug

Nightwalker-87, 2021-03-22:
There will probably be no further response from the original author due to inactivity.

tarek-bochkati, 2021-03-26:
why are you reading this address 0x1fff75e0 
IFAICT flash size data register is @ 0x0BFA05E0
and L5 idcode is @ 0xE0044000

I wanted to check further, but reviewing this part [src/common.c](https://github.com/stlink-org/stlink/pull/999/files#diff-f6c5883a603e481644b9ef028b158acace9eabf59598f164dbec7b97d8da231f) was a nightmare 

maybe I read it wrong and you mean by the ticket to support the secure state ?

Nightwalker-87, 2021-03-26:
Related to #999.

Ant-ON, 2021-03-27:
@tarek-bochkati I looked carefully at the author's problem. It looks like he tried to add support for L5 based on L4, but didn't fix everything...

Nightwalker-87, 2022-01-03:
@Ant-ON @tarek-bochkati The referenced implementation is incomplete. As far as I can see from reviewing we need at least a chip-id file in order to support the mentioned board series. Can you also describe the necessary changes more precisely?

Nightwalker-87, 2022-01-03:
A first contribution to this topic is a (preliminary) chip-id file for the STM32L5 series in branch `chipid_cleanup`.

Nightwalker-87, 2022-01-12:
Please provide the full device-specific parametres as well as the ID of the respective ST reference manual.
This would allow to complete the generated chip file for this device type which currently serves as a template only.

Evidlo, 2022-05-03:
I'm also interested in this.  I've taken a stab at creating my own chip file, and commented on which page I pulled each parameter from.

```
# Chip-ID file for STM32L552 device
#
dev_type STM32L552
ref_manual_id 0438
chip_id 0x472                // RM0438, pg. 2157
flash_type L5_U5             // ???
flash_size_reg 0x0bfa05e0    // RM0438, pg. 2166
flash_pagesize 0x1000        // 4 KB - RM0438, pg. 178
sram_size 0x40000            // 256 KB - RM0438, pg. 94
bootrom_base 0x0bf90000      // RM0438, pg. 98/179
bootrom_size 0x8000          // 32 KB - RM0438, pg. 179
option_base 0x0              // ???
option_size 0x0              // ???
flags none                   // ???
```

I cross referenced `F1xx_MD.chip` and the corresponding manual and tried to pull parameters from equivalent sections.

I'm still not able to detect my device.  I'm not sure if there's some other changes I need to make to other header files?

Ant-ON, 2022-05-04:
@Evidlo Can you attach the application launch log? In theory, the chip should be determined.

Ant-ON, 2022-05-04:
@Evidlo Console log after running something like: `st-flash --connect-under-reset read firmware.bin 0x8000000 0x10`

Evidlo, 2022-05-12:
So I've made some progress:

```
[evan@blackbox ~] st-info --probe
Failed to parse flash type or unrecognized flash type

detected chip_id parametres

# Device Type: STM32L552
# Reference Manual: RM0438
#
chip_id 0x472
flash_type 10
flash_size_reg 0xbfa05e0
flash_pagesize 0x1000
sram_size 0x40000
bootrom_base 0xbf90000
bootrom_size 0x8000
option_base 0x0
option_size 0x0
flags 0

Found 1 stlink programmers
  version:    V2J39S7
  serial:     57FF6E067186555527410267
  flash:      524288 (pagesize: 4096)
  sram:       262144
  chipid:     0x472

detected chip_id parametres

# Device Type: STM32L552
# Reference Manual: RM0438
#
chip_id 0x472
flash_type 10
flash_size_reg 0xbfa05e0
flash_pagesize 0x1000
sram_size 0x40000
bootrom_base 0xbf90000
bootrom_size 0x8000
option_base 0x0
option_size 0x0
flags 0

  dev-type:   STM32L552
```

```
[evan@blackbox ~] st-flash --connect-under-reset read /tmp/firmware.bin 0x8000000 0x10
st-flash 1.7.0-186-gc4762e6-dirty
Failed to parse flash type or unrecognized flash type
2022-05-12T10:48:16 WARN common.c: NRST is not connected

detected chip_id parametres

# Device Type: STM32L552
# Reference Manual: RM0438
#
chip_id 0x472
flash_type 10
flash_size_reg 0xbfa05e0
flash_pagesize 0x1000
sram_size 0x40000
bootrom_base 0xbf90000
bootrom_size 0x8000
option_base 0x0
option_size 0x0
flags 0

2022-05-12T10:48:16 INFO common.c: STM32L552: 256 KiB SRAM, 512 KiB flash in at least 4 KiB pages.
2022-05-12T10:48:16 INFO common.c: read from address 0x08000000 size 16
```

And I get a 16 byte file filled with 1's.

Evidlo, 2022-05-12:
Not able to get `st-flash write` working yet.

I produced a simple LED blink `.bin` with stm32duino that is confirmed working.

This is the console output when stm32duino calls the official programming utility:

```
sh /home/evan/.arduino15/packages/STMicroelectronics/tools/STM32Tools/2.1.0/stm32CubeProg.sh 0 /tmp/arduino_build_427557/blink.ino.bin -g 
      -------------------------------------------------------------------
                        STM32CubeProgrammer v2.10.0                  
      -------------------------------------------------------------------

ST-LINK SN  : 57FF6E067186555527410267
ST-LINK FW  : V2J39S7
Board       : --
Voltage     : 3.23V
SWD freq    : 4000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x472
Revision ID : Rev Z
Device name : STM32L5xx
Flash size  : 512 KBytes
Device type : MCU
Device CPU  : Cortex-M33
BL Version  : --

Memory Programming ...
Opening and parsing file: blink.ino.bin
  File          : blink.ino.bin
  Size          : 20.09 KB 
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 10]
Download in Progress:

File download complete
Time elapsed during download operation: 00:00:11.043

RUNNING Program ... 
  Address:      : 0x8000000
Application is running, Please Hold on...
Start operation achieved successfully
```

And then when I try to program with `st-flash`:

```
[evan@blackbox ~] st-flash write /tmp/arduino_build_427557/blink.ino.bin 0x8000000
st-flash 1.7.0-186-gc4762e6-dirty
Failed to parse flash type or unrecognized flash type

detected chip_id parametres

# Device Type: STM32L552
# Reference Manual: RM0438
#
chip_id 0x472
flash_type 10
flash_size_reg 0xbfa05e0
flash_pagesize 0x1000
sram_size 0x40000
bootrom_base 0xbf90000
bootrom_size 0x8000
option_base 0x0
option_size 0x0
flags 0

2022-05-12T12:20:06 INFO common.c: STM32L552: 256 KiB SRAM, 512 KiB flash in at least 4 KiB pages.
file /tmp/arduino_build_427557/blink.ino.bin md5 checksum: 11eea395bf81feee318bcef406d42c, stlink checksum: 0x001e2964
2022-05-12T12:20:06 INFO common_flash.c: Attempting to write 20568 (0x5058) bytes to stm32 address: 134217728 (0x8000000)
2022-05-12T12:20:06 ERROR common_flash.c: method 'is_flash_busy' is unsupported
2022-05-12T12:20:06 ERROR common_flash.c: method 'is_flash_busy' is unsupported
2022-05-12T12:20:06 ERROR common_flash.c: method 'is_flash_busy' is unsupported
2022-05-12T12:20:06 ERROR common_flash.c: method 'is_flash_busy' is unsupported
2022-05-12T12:20:06 ERROR common_flash.c: method 'is_flash_busy' is unsupported
2022-05-12T12:20:06 ERROR common_flash.c: method 'is_flash_busy' is unsupported
...
```

Ant-ON, 2022-05-13:
@Evidlo the STM32_FLASH_TYPE_L5_U5 flash type is not implemented yet. Need to add this flash type (mostly add correct register addresses) to the functions in `src/common_flash.c` and `src/flashloader.c`. It is better to take STM32_FLASH_TYPE_G0/STM32_FLASH_TYPE_G4 types as a base. They are close and there is no need to write and compile a native flashloader.

Evidlo, 2022-05-18:
@Ant-ON see https://github.com/stlink-org/stlink/pull/1247

Nightwalker-87, 2022-12-31:
I'm currently trying to pick up the last known state from #1247, but we would need someone with the respective hardware to test the results in order to get along here.

Nightwalker-87, 2023-01-02:
@tarek-bochkati Can you review and/or contribute to this issue?
Am I right in assuming that you may even have (easy) access to the (L5 & U5) board types from your professional background?

Nightwalker-87, 2023-01-08:
We need people to test the changes and relevant features.

Evidlo, 2023-01-09:
I have hardware for the L552ZEQ, but #1247 being closed from just two weeks of inactivity after spending significant time on that PR was a pretty discouraging experience.

Nightwalker-87, 2023-03-18:
Can someone with a L5 board check PR #1300?
Tests on a U5 device have been carried out successfully. :wink:

Ant-ON, 2023-03-28:
@Nightwalker-87 "read_debug32" is a programmer's call. This does not apply to L5 support. This applies to STLink v3 support. STLink v3 seems to work well and there are no problems with memory reading (U5 as I see it works well). I think this issue has already been resolved.

PR #1300 (2023-04-06): Initial support for STM32 L5 & U5 devices and minor changes
- Initial support for STM32 L5 & U5 devices (Closes #1005) (Closes #1096) (Reference #1247)
- Updated CHANGELOG.md
- Updating windows signal handling to not terminate the application before clean-up.
- Log message improvements for st-flash
- Fix for option bytes on STM32G031G8 (Closes #1194)
closed 2023-04-06 by Nightwalker-87