stlink-org/stlink #1006: stlink-gui installation issue on Ubuntu-18.04

cpackham, 2020-07-09:
I see that PR #1004 has tried to address an issue of where `stlink-gui.ui` ends up getting installed. For me (on Ubuntu 18.04) this still appears to get it wrong.

If I build with 
```
make release
sudo make install
```
`stlink-gui.ui` ends up in `/usr/local/share/stlink/` which is where I'd expect

If I build with
```
make CMAKEFLAGS=-DCMAKE_INSTALL_PREFIX=/usr release
sudo make install
```

`stlink-gui.ui` ends up in `/usr/share/stlink/` (again as expected)

In either case the `stlink-gui` program seems to try and open `share/stlink/stlink-gui.ui` (note no leading `/`)

```
strace -e access stlink-gui 2>&1 | tail -n3
access("share/stlink/stlink-gui.ui", F_OK) = -1 ENOENT (No such file or directory)
Failed to load UI file: stlink-gui.ui
+++ exited with 1 +++
```

I don't know much about CMake but I assume this has something to do with the way the definition of STLINK_UI_DIR= is defined and how CMAKE_INSTALL_DATAROOTDIR is defined (or perhaps it needs to be "${CMAKE_INSTALL_PREFIX}/${CMAKE_INSTALL_DATAROOTDIR}).

Nightwalker-87, 2020-07-09:
Thx for reporting, I'll check ASAP.

lentychang, 2020-07-13:
@Nightwalker-87, according to @cpackham 

I change stlink/src/stlink-gui/CMakelist.txt line 37,38 into the following and it worked.
`set_target_property` requires absolute path and `CMAKE_INSTALL_DATAROOTDIR` has stripped `CMAKE_INSTALL_PREFIX` part away.
```
set_target_properties(stlink-gui PROPERTIES  
                COMPILE_DEFINITIONS STLINK_UI_DIR="${CMAKE_INSTALL_PREFIX}/${CMAKE_INSTALL_DATAROOTDIR}/${PROJECT_NAME}")
```

Nightwalker-87, 2020-07-17:
@cpackham & @lentychang: Thx for reporting. Looks good to me. :+1:
I can reproduce both, the error and the successful fix - it will be pushed upstream soon.

commit 56a699d3 (2020-07-17): Added missing cmake install prefix

(Closes #1006)
closed 2020-07-17 by Nightwalker-87