stlink-org/stlink #1007: [feature] Known good `st-info --probe`

stappersg, 2020-07-13:
- [x] Programmer/board type: Stlink/v3mini
- [ ] Programmer firmware version: dont know, might be the problem
- [x] Operating system: Linux, Debian bullseye
- [x] Stlink tools version: 1.6.1+ds-2,  2020-06-21
- [x] Stlink commandline tool name: `st-info`
- [x] Target chip (and optional board):  STM32L072,  PX-HERO,  https://www.crowdsupply.com/piconomix/px-her0-board/

Commandline-Output:

```
$ ./st-info --probe
Found 1 stlink programmers
 serial:     303033413030323233343338353130323334333133393339
 hla-serial: "\x30\x30\x33\x41\x30\x30\x32\x32\x33\x34\x33\x38\x35\x31\x30\x32\x33\x34\
\x33\x31\x33\x39\x33\x39"
 flash:      0 (pagesize: 0)
 sram:       0
 chipid:     0x0000
 descr:      unknown device
```

Description:

Issue #716 was closed without a _this is how it is supposed to look_,  without clearly _target must be present_.

Expected:

Have documented how a _known good_ looks, wether or not the target device must be present.

In other words:  Have #716 amended.

stappersg, 2020-07-13:
Additional information from the issue reporter:
* With or without the target board, I get the same output from `st-info --probe`.
* When target board connected to _stlink V3mini_ does a `st-info --probe` reset the board.

stappersg, 2020-07-14:
My research show that updating the firmware of my STLINK is an option.
It is also what https://github.com/stlink-org/stlink/issues/716#issuecomment-605114001 says.

Time will tell who was here first on reporting a known good of `st-info --probe`  ...   :smiley:

Nightwalker-87, 2020-07-17:
@stappersg: Does it work now, after updating the firmware?
There is no (one) "known good" really, as the displayed information is MCU dependent.
You can expect that the connected chip is properly detected if the empty fields as well as the `chipid` field are all filled out.
Please also refer to `doc/devices_boards.md` to verify a correct chip detection and compare the read memory sizes with the known specifications-

>     * My plan of getting a _#1007 mentions this issue_ in #716 failed  ... `:-/`

This is because editing resolved old tickets is locked to prevent recycling of old issues.

stappersg, 2020-07-18:
> Does it work now, after updating the firmware?

ST-LINK V3MINI updated with en.stsw-link007_V2-37-26.zip from https://www.st.com/content/st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-programmers/stsw-link007.html

I see **no** improvement. Thing I didn't noticed before that the LED at USB connector of the STLINKV3MINI goes from red to green after the `st-info --probe`.

This _all new to me_ was reason for buying several sets. I  can send a set to developer. Those who are interested can email me at  stappers at stappers dot it

Nightwalker-87, 2020-07-18:
This could be related to incomplete support of the STLink-v3 programmer series by this toolset as well...

stappersg, 2020-07-31:
With `stlink-gui`
* Chip:  unknown device
* Core:  0xbc11477
* Flash size: 0 kB
* RAM size: 0 kB

Clicking on `connect` gets LED near USB connector from red to green.
Doing a _read_ seems succesfull, but I get an empty file.

Other thing I did at my side is ordering another STLINK V3MINI, this time with adaptor to 2.54mm header plus Dupont cables. When it arrives is the plan to test it an known good STM32 device.

Nightwalker-87, 2020-07-31:
STLINK V3MINI is not officially or fully supported by this toolset.
Please consider this.

stappersg, 2020-10-05:
Reporting _now it works ..._
```
$ st-info --probe
Found 1 stlink programmers
 serial:     303033413030323233343338353130323334333133393339
 hla-serial: "\x30\x30\x33\x41\x30\x30\x32\x32\x33\x34\x33\x38\x35\x31\x30\x32\x33\x34\x33\x31\x33\x39\x33\x39"
 flash:      131072 (pagesize: 128)
 sram:       20480
 chipid:     0x0447
 descr:      L0xx Category 5
```

I can't tell what did the trick.
Might be related to doing `cargo flash`.  ( Yes,  `cargo` is the [Rust](https://www.rust-lang.org/) _package manager_.)

Nightwalker-87, 2020-10-06:
@stappersg Did you use a STLINK-V3 as well?

stappersg, 2020-10-06:
On Tue, Oct 06, 2020 at 11:48:40AM -0700, nightwalker-87 wrote:
> @stappersg Did you use a STLINK-V3 as well?

Yes. (It is the only ST-LINK I have.)

Getting the STLINK-V3 in the right state
is most likely be down by `cargo flash`.

`cargo flash` is based upon https://probe.rs

My plan is to update  the stlink documention
with a known good example.  (To be continued)

Nightwalker-87, 2020-10-06:
I see. However it would be helpful to understand the change an executed `cargo flash` introduces in reference to this stlink toolset.

stappersg, 2020-10-10:
> My plan is to update the stlink documention with a known good example. (To be continued)

There is now merge request #1049

closed 2020-10-10 by Nightwalker-87