stlink-org/stlink #1008: NUCLEO-767ZI (STM32F767ZI) unsupported?

tdjastrzebski, 2020-07-27:
Trying to debug NUCLEO-767ZI (STM32F767ZI) board I get the following error:

```
020-07-27T09:18:32 WARN common.c: Invalid flash type, please check device declaration
2020-07-27T09:18:32 ERROR gdb-server.c: Unsupported Target (Chip ID is 0000000000, Core ID is 0x5ba02477).
```

Does the above mean this board is unsupported?
Environment: Windows 10, VSCode, Corex-Debug, STLink 6.1

tdjastrzebski, 2020-07-28:
More logs.
```
st-flash 1.6.1
libusb: info [cache_config_descriptors] could not access configuration descriptor 1 (dummy) for 'USB\VID_045E&PID_07C6\000001000000': [31] A device attached to the system is not functioning.
2020-07-28T22:38:40 WARN common.c: unknown chip id! 0x1a
Failed to connect to target
The terminal process "C:\Program Files\Git\bin\bash.exe '-c', 'st-flash write 'C:\Test Projects\STM32F767ZI/BUILD/NUCLEO_F767ZI/GCC_ARM-DEBUG/STM32F767ZI.bin' 0x08000000'" terminated with exit code: 127.
```
```
st-flash 1.6.0
2020-07-28T22:36:19 INFO common.c: Loading device parameters....
2020-07-28T22:36:19 INFO common.c: Device connected is: F76xxx device, id 0x10016451
2020-07-28T22:36:19 INFO common.c: SRAM size: 0x80000 bytes (512 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 2048 bytes
2020-07-28T22:36:19 INFO common.c: Attempting to write 36812 (0x8fcc) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08008000 erasedEraseFlash - Sector:0x1 Size:0x8000 
2020-07-28T22:36:19 INFO common.c: Finished erasing 2 pages of 32768 (0x8000) bytes
2020-07-28T22:36:19 INFO common.c: Starting Flash write for F2/F4/L4
2020-07-28T22:36:19 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 32768
size: 4044
2020-07-28T22:36:20 INFO common.c: Starting verification of write complete
2020-07-28T22:36:20 INFO common.c: Flash written and verified! jolly good!
```
```
STM32 ST-LINK CLI v3.5.0.0
STM32 ST-LINK Command Line Interface

ST-LINK SN: 066BFF3035344E5043141238
ST-LINK Firmware version: V2J37M26
Connected via SWD.
SWD Frequency = 4000K.
Target voltage = 3.3 V
Connection mode: Normal
Reset mode: Hardware reset        
Device ID: 0x451 
Device flash Size: 2048 Kbytes
Device family: STM32F76x
Loading file...
Flash Programming:
  File : C:\Test Projects\STM32F767ZI/BUILD/NUCLEO_F767ZI/GCC_ARM-DEBUG/STM32F767ZI.bin
  Address : 0x08000000
Memory programming...
██████████████████████████████████████████████████ 100%
Memory programmed in 1s and 250ms.
Verification...OK
Programming Complete.
Programmed memory Checksum: 0x0034D2E0
```

tdjastrzebski, 2020-07-28:
**UPDATE**: I am unable to flash any Nucleo board using version **1.6.1** while version **1.6.0** works just fine.
Is it possible I screwed up the build? But build is so simple that it is nearly impossible to break anything.
One thing I did differently: I used cmake which comes with VS Studio but I do not see how this could cause a problem.

Nightwalker-87, 2020-07-31:
@tdjastrzebski: Please provide information on the used programmer.

blazczak, 2020-08-16:
I can confirm the same flashing issues as above (`unknown chip id`, `LIBUSB_ERROR_BUSY`) when using stlink 1.6.1 with STM32F767ZI . Downgrading to 1.6.0 allowed me to program the board w/o issues.

@Nightwalker-87 using the embedded ST-LINK/V2-1.

Nightwalker-87, 2021-03-25:
@Ant-ON Would it be possible to include this in your current branch `Core ID reading rework` ?
What does `WARN common.c: unknown chip id! 0x1a` stand for?

Ant-ON, 2021-03-26:
@Nightwalker-87 I can add an extra warning for the error reading the core_id. It must be readable for all Cortex-M targets. In the event of a read error, I add a warning that the target is not available for comunication (check the connection of the "SWD" pins; try connecting the `NRST` pin and use `--connect-under-reset`).

Nightwalker-87, 2021-03-26:
Yeah, that sounds good to me.

Ant-ON, 2021-03-27:
@Nightwalker-87 See https://github.com/Ant-ON/stlink/tree/chipid_read_rework

Nightwalker-87, 2021-03-31:
@tdjastrzebski @blazczak Have you experienced any change when testing with the current `develop` branch?

tdjastrzebski, 2021-04-01:
@Nightwalker-87 My focus is currently on relatively new STM32H7A3 so I stopped using stlink months ago but I will check.

Ant-ON, 2021-04-01:
@tdjastrzebski You can check STM32H7A3 too. This target must also be supported.

blazczak, 2021-04-11:
@Nightwalker-87 the version in `develop` seems to fix the `unknown chip id` `LIBUSB_ERROR_BUSY` errors.

```
st-flash --version
v1.6.1-277-g6f793d6
```

However, I don't quite remember having to reset manually at the end of each write. Is that a regression (no autoreset)? F7 board.

blazczak, 2021-04-11:
@Nightwalker-87 Auto-reset occurred after using v1.6.0 to write the same binary.

Nightwalker-87, 2021-04-11:
@blazczak The relevant changes should have been introduced with the merge of #1120.

Ant-ON, 2021-04-12:
@blazczak To reset the target after flashing, you need to use `--reset` parametr. By default, after the firmware, the code continues to be executed. Judging by the documentation (https://github.com/stlink-org/stlink/blob/develop/doc/tutorial.md), this has always been the case, perhaps there were some flaws leading to reset after flashing. 
When using `--connect-under-reset`, a new firmware will start after flashing, although there was no reset do.

Nightwalker-87, 2021-04-15:
@tdjastrzebski Can you give us an update on this?

Nightwalker-87, 2021-04-18:
Intermediate summary:

- _"Does the above mean this board is unsupported?"_
_"According to this list Core-ID: 0x5ba02477 is supported."_
--> The NUCLEO-767ZI board and the STM32F767 MCU are both supported by this toolset.

- _"Is incorrect Chip ID what causes the problem? It looks like Chip ID should start with 0x4xx."
"Is it some Chip ID parsing problem?"_
--> has been addressed in #1120.

- _"I am unable to flash any Nucleo board using version 1.6.1 while version 1.6.0 works just fine. (#1009) 
Is it possible I screwed up the build? But build is so simple that it is nearly impossible to break anything.
The only thing I did differently: I used cmake which comes with VS Studio but I do not see how this could cause a problem."_
--> Yes, it is possible to screw up the build if the previous one has not been cleaned properly with leftovers remaining in place. Detailled instructions on how to proceed can be found in `/doc/tutorial.md`

- _"Version 1.6.1 stops not where breakpoints are set and generally is not helpful so it may have a regression issue."_
--> has been addressed in #1071.

Nightwalker-87, 2021-04-18:
Please don't work with any older commits than 9c1315a73296841ed3c25b4ea9500c187ab8236d from this point.

@blazczak: Are there any further leftovers?

We need testing with **STM32F746**, **STM32F767** and **STM32F769** to find out if this regression from **v1.6.1** has been finally fixed.

Nightwalker-87, 2021-04-22:
@tarek-bochkati Do you have a F7 board around to give it a short test? I think there is not much to test, really.

Nightwalker-87, 2021-04-24:
> OpenOCD 0.10 is also having a similar problem with this board.

Together with the result of #1009 this indicates a local hardware or OS-specific problem, which appears not to be related to this toolset or even OpenOCD 0.10.

All other issues mentioned in the summary above should be fixed by #1124 at the latest.

closed 2021-04-24 by Nightwalker-87