stlink-org/stlink #1010: [STM32L051C8]: Can flash on command-line, can flash using GUI

p-rodrigues, 2020-07-31:
Hi,

I've looked in other issues and could not find anything equal to this:

Compiled from source-code and I can upload the code through stlink-gui. When uploading with stlink-gui the code appears to work perfectly fine on the hardware. 

However, when using st-flash I get the following error:
$st-flash --opt --freq=4m write file.hex 0x8000000
st-flash 1.6.1-86-gf162a5d
2020-07-31T15:23:16 INFO common.c: L0x3: 8 KiB SRAM, 64 KiB flash in at least 128 byte pages.
file file.hex md5 checksum: 5adcb1cef8433736cf5a8cd8acbfd82, stlink checksum: 0x0061899c
2020-07-31T15:23:16 INFO common.c: Attempting to write 123012 (0x1e084) bytes to stm32 address: 134217728 (0x8000000)
2020-07-31T15:23:16 ERROR common.c: addr too high
stlink_fwrite_flash() == -1

The file I am using is the same on the UI and on command-line. Yes the file is larger than 64kB, however, how can I upload it through the UI and not through the command-line? Am I doing something wrong on the command-line? I have already tried some combinations on the command-line parameters (the one I posted is just the last one I ran).

Info:
- Programmer/board type: Stlink v2
- Programmer firmware version: [enter here] (e.g STSW-LINK007 2.27.15)
- Operating system and version: Ubuntu 18.04 Linux
- **Stlink tools version** and/or git commit hash: v1.6.1-86-gf162a5d
- Stlink commandline tool name: st-flash
- Target chip (and board if applicable): STM32L051C8

stappersg, 2020-07-31:
On Fri, Jul 31, 2020 at 07:27:46AM -0700, p-rodrigues wrote:
> When uploading with stlink-gui the code appears to work perfectly fine on the hardware. 
> However, when using st-flash I get an error.

Here someone who is new to st-link,  I'm with st-link since #1007 `;-)`

I do read the issue as

  How to translate a succesfull stlink-gui action
  into stlink command line action?

I don't know, but I want to say
> Thank you for sharing that stlink-gui has something, maybe "knowledge",
> that makes succesfull flashing possible.

Cheers
Stappers

p-rodrigues, 2020-07-31:
> On Fri, Jul 31, 2020 at 07:27:46AM -0700, p-rodrigues wrote: When uploading with stlink-gui the code appears to work perfectly fine on the hardware. However, when using st-flash I get an error.
> Here [...]

I cannot say you are wrong :). It's just that I do not know what else to do.

Ant-ON, 2021-03-25:
@p-rodrigues Everything works correctly. You forgot to specify the firmware format. The problem is solved by adding `--format=ihex`

Nightwalker-87, 2021-03-25:
Closed by https://github.com/stlink-org/stlink/issues/1010#issuecomment-806578123.

closed 2021-03-25 by Nightwalker-87