stlink-org/stlink #1012: [ST-Link V2 clone]: "unknown chip id! 0x3748" after FW update to V2J37S7

JelmerT, 2020-08-05:
- Programmer/board type: [v2-clone]
- Programmer firmware version: [V2J37S7]
- Operating system and version: [macOS]
- **Stlink tools version** and/or git commit hash: [1.6.1] (homebrew version)
- Stlink commandline tool name: [All]
- Target chip (and board if applicable): [none]

Commandline-Output:

```
$ st-flash --debug erase                                                                                                                                                                             
st-flash 1.6.1
2020-08-05T00:04:19 DEBUG common.c: *** looking up stlink version
2020-08-05T00:04:19 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-08-05T00:04:19 DEBUG common.c: stlink pid     = 0x3748
2020-08-05T00:04:19 DEBUG common.c: stlink version = 0x2
2020-08-05T00:04:19 DEBUG common.c: jtag version   = 0x25
2020-08-05T00:04:19 DEBUG common.c: swim version   = 0x7
2020-08-05T00:04:19 DEBUG common.c: *** looking up stlink version
2020-08-05T00:04:19 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-08-05T00:04:19 DEBUG common.c: stlink pid     = 0x3748
2020-08-05T00:04:19 DEBUG common.c: stlink version = 0x2
2020-08-05T00:04:19 DEBUG common.c: jtag version   = 0x25
2020-08-05T00:04:19 DEBUG common.c: swim version   = 0x7
2020-08-05T00:04:19 DEBUG common.c: stlink current mode: mass
2020-08-05T00:04:19 DEBUG usb.c: JTAG/SWD freq set to 0
2020-08-05T00:04:19 DEBUG common.c: *** set_swdclk ***
2020-08-05T00:04:19 DEBUG common.c: stlink current mode: mass
2020-08-05T00:04:19 DEBUG common.c: *** stlink_enter_swd_mode ***
2020-08-05T00:04:19 DEBUG common.c: *** stlink_jtag_reset ***
2020-08-05T00:04:19 DEBUG common.c: *** stlink_reset ***
2020-08-05T00:04:19 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-08-05T00:04:19 DEBUG common.c: Loading device parameters....
2020-08-05T00:04:19 DEBUG common.c: *** stlink_core_id ***
2020-08-05T00:04:19 DEBUG common.c: core_id = 0x00003748
2020-08-05T00:04:19 DEBUG common.c: *** stlink_read_debug32 3748 is 0xe0042000
2020-08-05T00:04:19 WARN common.c: unknown chip id! 0x3748
Failed to connect to target

$ st-info --probe                                                                                                                                                                                    
Found 1 stlink programmers
 serial:     513f6a06506752541626213f
 hla-serial: "\x51\x3f\x6a\x06\x50\x67\x52\x54\x16\x26\x21\x3f"
 flash:      0 (pagesize: 0)
 sram:       0
 chipid:     0x0748

$ st-util -v99                                                                                                                                                                                                
st-util
2020-08-05T00:05:06 DEBUG common.c: *** looking up stlink version
2020-08-05T00:05:06 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-08-05T00:05:06 DEBUG common.c: stlink pid     = 0x3748
2020-08-05T00:05:06 DEBUG common.c: stlink version = 0x2
2020-08-05T00:05:06 DEBUG common.c: jtag version   = 0x25
2020-08-05T00:05:06 DEBUG common.c: swim version   = 0x7
2020-08-05T00:05:06 DEBUG common.c: *** looking up stlink version
2020-08-05T00:05:06 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-08-05T00:05:06 DEBUG common.c: stlink pid     = 0x3748
2020-08-05T00:05:06 DEBUG common.c: stlink version = 0x2
2020-08-05T00:05:06 DEBUG common.c: jtag version   = 0x25
2020-08-05T00:05:06 DEBUG common.c: swim version   = 0x7
2020-08-05T00:05:06 DEBUG common.c: stlink current mode: mass
2020-08-05T00:05:06 DEBUG usb.c: JTAG/SWD freq set to 0
2020-08-05T00:05:06 DEBUG common.c: *** set_swdclk ***
2020-08-05T00:05:06 DEBUG common.c: stlink current mode: mass
2020-08-05T00:05:06 DEBUG common.c: *** stlink_enter_swd_mode ***
2020-08-05T00:05:06 DEBUG common.c: *** stlink_jtag_reset ***
2020-08-05T00:05:06 DEBUG common.c: *** stlink_reset ***
2020-08-05T00:05:06 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-08-05T00:05:06 DEBUG common.c: Loading device parameters....
2020-08-05T00:05:06 DEBUG common.c: *** stlink_core_id ***
2020-08-05T00:05:06 DEBUG common.c: core_id = 0x00003748
2020-08-05T00:05:06 DEBUG common.c: *** stlink_read_debug32 3748 is 0xe0042000
2020-08-05T00:05:06 WARN common.c: unknown chip id! 0x3748
2020-08-05T00:05:06 DEBUG common.c: *** stlink_reset ***
2020-08-05T00:05:06 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-08-05T00:05:06 DEBUG gdb-server.c: Chip ID is 0x00000748, Core ID is 0x003748.
2020-08-05T00:05:06 INFO gdb-server.c: Listening at *:4242...
```

Possible duplicate of #956

V2 clone was working just fine, noticed it was on older firmware, upgraded it with `STSW-LINK007` (2.37.26) - `STLinkUpgrade 3.3.4` to version  `V2J37S7`.

Upgrade went fine, but independent of what is connected to the programmer I'm getting the `unknown chip id! 0x3748` error.
Tested with known good board (bluepill).

[this comment](https://github.com/stlink-org/stlink/issues/956#issuecomment-638268510) in #956 reports `1.6.1` working with a V2 Clone with firmware `V2J37S7`. So I have no clue what's going wrong here.

Nightwalker-87, 2020-08-09:
@JelmerT: Have you tried to overwrite the firmware on the programmer again?

JelmerT, 2020-08-13:
@Nightwalker-87 do you mean re-flashing the st-link clone to `V2J37S7` with the ST upgrade tool? -> Yes tried that a couple times. No difference.

I have not tried downgrading it to an earlier version. Is this possible? (I'm struggling making earlier versions of the tool work on macOS)

dwinters42, 2020-08-19:
- Programmer/board type: [v2-clone]
- Programmer firmware version: [V2J36S7] and [V2J37S7]
- Operating system and version: [macOS Catalina]
- Stlink tools version and/or git commit hash: [1.6.1] (homebrew version) + lastest master baab8ca

FWIW, I can reproduce the same behavior as seen by @JelmerT with several bluepill (STM32F103) boards on both 1.6.1 and latest master after a recent firmware update to versions V2J36S7 and V2J37S7.

Interestingly,  STM32F401 controllers still get detected fine:

$ st-info --probe
Found 1 stlink programmers
 serial:     573f6f06483f52553241063f
 hla-serial: "\x57\x3f\x6f\x06\x48\x3f\x52\x55\x32\x41\x06\x3f"
 flash:      262144 (pagesize: 16384)
 sram:       65536
 chipid:     0x0423
 descr:      F4xx (low power)

Nightwalker-87, 2020-08-19:
@JelmerT: I don't know if a downgrade is possible. Maybe you can try an older version of the firmware updater?

alexandrefdutra, 2020-08-19:
Same problem here! 

$ st-info --probe
Found 1 stlink programmers
 serial:     46310a002c135737334d4e00
 hla-serial: "\x46\x31\x0a\x00\x2c\x13\x57\x37\x33\x4d\x4e\x00"
 flash:      0 (pagesize: 0)
 sram:       0
 chipid:     0x0748

> @JelmerT: I don't know if a downgrade is possible. Maybe you can try an older version of the firmware updater?

I've done some downgrades but it didn't solve. I can only write once and then I lose the connection with the message unknown chip id! 0x3748!  I don't know what else to do.

larsks, 2020-08-29:
I came by to report the same problem, but I found this issue already here. I have an st-link v2 clone, and I upgraded it to the latest firmware while trying to get it working with the arduino ide. Now it reports "chipid: 0x0748", and and flashing operations seem to fail with `unknown chip id!` errors.

I've tried downgrading to 2.32.22, which is the earliest version offered on https://www.st.com/en/development-tools/stsw-link007.html, but the problem persists.

```
$ st-info --probe
Found 1 stlink programmers
 serial:     343f72064155353410300257
 hla-serial: "\x34\x3f\x72\x06\x41\x55\x35\x34\x10\x30\x02\x57"
 flash:      0 (pagesize: 0)
 sram:       0
 chipid:     0x0748
```

dltech, 2020-08-30:
> clive1 (NFA Crew) (Community Member)
> Edited by STM Community July 21, 2018 at 5:44 PM
> Posted on March 06, 2014 at 16:28
> Google had these when I went fishing yesterday..
>  
> https://drive.google.com/folderview?id=0Bzv7UpKpOQhnbXJVVEg4VUo2M1k

Clones with new firmware does not work at all, but with V12.J21.S4 works well, I downgraded my clone in the Virtualbox. By the way if you are interested in my homemade clone you can find it here https://github.com/dltech/stlinkv2.

kusumah99, 2020-10-14:
I'm using mac, it only happen with my STM32H743, not my STM32F4xx using V2J37S7, but downgrading stlink fw not helped.
I solved all problems by:
1. Use DFU Mode (Boot0 = 3.3v)
2. Connect using stlink or USB
3. st-flash or st-info CLI won't work with my chip, so I use STM32CubeIDE (using stlink) and STM32CubeProgrammer (using stlink or USB) they will work.
4. Power up
5. Connect
6. Chip Erase (if SWDIO and SWDCLK pins used by your code) (PA13 and PA14)

Ant-ON, 2020-11-12:
It looks like a reset issue. The fix can be checked in this branch: https://github.com/Ant-ON/stlink/tree/try_h7_debug

2ni, 2020-12-20:
When compiling  try_h7_debug I get the following errors:
`
Scanning dependencies of target stlink-static
[  2%] Building C object CMakeFiles/stlink-static.dir/src/common.c.o
[  5%] Building C object CMakeFiles/stlink-static.dir/src/stlink-lib/chipid.c.o
[  8%] Building C object CMakeFiles/stlink-static.dir/src/stlink-lib/flash_loader.c.o
[ 11%] Building C object CMakeFiles/stlink-static.dir/src/stlink-lib/logging.c.o
[ 13%] Building C object CMakeFiles/stlink-static.dir/src/stlink-lib/md5.c.o
[ 16%] Building C object CMakeFiles/stlink-static.dir/src/stlink-lib/sg.c.o
[ 19%] Building C object CMakeFiles/stlink-static.dir/src/stlink-lib/usb.c.o
/www/lora-e5/stlink-try_h7_debug/src/stlink-lib/usb.c:1121:75: error: implicit conversion loses integer precision: 'size_t' (aka 'unsigned long') to 'int'
      [-Werror,-Wshorten-64-to-32]
        int t = libusb_bulk_transfer(slu->usb_handle, slu->ep_trace, buf, trace_count, &res, 3000);
                ~~~~~~~~~~~~~~~~~~~~                                      ^~~~~~~~~~~
/www/lora-e5/stlink-try_h7_debug/src/stlink-lib/usb.c:1129:12: error: implicit conversion loses integer precision: 'size_t' (aka 'unsigned long') to 'int'
      [-Werror,-Wshorten-64-to-32]
    return trace_count;
    ~~~~~~ ^~~~~~~~~~~
2 errors generated.
make[3]: *** [CMakeFiles/stlink-static.dir/src/stlink-lib/usb.c.o] Error 1
make[2]: *** [CMakeFiles/stlink-static.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [release] Error 2
`

Ant-ON, 2020-12-21:
@2ni Now all the changes are merged in `develop`. Compilation issue fixed. Can you try the `develop` branch instead?

2ni, 2020-12-21:
I uploaded some code with cubeIDE which is running on a STM32WLE5JB module connected to a chinese stlink v2. This is what I get now:
```
09:38 $ ./build/Release/bin/st-info --probe
Found 1 stlink programmers
  version:    V2J37S7
  serial:     321c130012144d43574d4e00
  hla-serial: "\x32\x1c\x13\x00\x12\x14\x4d\x43\x57\x4d\x4e\x00"
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x001f
  ```
  
  I hoped to get rid of cubeIDE...

elliott-maguire, 2020-12-27:
Programmer/board type: V2 clone (initeq)
Programmer firmware version: V2J36S7
Operating system and version: Linux (5.9.11-3-MANJARO)
ST-Link version: Latest (`develop
Target chip/board: STM32F103C8T6/Blue Pill (initeq)

```
$ st-info --probe
  Found 1 stlink programmers
    version:    V2J36S7
    serial:     553f6a06713f535339150867
    hla-serial: "\x55\x3f\x6a\x06\x71\x3f\x53\x53\x39\x15\x08\x67"
    flash:      0 (pagesize: 0)
    sram:       0
    chipid:     0x0748
$ st-flash read test.bin 0 0xFFFF
  st-flash 1.6.1-200-gddac731
    2020-12-27T11:16:13 WARN usb.c: NRST is not connected
    2020-12-27T11:16:13 WARN common.c: unknown chip id! 0x3748
    Failed to connect to target
```

@Ant-ON What was the context in which you got it working?

geocine, 2020-12-28:
Programmer/board type: V2 clone
Programmer firmware version: V2J37S7
Operating system and version: Mac OS (10.14.6)
ST-Link version: v1.6.1-200-gddac731 (latest develop)
Target chip/board: STM32F103C8T6/Blue Pill 
```
$ st-info --probe
 Found 1 stlink programmers
   version:    V2J37S7
   serial:     0a240f002b135937334d4e00
   hla-serial: "\x0a\x24\x0f\x00\x2b\x13\x59\x37\x33\x4d\x4e\x00"
   flash:      0 (pagesize: 0)
   sram:       0
   chipid:     0x0748
```

geocine, 2020-12-28:
I managed to solve this problem, by following what has been stated [here](https://electronics.stackexchange.com/questions/204996/stm32-st-link-cannot-connect-to-mcu-after-successful-programming) and [here](https://stackoverflow.com/questions/57628401/eclipse-error-init-mode-failed-unable-to-connect-to-the-target)

You need a Windows computer and the [ST-Link Utility](https://www.st.com/en/development-tools/stsw-link004.html)
1. In ST-Link Utility go to Settings -> Mode -> Connect under reset
2. Then connect the last board you used before it malfunctioned with your ST-Link V2 and hold RESET button of that board
3. Click "connect to the target" , just after clicking, release the RESET button
2. Then click "Target" -> "Erase chip" 

You can now reprogram your MCU. 

While I don't know the exact reason what causes this, ***I hope this feature can be added unto this `st-flash` as well***.

Ant-ON, 2020-12-29:
@geocine `st-flash` alredy have `--connect-under-reset` options. But there is no information about it in the help. Can you try it option?

geocine, 2020-12-29:
@Ant-ON  
I just tried literally just now, yes it is working I will update my comment. Have to quickly execute the flash with `connect-under-reset` once you tap reset , 
```
st-flash --connect-under-reset —reset --format binary write bootloader-generic.bin 0x8000000
```

Nightwalker-87, 2020-12-29:
So this reads as if this issue is about to be resolved very soon...

geocine, 2020-12-29:
I think I derailed the issue a little bit, the way I understood the **original** issue is `st-*` tools detecting the chip as `chipid:     0x0748` in the following scenarios
- No target chip/board is connected to programmer
- A target chip/board is connected but cannot be detected (due to a number of reasons)

Nightwalker-87, 2020-12-30:
Another case: https://community.st.com/s/question/0D53W00000Ji4awSAB/programming-stm32f103zet6-or-stm32f105vct6-for-the-very-first-time-using-stlink-utility

digitalentity, 2021-02-20:
FWIW I'm also affected bt the same issue. On firmware V2J37S7 `st-flash` doesn't work

Nightwalker-87, 2021-03-11:
I can confirm that `st-flash` is working with STLink/V2-clone, FW version V2J37S7 and v1.6.1:

```
$ st-info --probe
Found 1 stlink programmers
 serial:     3f70050132124647524b4e00
 hla-serial: "\x3f\x70\x05\x01\x32\x12\x46\x47\x52\x4b\x4e\x00"
 flash:      65536 (pagesize: 1024)
 sram:       20480
 chipid:     0x0410
 descr:      F1xx Medium-density
```

```
$ st-flash write /[...]/Blinker.bin 0x8000000
st-flash 1.6.1
2021-03-11T21:45:23 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
file /[...]/Blinker.bin md5 checksum: 16277ff2c027d497361dbae4255081a1, stlink checksum: 0x0004644a
2021-03-11T21:45:23 INFO common.c: Attempting to write 3520 (0xdc0) bytes to stm32 address: 134217728 (0x8000000)
2021-03-11T21:45:23 INFO common.c: Flash page at addr: 0x08000000 erased
2021-03-11T21:45:23 INFO common.c: Flash page at addr: 0x08000400 erased
2021-03-11T21:45:23 INFO common.c: Flash page at addr: 0x08000800 erased
2021-03-11T21:45:23 INFO common.c: Flash page at addr: 0x08000c00 erased
2021-03-11T21:45:23 INFO common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2021-03-11T21:45:23 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2021-03-11T21:45:23 INFO flash_loader.c: Successfully loaded flash loader in sram
  4/4 pages written
2021-03-11T21:45:23 INFO common.c: Starting verification of write complete
2021-03-11T21:45:23 INFO common.c: Flash written and verified! jolly good!
```

salacpavel, 2021-03-24:
Experiencing some issue with GD32F105RBT6. I was able to flash once, but I get unknown chip id! 0x4278b7 ever since then.
As this is no target board with no rst pins exposed, I would like to ask you guys to release 1.6.1 with bugfix backported, please?
As I assume 1.6.2 release date is not set.

dltech, 2021-03-24:
> Experiencing some issue with GD32F105RBT6. I was able to flash once, but I get unknown chip id! 0x4278b7 ever since then.
Unknown chip ID usually means bad physical connection, check supply voltage and mcu connection.

Ant-ON, 2021-03-24:
@salacpavel Try add `--connect-under-rese` to command line

salacpavel, 2021-03-24:
> @salacpavel Try add `--connect-under-rese` to command line

```
C:\apps\stlink-1.6.1\bin>st-flash --format ihex --connect-under-reset write 323.hex
st-flash 1.6.1
[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_READCOREID
2021-03-24T10:49:00 ERROR common.c: Failed to read core_id
[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
2021-03-24T10:49:00 WARN common.c: unknown chip id! 0x4278b7
Failed to connect to target
```

dltech, 2021-03-24:
@salacpavel  --connect-under-reset affects reset pin, but we do not have it.

dltech, 2021-03-24:
libusb error? try "apt install stlink-tools" in ubuntu

Ant-ON, 2021-03-24:
@salacpavel Are you using stlink v1?

salacpavel, 2021-03-24:
some clone of ST-LINK v2. And I am on Windows, sadly. I should highlight I am not having such a problem with supposedly genuine STM32F105RBT6, I am seeing this with GD32F105RBT6 by GigaDevice only so far. hese two are supposed to be compatible tho. And the first flash went all right.

digitalentity, 2021-03-24:
@salacpavel the firmware you flashed on your GD32F105RBT6 probably remapped the SWD pins (used them as GPIO, AF or entirely disabled JTAG/SWD module). The only way to recover is to connect under reset unfortunately.

salacpavel, 2021-03-24:
Ok, I have disassembled the board and managed to connect jumper wire to nrst pin (7). Once ground, I run:
C:\apps\stlink-1.6.1\bin>st-flash --format=ihex --reset --connect-under-reset write 323.hex
st-flash 1.6.1
2021-03-24T11:11:59 WARN common.c: unknown chip id! 0xffffffff
Failed to connect to target

Nightwalker-87, 2021-03-24:
@salacpavel I don't really see **_any_** relation to this issue. Not the same chip-id error, not the same chip, no obvious coincidence of your problem with the mentioned programmer firmware version and also the chip id error message is different. **_Please open a new ticket and do not recycle this one._** To answer your question: there will be no patches or any maintenance for older versions of this toolset. Users are encouraged to work with the latest version or the `develop` tree if necessary.

Nightwalker-87, 2021-03-25:
This reason for the appearance of `unknown chip id! 0x3748` is now also mentioned in our tutorial.

Nightwalker-87, 2021-03-26:
@digitalentity @geocine What is the current state of this by now?
Can you re-test with the latest `develop`and summarize your findings?

digitalentity, 2021-03-27:
@Nightwalker-87 my original issue reported here https://github.com/stlink-org/stlink/issues/1012#issuecomment-782602708 was root-caused to a faulty ST-Link/V2 clone. No idea why it works on earlier firmwares, but V2J37S7 makes the device useless regardless of `st-flash` version. I've replaced the debugger with another cloned ST-Link/V2 which works perfectly fine with either `1.6.1` or the latest `develop` and firmware V2J37S7.

Nightwalker-87, 2021-03-28:
The input by @geocine has been added to our tutorial in commit  229c721189587760db5509c59b3c02e93e7035c8.
Based on the issue context and the feedback from @digitalentity we can conclude that some (obviously cheap) ST-LINK/V2 clone programmers brick themselves on official firmware updates, which makes this topic related to individual hardware.

To give a statement on this I personally can't generally recommend not to use ST-LINK/V2 clone programmers, as quite a few seem to have a good cost benefit ratio and are implemented quite well. However one should be aware of that there is scrap on the market as well. The best advise I can give is to check reviews and sites like e.g. this one: https://wiki.cuvoodoo.info/doku.php?id=jtag to help finding the appropriate ones. I personally have two STLINK/V2 clones around which have so far accepted all official firmware updates without any special occurrences.

Against this background we can not allocate a general malfunction in the toolset and I am closing this issue now as resolved.

closed 2021-03-28 by Nightwalker-87

Rombersoft, 2025-05-11:
> 1. In ST-Link Utility go to Settings -> Mode -> Connect under reset
After this action I got error:

[image]

I've tried to set all Frequency mode but without success.