stlink-org/stlink #1014: [feature] Tag  "considered a duplicate so closed"

stappersg, 2020-08-09:
The tag "considered a duplicate so closed" is for  quickly closing issues that a considered duplicated.

It purpose is two sided:
* Developers close issues without wasting energy on explaining it is a duplicate
* Users spent energy on adding information on issue and so helping closing issues

The template of new issue has 
- [ ] I made serious effort to avoid creating duplicate / nearly similar issue

This tag is an alternative for _milestone "Unreviewed issues", preset in issue-templates_ from  https://github.com/stlink-org/stlink/issues/1000#issuecomment-671040925

The idea behind the different approach is that good new issues  are just good, developers don't need to remove the "Unreviewed issues" tag.   Poor issues can quickly be closed by developers without energy drain and the lazy  and/or cheap users do get some feedback.   Yes, quite a minimal feedback, but hey, they started with minimal effort.

Nightwalker-87, 2020-08-09:
We have a "duplicate" tag already and it would be preferable to close duplicates along with linked PRs in order to ease changelog referencing afterwards (instead of e.g. searching the history for duplicates). Also we use automated GitHub issue handling to close linked issues that have been solved in PRs without further interaction. Your proposal would open up a parallel and manual handling path. Lastly this does not take the codebase any further, if the actual issue remains unsolved. The main intention should be to ease maintenance and support dev-work from my point of view.

stappersg, 2020-08-09:
Yes, supporting
> The main intention should be to ease maintenance and support dev-work from my point of view.

Feel free to add
```
- [ ] I made serious effort to avoid creating duplicate or nearly similar issue
```
to the  issue template.

And please be  _straight_,  _bold_ or even _rude_ in tagging issues `duplicate` and close it without further explaining.

Don't let [Eternal September](https://en.wikipedia.org/wiki/Eternal_September) wear you down.

Nightwalker-87, 2020-08-09:
@stappersg: Don't worry about this too much, our guidelines and instructions are not just for show. I tend to be very consequent on careless interaction. Thx for the input and support from your side. :+1: Looks like this topic hit a nerve somehow. Taking into account your suggestion, I'll give the issue templates another update soon. Thus I'm closing this maintenance-related feedback issue now.

commit baab8ca2 (2020-08-15): Updated ISSUE_TEMPLATES

(Closes #1014)
commit 6f365747 (2020-08-15): Updated ISSUE_TEMPLATES

(Closes #1014)
closed 2020-08-09 by Nightwalker-87