stlink-org/stlink #1023: [STM32F401CC]: [_$Unable to load F401 cusbom board with STM32 _]

dingyifei, 2020-09-07:
Thank you for giving feedback to the stlink project.

**NOTICE: Please read and follow instructions in #906 before submitting a ticket. This feature request will be deleted without notice when not enough information is provided! So please ensure that all fields are filled out.**

- [ x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: [STlink V3SET] (e.g Stlink /v1, /v2, /v2-clone, /v2-1)
- Operating system and version: [Windows] (e.g Linux, Mac OS X, Windows)
- **Stlink tools version** and/or git commit hash: [v1.6.1] (e.g v1.1.0/git-c722056)
- Stlink commandline tool name: [st-util] (e.g `st-info`, `st-flash`, `st-util`)
- Target chip (and board if applicable): [STM32F401CC] (e.g STM32F402VG)

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:

GDB:
```
(gdb) target extended localhost:4242
Remote debugging using localhost:4242
0x00000000 in ?? ()
(gdb) load
Loading section .text, size 0xb86c lma 0x8000000
Loading section .ARM.exidx, size 0x8 lma 0x800b86c
Loading section .data, size 0xa08 lma 0x800b874
Error finishing flash operation
(gdb) disconnect
Ending remote debugging.
(gdb) target extended localhost:4242
Remote debugging using localhost:4242
0x00000000 in ?? ()
(gdb) load
Loading section .text, size 0xb86c lma 0x8000000
Loading section .ARM.exidx, size 0x8 lma 0x800b86c
Loading section .data, size 0xa08 lma 0x800b874
Error finishing flash operation
(gdb)
```
stlink-util.exe
```
$ ./st-util.exe
st-util
2020-09-06T18:16:59 INFO usb.c: Unable to match requested speed 1800 kHz, using 1000 kHz
2020-09-06T18:16:59 INFO common.c: F4xx (low power): 64 KiB SRAM, 256 KiB flash in at least 16 KiB pages.
2020-09-06T18:16:59 INFO gdb-server.c: Listening at *:4242...
2020-09-06T18:17:10 INFO gdb-server.c: Found 0 hw breakpoint registers
2020-09-06T18:17:10 INFO gdb-server.c: GDB connected.
2020-09-06T18:17:12 INFO common.c: Attempting to write 16384 (0x4000) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Sector:0x0 Size:0x4000 2020-09-06T18:17:12 INFO common.c: Flash page at addr: 0x08000000 erased
2020-09-06T18:17:12 INFO common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
2020-09-06T18:17:12 INFO common.c: Starting Flash write for F2/F4/L4
2020-09-06T18:17:12 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 16384
2020-09-06T18:17:13 ERROR flash_loader.c: flash loader run error
2020-09-06T18:17:13 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1

```

Expected/description:

`correctly flash STM32F401CC`

Thank you for your support.

The stlink project maintainers

chenguokai, 2020-09-07:
STLink v3 is not fully supported, I would recommend using v2 or v2.1 for now.

dingyifei, 2020-09-07:
Thank you for your reply, I'll try STlink v2 see if it works.

dingyifei, 2020-09-07:
I just test it with V2, F401CC works with the STlink v2.

Nightwalker-87, 2021-04-03:
@dingyifei Please test this again with the current `develop` branch.
This should be fixed since #1025.

Nightwalker-87, 2021-04-11:
Ping @dingyifei

Nightwalker-87, 2021-04-11:
Closed by #954.

closed 2021-04-11 by Nightwalker-87