stlink-org/stlink #1024: STM32F103C8T6 using St-link v2 doesn't flash on Ubuntu (works properly on windows)

marcosQuesada, 2020-09-07:
Thank you for giving feedback to the stlink project.

**NOTICE: Please read and follow instructions in #906 before submitting a ticket. This feature request will be deleted without notice when not enough information is provided! So please ensure that all fields are filled out.**

- [ ] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: [st-link v2]
- Operating system and version: [Linux] (Ubuntu 20.04)
- **Stlink tools version** and/or git commit hash: [v1.6.1, v.1.6.0, v.1.5.1]
- Stlink commandline tool name: [st-flash]
- Target chip (and board if applicable): [STM32F103C8T6]

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:

```
st-flash --reset write /home/marcos/code/stm32/stm32f103vc-tempalte/main.bin 0x08000000

st-flash 1.6.0
2020-09-07T15:38:53 INFO common.c: Loading device parameters....
2020-09-07T15:38:53 INFO common.c: Device connected is: F1 Medium-density device, id 0x20000410
2020-09-07T15:38:53 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0 bytes (0 KiB) in pages of 1024 bytes
Unknown memory region

```
```
st-flash --format=ihex write /home/marcos/code/stm32/stm32f103vc-tempalte/main.hex

st-flash 1.6.0
2020-09-07T15:38:53 INFO common.c: Loading device parameters....
2020-09-07T15:38:53 INFO common.c: Device connected is: F1 Medium-density device, id 0x20000410
2020-09-07T15:38:53 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0 bytes (0 KiB) in pages of 1024 bytes
Unknown memory region
```

Expected/description:

 I reviewed all issues related to this "Unknown memory region", many of them point to a bad cable connection between st-link and bluepill (SCLK SDIO), but, i tested everything on windows 10 too, and i am able to flash successfully, so connection must be fine.

 Everything else seems to be well recognized, as:
```
 st-info --probe  

Found 1 stlink programmers
 serial: 371a0b002c135737334d4e00
openocd: "\x37\x1a\x0b\x00\x2c\x13\x57\x37\x33\x4d\x4e\x00"
  flash: 0 (pagesize: 1024)
   sram: 20480
 chipid: 0x0410
  descr: F1 Medium-density device

``` 
 Maybe is my fault, but i don't know what else to do, i even installed everything from scratch in a different ubuntu machine (18.04) with same results.

Thank you for your support.

The stlink project maintainers

Nightwalker-87, 2020-09-07:
I have the same board and may be able to test in this scenario soon. Please give me some time.

fivdi, 2020-09-11:
For me, everything works as expected with v1.6.1 on Ubuntu 16.04 with a bluepill:
```
$ st-info --version
v1.6.1

$ st-info --probe
Found 1 stlink programmers
 serial:     500024001800003250334d4e
 hla-serial: "\x50\x00\x24\x00\x18\x00\x00\x32\x50\x33\x4d\x4e"
 flash:      65536 (pagesize: 1024)
 sram:       20480
 chipid:     0x0410
 descr:      F1xx Medium-density

$ st-flash --version
v1.6.1

$ st-flash write blinky.bin 0x8000000
st-flash 1.6.1
2020-09-11T06:39:55 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
file blinky.bin md5 checksum: e378e365fc9787922c1d3514fb7d8d3e, stlink checksum: 0x0000c4a8
2020-09-11T06:39:55 INFO common.c: Attempting to write 688 (0x2b0) bytes to stm32 address: 134217728 (0x8000000)
2020-09-11T06:39:55 INFO common.c: Flash page at addr: 0x08000000 erased
2020-09-11T06:39:55 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2020-09-11T06:39:55 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-09-11T06:39:55 INFO flash_loader.c: Successfully loaded flash loader in sram
  1/1 pages written
2020-09-11T06:39:55 INFO common.c: Starting verification of write complete
2020-09-11T06:39:55 INFO common.c: Flash written and verified! jolly good!
```

Nightwalker-87, 2020-09-12:
That is actually what I had in mind regarding my bluepill board used on Debian Bullseye...

marcosQuesada, 2020-09-12:
I see... I just can say that i tried from scratch again, with same results (pointing to 0x8000000 and 0x0800000) After that i installed platformio, i been able to flash from there successfully, so no hardware failure is related imho.
 I will be glad to do any test if you need to.

fivdi, 2020-09-13:
What's very unusual about the information in the first post above is that a flash size of 0 bytes is reported:
```
2020-09-07T15:38:53 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0 bytes (0 KiB) in pages of 1024 bytes
```
The flash size should be reported as 65536 bytes (64 KiB).

The device ID reported is 0x0410 which is correct. The revision ID reported is 0x2000 which is valid. The revision ID reported is a little unusual in my opinion as I would expect an STM32F103C8T6 that was manufactured within the last few years to have a revision ID of 0x2003. Nevertheless, 0x2000 is valid.

The STM32F103C8T6 has a 16-bit flash size register at location 0x1ffff7e0. The content of this register should be 0x40 (64) indicating that the device has 64 KiB flash. STLink relies on the value of this register.

I'm not familiar with PlatformIO or how it flashes devices. However, you do mention that you can flash the STM32F103C8T6 with PlatformIO. This means that you could implement a program that examines the content of the flash size register to confirm whether or not it's value is 0x40. If the value is 0x40, we have made no progress. If the value is 0, it would help to explain the issue.

Nightwalker-87, 2020-09-26:
@marcosQuesada: You were using an old version of the tools previously (referring to your output). Please try it again with v1.6.1 and compare the result with @fivdi.

marcosQuesada, 2020-09-29:
Hi @Nightwalker-87 
I just followed your suggestion, and i tested on v1.61, development and master:
- v1.6.1
```
./st-info --version
v1.6.1

./st-info --probe
Found 1 stlink programmers
 serial:     371a0b002c135737334d4e00
 hla-serial: "\x37\x1a\x0b\x00\x2c\x13\x57\x37\x33\x4d\x4e\x00"
 flash:      0 (pagesize: 1024)
 sram:       20480
 chipid:     0x0410
 descr:      F1xx Medium-density

./st-flash --reset write /home/marcos/code/stm32/stm32f103vc-tempalte/main.bin 0x80000000
st-flash 1.6.1
2020-09-29T15:27:07 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 0 KiB flash in at least 1 KiB pages.
Unknown memory region

./st-flash --reset write /home/marcos/code/stm32/stm32f103vc-tempalte/main.bin 0x08000000 
st-flash 1.6.1
2020-09-29T15:27:13 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 0 KiB flash in at least 1 KiB pages.
Unknown memory region
```
- master
```
./st-info --version
v1.6.1-1-gbaab8ca

 ./st-info --probe  
Found 1 stlink programmers
 serial:     371a0b002c135737334d4e00
 hla-serial: "\x37\x1a\x0b\x00\x2c\x13\x57\x37\x33\x4d\x4e\x00"
 flash:      0 (pagesize: 1024)
 sram:       20480
 chipid:     0x0410
 descr:      F1xx Medium-density

./st-flash --reset write /home/marcos/code/stm32/stm32f103vc-tempalte/main.bin 0x80000000
st-flash 1.6.1-1-gbaab8ca
2020-09-29T15:32:09 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 0 KiB flash in at least 1 KiB pages.
Unknown memory region

./st-flash --reset write /home/marcos/code/stm32/stm32f103vc-tempalte/main.bin 0x08000000 
st-flash 1.6.1-1-gbaab8ca
2020-09-29T15:32:13 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 0 KiB flash in at least 1 KiB pages.
Unknown memory region

```
- development
```
./st-info --version
v1.6.1
 ./st-info --probe
Found 1 stlink programmers
 serial:     371a0b002c135737334d4e00
 hla-serial: "\x37\x1a\x0b\x00\x2c\x13\x57\x37\x33\x4d\x4e\x00"
 flash:      0 (pagesize: 1024)
 sram:       20480
 chipid:     0x0410
 descr:      F1xx Medium-density

./st-flash --reset write /home/marcos/code/stm32/stm32f103vc-tempalte/main.bin 0x08000000
st-flash 1.6.1
2020-09-29T15:29:28 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 0 KiB flash in at least 1 KiB pages.
Unknown memory region
./st-flash --reset write /home/marcos/code/stm32/stm32f103vc-tempalte/main.bin 0x80000000
st-flash 1.6.1
2020-09-29T15:29:33 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 0 KiB flash in at least 1 KiB pages.
Unknown memory region
```

marcosQuesada, 2020-09-29:
About PlatformIO workaround, It's a plugin for Visual Studio, and uses OpenOCD in the backgrounds to flash, those are the logs during flashing:
```
Uploading .pio/build/genericSTM32F103C8/firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

0x2ba01477
hla_swd
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000380 msp: 0x20005000
** Programming Started **
Warn : STM32 flash size failed, probe inaccurate - assuming 128k flash
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
```
Maybe that warning could help: Warn : STM32 flash size failed, probe inaccurate - assuming 128k flash

Nightwalker-87, 2020-09-29:
@marcosQuesada: Thx for the update. Indeed it looks as if your MCU comes with 128k flash which is not the standard configuration and thus not detected properly by the tools. What comes to my mind is, that you can try our `--flash=n[k][m]` option (check the tutorial) to override the standard chip memory detection upon flashing to see if it works.

As far as I can remember, the issue that causes this is that some F103C8T6 technically ship with 128k though only 64k are addressable per default (as compliant with the specs). To make use of the full memory available in this case was the intention for implementing the mentioned option. You can give it a try to see if it helps in your case.

However I see that the detection routine could see some improvements here though...

marcosQuesada, 2020-09-29:
Flashing using --flash options seems to flash properly, but fails on validation. The bluepill remains the same, so erasing has been succeed , but has not been updated. That's definitely one step in the correct direction :)
Here's the log:
```
./st-flash --flash=128k write /home/marcos/Documents/PlatformIO/Projects/tracker/.pio/build/genericSTM32F103C8/firmware.bin 0x08000000 
st-flash 1.6.1
2020-09-29T19:26:36 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 0 KiB flash in at least 1 KiB pages.
Forcing flash size: --flash=0x00020000
file /home/marcos/Documents/PlatformIO/Projects/tracker/.pio/build/genericSTM32F103C8/firmware.bin md5 checksum: dc502fec66215527ce84f1fe3414d8f, stlink checksum: 0x0053f3a9
2020-09-29T19:26:36 INFO common.c: Attempting to write 57808 (0xe1d0) bytes to stm32 address: 134217728 (0x8000000)
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08000000 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08000400 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08000800 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08000c00 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08001000 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08001400 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08001800 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08001c00 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08002000 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08002400 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08002800 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08002c00 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08003000 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08003400 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08003800 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08003c00 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08004000 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08004400 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08004800 erased
2020-09-29T19:26:36 INFO common.c: Flash page at addr: 0x08004c00 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08005000 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08005400 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08005800 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08005c00 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08006000 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08006400 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08006800 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08006c00 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08007000 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08007400 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08007800 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08007c00 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08008000 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08008400 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08008800 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08008c00 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08009000 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08009400 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08009800 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x08009c00 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800a000 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800a400 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800a800 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800ac00 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800b000 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800b400 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800b800 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800bc00 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800c000 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800c400 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800c800 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800cc00 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800d000 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800d400 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800d800 erased
2020-09-29T19:26:37 INFO common.c: Flash page at addr: 0x0800dc00 erased
2020-09-29T19:26:38 INFO common.c: Flash page at addr: 0x0800e000 erased
2020-09-29T19:26:38 INFO common.c: Finished erasing 57 pages of 1024 (0x400) bytes
2020-09-29T19:26:38 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-09-29T19:26:38 INFO flash_loader.c: Successfully loaded flash loader in sram
  3/57 pages written2020-09-29T19:26:38 ERROR flash_loader.c: flash loader run error
2020-09-29T19:26:38 ERROR common.c: stlink_flash_loader_run(0x8000c00) failed! == -1
stlink_fwrite_flash() == -1
```

Nightwalker-87, 2020-09-29:
Hmm, I wouldn't assess that writing concluded successfully:

```
020-09-29T19:26:38 INFO flash_loader.c: Successfully loaded flash loader in sram
  **3/57 pages written**2020-09-29T19:26:38 ERROR flash_loader.c: flash loader run error
2020-09-29T19:26:38 ERROR common.c: stlink_flash_loader_run(0x8000c00) failed! == -1
stlink_fwrite_flash() == -1
```
We need some improvement here as well. :confused:
I'd need a STM32F1C8T6 Bluepill (Note: Not to be confused with STM32F1C**B**T6 !) with 128k to look into this.
Mine only has the standard 64k flash.

stappersg, 2020-09-29:
On Tue, Sep 29, 2020 at 12:27:53PM -0700, nightwalker-87 wrote:
> I'd need a STM32F1C8T6 Bluepill (Note: Not to be confused with STM32F1C**B**T6 !)
> with 128k to look into this.

What is the "protocol" to send hardware to team members?

(And is it is OK to document that is a separate issue?)

Nightwalker-87, 2020-09-29:
@stappersg: Eh, we never had the case, and I don't really think it should be common practise, as we don't offer any kind of support for any commercial products. However if anybody feels himself committed to help somebody in a specific issue and arrange sharing of hardware on a personal basis, he / they should be fine with that - but not on behalf of the project.

Targeting your point it could be more helpful to gain common recognition or support from ST for this project e.g. resulting in selective free hardware-giveaways or external dev-support or whatever may appear applicable.

fivdi, 2020-09-30:
Here is what I see on a bluepill that has an STM32F103C8T6 with 128KB of flash.
```console
$ st-info --version
v1.6.1
$ st-info --probe
Found 1 stlink programmers
 serial:     500024001800003250334d4e
 hla-serial: "\x50\x00\x24\x00\x18\x00\x00\x32\x50\x33\x4d\x4e"
 flash:      131072 (pagesize: 1024)
 sram:       20480
 chipid:     0x0410
 descr:      F1xx Medium-density
$ st-flash --reset write blinky.bin 0x8000000
st-flash 1.6.1
2020-09-30T22:44:54 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 128 KiB flash in at least 1 KiB pages.
file blinky.bin md5 checksum: e378e365fc9787922c1d3514fb7d8d3e, stlink checksum: 0x0000c4a8
2020-09-30T22:44:54 INFO common.c: Attempting to write 688 (0x2b0) bytes to stm32 address: 134217728 (0x8000000)
2020-09-30T22:44:54 INFO common.c: Flash page at addr: 0x08000000 erased
2020-09-30T22:44:54 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2020-09-30T22:44:54 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-09-30T22:44:54 INFO flash_loader.c: Successfully loaded flash loader in sram
  1/1 pages written
2020-09-30T22:44:54 INFO common.c: Starting verification of write complete
2020-09-30T22:44:54 INFO common.c: Flash written and verified! jolly good!
$ 
```
The flash size register at location 0x1ffff7e0 contains the value 0x80 which corresponds to 128KB.

What's also of interest is the following text in section stm32f1x on page 100 of the [OpenOCD User’s Guide](http://openocd.org/doc/pdf/openocd.pdf),

> Note that some devices have been found that have a flash size register that contains an invalid value, to workaround this issue you can override the probed value used by the flash driver.

marcosQuesada, 2020-10-01:
Thank you @fivdi. 
Maybe it's a shot in the dark, but, to make it work under OpenCD (inside platformIO) the first try that i did failed with an error "Wrong idcode":
```
Warn : UNEXPECTED idcode: 0x2ba01477
Error: expected 1 of 1: 0x1ba01477
``` 
Digging around i found that the explanation to that error is: "Your BluePill board contains a CS32F103C8T6 chip, a Chinese clone instead of a real STM32F103C8T6." 
And the fix was to force the id code as:
```
upload_flags = -c set CPUTAPID 0x2ba01477
```

Nightwalker-87, 2020-10-01:
Ah, it's **that** old problem again - and the chip is marked as "STM32F103C8T8" instead of "CKS32F103C8T8"?

marcosQuesada, 2020-10-01:
Correct, chip is marked as STM32F103C8T8

Nightwalker-87, 2020-10-01:
It is really annoying to see that these fake-chips are still on the market.
Looking at this, we can conclude that the toolset itself is not the actual problem in this case.
However your input how to force a chip-ID is very interesting.
We should add that to our documentation to help others along that face this problem as well.

@marcosQuesada Thank you very much for your help.
This ticket will be closed automatically as soon as the related commit or PR is pushed/merged.

marcosQuesada, 2020-10-01:
@Nightwalker-87 Thank you so much for your time and even more for your patience.

Nightwalker-87, 2020-12-13:
Duplicate of #855.

commit a66557a1 (2020-12-13): [doc] STM32F103 fake chips (Closes #1024)
closed 2020-12-13 by Nightwalker-87