stlink-org/stlink #1028: NUCLEO-L452RE-P: ST-Link not found

underpickled, 2020-09-14:
Thank you for giving feedback to the stlink project.

**NOTICE: Please read and follow instructions in #906 before submitting a ticket. This feature request will be deleted without notice when not enough information is provided! So please ensure that all fields are filled out.**

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: ST LINK V2-1
- Operating system and version: MacOS 10.15.6
- v1.6.1
- Stlink commandline tool name: st-info
- Target chip (and board if applicable): STM32L452RE (NUCLEO-L452RE-P)

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:

```
st-info --probe
Found 0 stlink programmers
```

Expected/description:

I expect to see one programmer attached. Board is clearly powered over USB (blinking lights, etc).

Thank you for your support.

The stlink project maintainers

stappersg, 2020-09-14:
On Mon, Sep 14, 2020 at 04:07:29PM +0000, Dylan Jackson wrote:
> - Programmer/board type: ST LINK V2-1
> - Operating system and version: MacOS 10.15.6
> - v1.6.1
> 
> ```
> st-info --probe
> Found 0 stlink programmers
> ```

Is the USB device visible for MacOS?

If yes:  Tell us how.

Regards
Linux User
Who has `lsusb` and `lsusb -v -s N:M`

underpickled, 2020-09-14:
> On Mon, Sep 14, 2020 at 04:07:29PM +0000, Dylan Jackson wrote:
>
> Is the USB device visible for MacOS?
>
> If yes:  Tell us how.
>
>
> Regards
> Linux User
> Who has `lsusb` and `lsusb -v -s N:M`

It is not visible in MacOS that I can see. I looked in the system profiler and the USB devices all are just internal Apple devices as well as the dongle itself (the dongle works with other USB devices so I know that works).

stappersg, 2020-09-14:
On Mon, Sep 14, 2020 at 12:21:30PM -0700, Dylan Jackson wrote:
> stapperg
> > 
> > Is the USB device visible for MacOS?
> > 
> > If yes:  Tell us how.
> > 
> 
> It is not visible in MacOS that I can see. I looked in the system
> profiler and the USB devices all are just internal Apple devices as
> well as the dongle itself (the dongle works with other USB devices so
> I know that works).

I'm confused.

Retransmitting my question, in improved form.

 Is the USB device itself visible for MacOS?
 If yes: Tell us how.

Same question translated for other universum   :-)

 Can MacOS see the ST-link dongle?
 If so, what is reported by MacOS?

underpickled, 2020-09-14:
Sorry, the "dongle" I was referring to is the USB-C to USB-A dongle required for the USB-C macbook pros.
I am plugged into the USB port on the Nucleo board, which, as far as I can discern, is supposed to have ST-LINK hardware built in.

stappersg, 2020-09-14:
That is indeed what https://www.st.com/en/evaluation-tools/nucleo-l452re-p.html says.

At https://duckduckgo.com/?q=lsusb+for+macintosh is said that lsusb for
macintosh should be

   brew install lsusb

HTH

underpickled, 2020-09-14:
Thanks @stappersg, I did install `lsusb` as you suggested but unfortunately it still is not showing up.

Nightwalker-87, 2020-09-15:
@underpickled: Please paste the terminal output of `lsusb -t` below.

underpickled, 2020-09-15:
```
lsusb -t
2020-09-14 22:42:39.224 system_profiler[64929:2091569] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2020-09-14 22:42:39.224 system_profiler[64929:2091569] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2020-09-14 22:42:39.225 system_profiler[64929:2091569] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2020-09-14 22:42:39.225 system_profiler[64929:2091569] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2020-09-14 22:42:39.230 system_profiler[64929:2091569] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2020-09-14 22:42:39.231 system_profiler[64929:2091569] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
0.Dev 000: Apple T2 Bus, 
  Bus 000.Dev 001: 3.0 root hub, 5000M
/:  Bus 000.Dev 001: USB 3.1 Bus, 
/:  Bus 001.Dev 001: USB 3.1 Bus, 
    |__ Bus 001.Dev 001: USB3.1 Hub, 5Gb/s
    |__ Bus 020.Dev 001: USB2.0 Hub, 480Mb/s
        |__ Bus 020.Dev 002: USB-C Digital AV Multiport Adapter, 12Mb/s
/:  Bus 128.Dev 000: Apple T2 Controller, 
/:  Bus 128.Dev 000: FaceTime HD Camera (Built-in), 
/:  Bus 128.Dev 000: Ambient Light Sensor, 
/:  Bus 128.Dev 000: Headset, 
    |__ Bus 128.Dev 009: Apple Internal Keyboard / Trackpad, 480Mb/s
/:  Bus 128.Dev 000: Touch Bar Display, 
/:  Bus 128.Dev 000: Touch Bar Backlight, 
    |__ Bus 128.Dev 005: Composite Device, 480Mb/s
```

stappersg, 2020-09-15:
I think it is a hardware issue.

Fact:  the Operation System doesn't see the USB side of the ST-Link /
       Nucleo board

Things to try:
* Other USB cable
* Other Computer
* Other Nucleo board

Nightwalker-87, 2020-09-15:
> Additionally, the STLinkUpgrade.jar fails to launch. Running java -jar STLinkUpgrade.jar from the terminal results in [...]

This tool is not part of this project an thus none of our business.
You need to contact the Support of ST Microelectronics for this particular issue.

Nightwalker-87, 2020-10-22:
@underpickled Any update on this?

underpickled, 2020-10-22:
I gave up and just waited until my own PCBs arrived. I'm using SWD directly on that and have no issues.

Nightwalker-87, 2020-10-22:
I see, but going through this again indicates that this is likely related to faulty hardware without any relation to the `stlink` toolset. Thus I am closing this issue now as resolved.

closed 2020-10-22 by Nightwalker-87