stlink-org/stlink #1030: After update my old STM32VL1discovery not flashing microcontroller device

leha2001, 2020-09-16:
Hello friends!
I updated my Archlinux system and new version of st-util not working correctly.
My arm-none-eabi-gdb reporting about write flash operation error:
```
(gdb) load main.elf 
Loading section .text, size 0x1c6c lma 0x8000000
Loading section .rodata, size 0x34 lma 0x8001c6c
Loading section .data, size 0x8 lma 0x8001ca0
Error finishing flash operation
(gdb) 

```
st-util speaks what detecting chip correctly, but cannot interaction with it:
```
leha@dom /home/leha/work/Electronics/HPS_growbox/BrygðiOS $ st-util 
st-util
2020-09-16T15:41:45 INFO common.c: F1xx Value Line: 8 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
2020-09-16T15:41:45 INFO gdb-server.c: Listening at *:4242...
2020-09-16T15:41:46 INFO gdb-server.c: Found 6 hw breakpoint registers
2020-09-16T15:41:46 INFO gdb-server.c: GDB connected.
2020-09-16T15:41:49 INFO common.c: Attempting to write 1024 (0x400) bytes to stm32 address: 134217728 (0x8000000)
2020-09-16T15:41:49 INFO common.c: Flash page at addr: 0x08000000 erased
2020-09-16T15:41:49 INFO common.c: Finished erasing 1 pages of 1024 (0x400) bytes
2020-09-16T15:41:49 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-09-16T15:41:49 WARN flash_loader.c: Failed to write flash loader to sram!
2020-09-16T15:41:49 ERROR common.c: stlink_flash_loader_init() == -1
```

I trying to compile various version from git but I finded last working version in commit 6dfb8bafdb95e244271cf232ed2da8dbedbad08c
```
(gdb) load main.elf 
Loading section .text, size 0x1c6c lma 0x8000000
Loading section .rodata, size 0x34 lma 0x8001c6c
Loading section .data, size 0x8 lma 0x8001ca0
Start address 0x08000770, load size 7336
Transfer rate: 4 KB/sec, 2445 bytes/write.
(gdb)
```
But and one is working not stable since connection between STM32VL1discovery and microcontroller periodically disapear...

stappersg, 2020-09-16:
> I trying to compile various version from git but I finded last working version in commit [6dfb8ba](https://github.com/stlink-org/stlink/commit/6dfb8bafdb95e244271cf232ed2da8dbedbad08c)

That commit did **not change** source code lines.  It did only change `.version` and `ChangeLog.md`.

leha2001, 2020-09-16:
May be... This commit build and working, but it very old. I trying various commits with release tags or comment, many more new cannot build  because of cmake error.
```
leha@dom /home/leha/local_repo/stlink-git/src/stlink-git $ git checkout 3eab7b960ff180df5f3678def13dceb4cf7ced93
leha@dom /home/leha/local_repo/stlink-git/src/stlink-git $ cmake .
-- stlink version: v1.5.1
--           Major  Minor  Patch 
-- found USB
CMake Error at CMakeLists.txt:121 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

-- Manpage generation disabled
-- Configuring incomplete, errors occurred!
See also "/home/leha/local_repo/stlink-git/src/stlink-git/CMakeFiles/CMakeOutput.log".
See also "/home/leha/local_repo/stlink-git/src/stlink-git/CMakeFiles/CMakeError.log".
leha@dom /home/leha/local_repo/stlink-git/src/stlink-git $ 
```

stappersg, 2020-09-17:
> -- Configuring incomplete, errors occurred!
> See also "/home/leha/local_repo/stlink-git/src/stlink-git/CMakeFiles/CMakeOutput.log".
> See also "/home/leha/local_repo/stlink-git/src/stlink-git/CMakeFiles/CMakeError.log".

Succes