stlink-org/stlink #1038: [STM32F429]: Semihosting is not working after 1.5.1

jiayihu, 2020-09-23:
Thank you for giving feedback to the stlink project.

**NOTICE: Please read and follow instructions in #906 before submitting a ticket. This feature request will be deleted without notice when not enough information is provided! So please ensure that all fields are filled out.**

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: stlink-v2
- Operating system and version: MacOS
- **Stlink tools version** and/or git commit hash: 1.6.1 via brew
- Stlink commandline tool name: st-util --semihosting
- Target chip (and board if applicable): STM32F429I-DISC

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:

```
2020-09-23T23:35:30 INFO common.c: F42x/F43x: 256 KiB SRAM, 2048 KiB flash in at least 16 KiB pages.
2020-09-23T23:35:30 INFO gdb-server.c: Listening at *:4242...
2020-09-23T23:35:33 INFO gdb-server.c: Found 6 hw breakpoint registers
2020-09-23T23:35:33 INFO gdb-server.c: GDB connected.
2020-09-23T23:35:33 INFO common.c: Attempting to write 16384 (0x4000) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Sector:0x0 Size:0x4000 2020-09-23T23:35:33 INFO common.c: Flash page at addr: 0x08000000 erased
2020-09-23T23:35:33 INFO common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
2020-09-23T23:35:33 INFO common.c: Starting Flash write for F2/F4/L4
2020-09-23T23:35:33 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 16384
2020-09-23T23:35:34 INFO common.c: Starting verification of write complete
2020-09-23T23:35:34 INFO common.c: Flash written and verified! jolly good!
2020-09-23T23:35:34 INFO common.c: Attempting to write 16384 (0x4000) bytes to stm32 address: 134234112 (0x8004000)
EraseFlash - Sector:0x1 Size:0x4000 2020-09-23T23:35:34 INFO common.c: Flash page at addr: 0x08004000 erased
2020-09-23T23:35:34 INFO common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
2020-09-23T23:35:34 INFO common.c: Starting Flash write for F2/F4/L4
2020-09-23T23:35:34 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 16384
2020-09-23T23:35:34 INFO common.c: Starting verification of write complete
2020-09-23T23:35:34 INFO common.c: Flash written and verified! jolly good!
2020-09-23T23:35:34 INFO common.c: Attempting to write 16384 (0x4000) bytes to stm32 address: 134250496 (0x8008000)
EraseFlash - Sector:0x2 Size:0x4000 2020-09-23T23:35:35 INFO common.c: Flash page at addr: 0x08008000 erased
2020-09-23T23:35:35 INFO common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
2020-09-23T23:35:35 INFO common.c: Starting Flash write for F2/F4/L4
2020-09-23T23:35:35 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 16384
2020-09-23T23:35:35 INFO common.c: Starting verification of write complete
2020-09-23T23:35:35 INFO common.c: Flash written and verified! jolly good!
2020-09-23T23:35:35 INFO common.c: Attempting to write 16384 (0x4000) bytes to stm32 address: 134266880 (0x800c000)
EraseFlash - Sector:0x3 Size:0x4000 2020-09-23T23:35:35 INFO common.c: Flash page at addr: 0x0800c000 erased
2020-09-23T23:35:35 INFO common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
2020-09-23T23:35:35 INFO common.c: Starting Flash write for F2/F4/L4
2020-09-23T23:35:35 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 16384
2020-09-23T23:35:36 INFO common.c: Starting verification of write complete
2020-09-23T23:35:36 INFO common.c: Flash written and verified! jolly good!
2020-09-23T23:35:36 INFO common.c: Attempting to write 65536 (0x10000) bytes to stm32 address: 134283264 (0x8010000)
EraseFlash - Sector:0x4 Size:0x10000 2020-09-23T23:35:36 INFO common.c: Flash page at addr: 0x08010000 erased
2020-09-23T23:35:36 INFO common.c: Finished erasing 1 pages of 65536 (0x10000) bytes
2020-09-23T23:35:36 INFO common.c: Starting Flash write for F2/F4/L4
2020-09-23T23:35:36 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 32768
size: 32768
2020-09-23T23:35:37 INFO common.c: Starting verification of write complete
2020-09-23T23:35:38 INFO common.c: Flash written and verified! jolly good!
```

Expected/description:

I have been debugging today using semihosting but it was not working despite using an application which I was sure it was fine some months ago. Then I tried to compile version 1.5.1 which I was using and `st-util --semihosting` works there, whereas with the new versions results in SIGTRAP

Thank you for your support.

The stlink project maintainers

Ant-ON, 2020-10-25:
I think it regression is fixed by #1027. 
@jiayihu May you try latest develop branch?

kokkonisd, 2020-12-29:
I have the same experience with an STM32F103RB and an STM32L452RE. Fetched latest develop via homebrew (`brew install stlink --HEAD`), and the issue persists.

Nightwalker-87, 2020-12-29:
@jiayihu to be more precisely: Please test commit ba5c6797f3bae57cb79be961250ce4a28dfc2675 and the previous commit 5c03678ab02f0305dae80978952dc96b321a9834 to verify this.

jiayihu, 2021-01-04:
Sorry guys, I'm not using semihosting anymore so I can't test it again master for the time being. Maybe someone else can do it, there was a deleted comment by someone else some days ago having the same issue.

Nightwalker-87, 2021-03-09:
@Ant-ON Can you or anybody else test instead?
I have never used this feature, so I'm not sure if I can give any useful assessment here.

Nightwalker-87, 2021-04-03:
@jiayihu Even though you don't use `--semihosting` any more: Can you please provide the commands you executed in `gdb` and tell us more about the actual issue to be able to have it investigated? This ticket is now open and thus should not be closed without a solution.

jiayihu, 2021-04-03:
I didn't use gdb, I just ran `st-util --semihosting`. Btw, if nobody else can test the fix just give me some time. I'm writing my master thesis right now, but after the deadline on the 15th of this month, I'll try to find a moment to test it on my board

Nightwalker-87, 2021-04-03:
@jiayihu: Thx for coming back to this. Note that `st-util --semihosting` alone does not do anything apart from opening a debug server listening and waiting for interaction with a `gdb` debugger. (As far as I am aware of.) Also check our tutorial (`/doc/tutorial.md`) on this.
I believe it is important to understand the behaviour you expected in contrast to the actual implementation. Maybe there was some confusion, please let us know. Shall there be any need for further testing later on, the 15th should be fine - we'll see.

jiayihu, 2021-04-03:
Oh you're right, I went back to my workflow and I used this `.gdbinit`

```
file "./obj/gee"

target extended-remote :4242

load
break main

continue
```

And ran with `arm-none-eabi-gdb`

Nightwalker-87, 2021-04-03:
Ok, thanks - I'll try this out with my Bluepill board and possibly with a borrowed Nucleo STM32F429ZI (if functional... :confused:).

Nightwalker-87, 2021-04-03:
Here is my output when trying to reproduce your command sequence with a Bluepill (CKS32F103C8T6):

```
$ gdb-multiarch
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".

(gdb) target extended-remote :4242
Remote debugging using :4242
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x00000000 in ?? ()
(gdb) file /[...]/Blinker/Release/Blinker.elf
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from /[...]/Blinker/Release/Blinker.elf...
(No debugging symbols found in /[...]/Blinker/Release/Blinker.elf)
(gdb) load
Loading section .isr_vector, size 0x130 lma 0x8000000
Loading section .text, size 0xbac lma 0x8000130
Loading section .rodata, size 0x74 lma 0x8000cdc
Loading section .init_array, size 0x4 lma 0x8000d50
Loading section .fini_array, size 0x4 lma 0x8000d54
Loading section .data, size 0x68 lma 0x8000d58
Start address 0x080002b4, load size 3520
Transfer rate: 13 KB/sec, 586 bytes/write.
(gdb) break main
Breakpoint 1 at 0x8000234
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) continue
Continuing.

Breakpoint 1, 0x08000234 in main ()
(gdb) 
```

... which looks good to me and is what to expect AFAIK.

Nightwalker-87, 2021-04-03:
Here is my output when trying to reproduce your command sequence on develop (HEAD) with a Nucleo STM32F429ZI:

```
$ gdb-multiarch
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) file /home/nightwalker-87/Schreibtisch/Blinker/Release/Blinker.elf
Reading symbols from /home/nightwalker-87/Schreibtisch/Blinker/Release/Blinker.elf...
(No debugging symbols found in /home/nightwalker-87/Schreibtisch/Blinker/Release/Blinker.elf)
(gdb) target extended-remote localhost:4242
Remote debugging using localhost:4242
0x08000234 in main ()
(gdb) load
Loading section .isr_vector, size 0x130 lma 0x8000000
Loading section .text, size 0xbac lma 0x8000130
Loading section .rodata, size 0x74 lma 0x8000cdc
Loading section .init_array, size 0x4 lma 0x8000d50
Loading section .fini_array, size 0x4 lma 0x8000d54
Loading section .data, size 0x68 lma 0x8000d58
```

... and nothing else happened in `gdb` while receiving continuous output of

```
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
[!] send_recv send request failed: LIBUSB_ERROR_NO_DEVICE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
[!] send_recv send request failed: LIBUSB_ERROR_NO_DEVICE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
[!] send_recv send request failed: LIBUSB_ERROR_NO_DEVICE
```

on the gdb-server (`st-util --semihosting`) side.

This led me to the approach to update the firmware of the STLINK/V2-1 on-board programmer to **V2.J37.M27**.

This leads to:

```
$ gdb-multiarch
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) file /[...]/Blinker/Release/Blinker.elf
Reading symbols from /[...]/Blinker/Release/Blinker.elf...
(No debugging symbols found in /[...]/Blinker/Release/Blinker.elf)
(gdb) target extended-remote localhost:4242
Remote debugging using localhost:4242
0x00000000 in ?? ()
(gdb) load
Loading section .isr_vector, size 0x130 lma 0x8000000
Loading section .text, size 0xbac lma 0x8000130
Loading section .rodata, size 0x74 lma 0x8000cdc
Loading section .init_array, size 0x4 lma 0x8000d50
Loading section .fini_array, size 0x4 lma 0x8000d54
Loading section .data, size 0x68 lma 0x8000d58
Start address 0x080002b4, load size 3520
Transfer rate: 5 KB/sec, 586 bytes/write.
(gdb) break main
Breakpoint 1 at 0x8000234
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) continue
Continuing.

Breakpoint 1, 0x08000234 in main ()
(gdb) 
```

... which again looks good to me and is what to expect AFAIK.

Nightwalker-87, 2021-04-03:
What I can still do is to verify the observed results with v1.6.0 and v1.6.1 in order to find out about any possible inconsistencies in the codebase between v1.5.1 and v1.6.1.

Nightwalker-87, 2021-04-04:
Here is my output when trying to reproduce your command sequence on `develop` (HEAD) with a Bluepill (CKS32F103C8T6):

```
$ gdb-multiarch
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) file /[...]/Blinker/Release/Blinker.elf
Reading symbols from /[...]/Blinker/Release/Blinker.elf...
(No debugging symbols found in /[...]/Blinker/Release/Blinker.elf)
(gdb) target extended-remote localhost:4242
Remote debugging using localhost:4242
0x08000bb4 in _write_r ()
(gdb) load
Loading section .isr_vector, size 0x130 lma 0x8000000
Loading section .text, size 0xbac lma 0x8000130
Loading section .rodata, size 0x74 lma 0x8000cdc
Loading section .init_array, size 0x4 lma 0x8000d50
Loading section .fini_array, size 0x4 lma 0x8000d54
Loading section .data, size 0x68 lma 0x8000d58
Start address 0x080002b4, load size 3520
Transfer rate: 15 KB/sec, 586 bytes/write.
(gdb) break main
Breakpoint 1 at 0x8000234
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) continue
Continuing. <-- runs through previously set breakpoint, instead of 
^C <-- Abort with CTRL+C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000bb4 in _write_r ()
(gdb)
```

... which does not work as expected.

Nightwalker-87, 2021-04-04:
Here is my output when trying to reproduce your command sequence for v1.6.1 with a Nucleo STM32F429ZI:

$ gdb-multiarch
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) file /[...]/Blinker/Release/Blinker.elf
Reading symbols from /[...]/Blinker/Release/Blinker.elf...
(No debugging symbols found in /[...]/Blinker/Release/Blinker.elf)
(gdb) target extended-remote localhost:4242
Remote debugging using localhost:4242
0x00000000 in ?? ()
(gdb) load
Loading section .isr_vector, size 0x130 lma 0x8000000
Loading section .text, size 0xbac lma 0x8000130
Loading section .rodata, size 0x74 lma 0x8000cdc
Loading section .init_array, size 0x4 lma 0x8000d50
Loading section .fini_array, size 0x4 lma 0x8000d54
Loading section .data, size 0x68 lma 0x8000d58
Start address 0x080002b4, load size 3520
Transfer rate: 5 KB/sec, 586 bytes/write.
(gdb) break main
Breakpoint 1 at 0x8000234
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) continue
Continuing. <-- runs through previously set breakpoint, instead of halting...
^C <-- abort with CTRL+C, to interrupt endless loop...
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000bb4 in _write_r ()
(gdb) 

... which also does not work as expected.

Nightwalker-87, 2021-04-04:
... so from my findings I can verify that `st-util --semihosting` (still) does not work correctly in v1.6.1, but on current `develop`.

Ant-ON, 2021-04-05:
@Nightwalker-87 Yes, this bug is related to #1027. Semihost uses the `stlink_read_all_regs` for determine a state:
https://github.com/stlink-org/stlink/blob/fe526ec9ebc77224ed3ac0e688e07cd250cd59a0/src/st-util/gdb-server.c#L1458-L1469

Nightwalker-87, 2021-04-05:
Fixed in #1027.

closed 2021-04-05 by Nightwalker-87