stlink-org/stlink #1041: cannot use st-flash  with coreid: 2ba01477, chipid: 414

satfan52, 2020-09-26:
Hello,

I am not a regular user of stlink, so please excuse the naivity of the question. 

I have received a prototype board and an stlink-v2 dongle. I need to flash the prototype board with a specific bootloader using st-flash. I am using Ubuntu 18.04.

I have compiled  the nightly build and the development version of stlink without issues hoping that one of the two versions would allow me to do the job with the command "st-flash binary.bin address"

Unfortunately, it does not work, the flash memory on the board is erased but the bootloader is unable to load (see console extract below). I had a search on the list of issues and my hint is that it is probably required to modify one of the files and recompile st-link so my prototype board with coreid: 2ba01477, chipid: 414 is supported, but I have no idea what needs to be updated/patched. If someone could guide me into this, it would very much appreciated.

Thanks
Regards
Peter

peter@on7yi-ubuntu:/opt/stlink/bin$ ./st-flash write /home/peter/Downloads/binary.bin 0x08000000
st-flash 1.6.1-127-g0a99197
2020-09-26T13:17:19 INFO common.c: F1xx High-density: 64 KiB SRAM, 256 KiB flash in at least 2 KiB pages.
file /home/peter/Downloads/binary.bin md5 checksum: c4166e78c6fb60737954837d291b4491, stlink checksum: 0x002c8660
2020-09-26T13:17:19 INFO common.c: Attempting to write 27576 (0x6bb8) bytes to stm32 address: 134217728 (0x8000000)
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08000000 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08000800 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08001000 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08001800 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08002000 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08002800 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08003000 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08003800 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08004000 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08004800 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08005000 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08005800 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08006000 erased
2020-09-26T13:17:19 INFO common.c: Flash page at addr: 0x08006800 erased
2020-09-26T13:17:19 INFO common.c: Finished erasing 14 pages of 2048 (0x800) bytes
2020-09-26T13:17:19 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-09-26T13:17:19 ERROR flash_loader.c: unknown coreid, not sure what flash loader to use, aborting! coreid: 2ba01477, chipid: 414
2020-09-26T13:17:19 WARN flash_loader.c: Failed to write flash loader to sram!
2020-09-26T13:17:19 ERROR common.c: stlink_flash_loader_init() == -1
stlink_fwrite_flash() == -1
peter@on7yi-ubuntu:/opt/stlink/bin$

Ant-ON, 2020-09-26:
Try to apply the patch:

```diff
--- a/src/stlink-lib/flash_loader.c
+++ b/src/stlink-lib/flash_loader.c
@@ -248,6 +248,12 @@ int stlink_flash_loader_write_to_sram(stlink_t *sl, stm32_addr_t* addr, size_t*
         loader_size = sizeof(loader_code_stm32l);
     } else if (sl->core_id == STM32VL_CORE_ID ||
                sl->chip_id == STLINK_CHIPID_STM32_F1_MEDIUM ||
+               sl->chip_id == STLINK_CHIPID_STM32_F1_HIGH ||
+               sl->chip_id == STLINK_CHIPID_STM32_F1_LOW ||
+               sl->chip_id == STLINK_CHIPID_STM32_F1_VL_MEDIUM_LOW ||
+               sl->chip_id == STLINK_CHIPID_STM32_F1_VL_HIGH ||
+               sl->chip_id == STLINK_CHIPID_STM32_F1_XL ||
+               sl->chip_id == STLINK_CHIPID_STM32_F1_CONN ||
                sl->chip_id == STLINK_CHIPID_STM32_F3 ||
                sl->chip_id == STLINK_CHIPID_STM32_F3_SMALL ||
                sl->chip_id == STLINK_CHIPID_STM32_F303_HIGH ||
```

satfan52, 2020-09-26:
Great !!! It worked ! Many Thanks !!!!!!
**" 2020-09-26T14:01:39 INFO common.c: Flash written and verified! jolly good! "**

closed 2020-10-02 by Nightwalker-87