stlink-org/stlink #1045: [STM32F100RB]: [Failed to read core_id and unknown device]

WK-M, 2020-10-05:
Thank you for giving feedback to the stlink project.

**NOTICE: Please read and follow instructions in #906 before submitting a ticket. This feature request will be deleted without notice when not enough information is provided! So please ensure that all fields are filled out.**

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: Onboard STLINK/V1 
- Operating system and version: Fedora 32
- **Stlink tools version** and/or git commit hash: v1.6.1
- Stlink commandline tool name: st-info and st-util
- Target chip (and board if applicable): STM32F100RBT6

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:

For **st-util**:
```
st-util -1
st-util: invalid option -- '1'
st-util
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_GET_VERSION
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_ENTER
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_READCOREID
2020-10-05T03:31:27 ERROR common.c: Failed to read core_id
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
2020-10-05T03:31:27 WARN common.c: Invalid flash type, please check device declaration
2020-10-05T03:31:27 ERROR gdb-server.c: Unsupported Target (Chip ID is 0000000000, Core ID is 0000000000).
```

For **st-info**:

```
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_GET_VERSION
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_ENTER
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_READCOREID
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
Found 1 stlink programmers
 serial:     303030303030303030303031
 hla-serial: "\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x31"
 flash:      0 (pagesize: 0)
 sram:       0
 chipid:     0x0000
 descr:      unknown device
```

Expected/description:

The device should not be read as an unknown device when executing `st-info --probe`, and there should not be a gdb-server error when running `st-util`. However, the board is able to be detected when running lsusb.

```
Bus 001 Device 013: ID 0483:3744 STMicroelectronics ST-LINK/V1
```

I'm still relatively new to STM32 and ARM, please let me know if there's any other details that I can provide.

Thank you for your support.

The stlink project maintainers

Nightwalker-87, 2020-10-05:
Thx for reporting. I can verify this with my STM32VL Discovery board that comes with this chip.

corecell, 2020-10-23:
Have the same problem.
Checked with version from repository `community/stlink 1.6.1-1` and latest develop branch.
Board - **STM32VLDISCOVERY**
```
Linux i5 5.4.61-rt37-MANJARO #1 SMP PREEMPT_RT Sat Aug 29 17:09:34 UTC 2020 x86_64 GNU/Linux
core/libusb 1.0.23-2
extra/libusb-compat 0.1.7-1
```

Latest working commit: e3aa887f7ffdfcfb3a0529ee3b94a7d3fe39b5d8

Ant-ON, 2020-10-25:
How I wrote in stlinkV3 issue it may break v1 support:
https://github.com/stlink-org/stlink/issues/820#issuecomment-617591438

@WK-M Can you try patch to develop branch:
```
index 9726ddc..a683267 100644
--- a/src/stlink-lib/usb.c
+++ b/src/stlink-lib/usb.c
@@ -261,7 +261,10 @@ int _stlink_usb_write_debug32(stlink_t *sl, uint32_t addr, uint32_t data) {
 }
 
 int _stlink_usb_get_rw_status(stlink_t *sl) {
-    if (sl->version.jtag_api == STLINK_JTAG_API_V1) { return(-1); }
+    if (sl->version.jtag_api == STLINK_JTAG_API_V1) { 
+        /* STLINK does not support getting statuses. Returns the IO operation is success */
+        return 0;
+    }
 
     unsigned char* const rdata = sl->q_buf;
     struct stlink_libusb * const slu = sl->backend_data;
@@ -300,7 +303,7 @@ int _stlink_usb_write_mem32(stlink_t *sl, uint32_t addr, uint16_t len) {
 
     if (ret == -1) { return(ret); }
 
-    ret = send_only(slu, 0, data, len);
+    ret = send_only(slu, 1, data, len);
 
     if (ret == -1) { return(ret); }
 
```

WK-M, 2020-10-25:
> How I wrote in stlinkV3 issue: it patch may break v1 support ( [#820 (comment)](https://github.com/stlink-org/stlink/issues/820#issuecomment-617591438) )
> 
> @WK-M Can you try patch to develop branch:
> 
> ```
> index 9726ddc..a683267 100644
> --- a/src/stlink-lib/usb.c
> +++ b/src/stlink-lib/usb.c
> @@ -261,7 +261,10 @@ int _stlink_usb_write_debug32(stlink_t *sl, uint32_t addr, uint32_t data) {
>  }
>  
>  int _stlink_usb_get_rw_status(stlink_t *sl) {
> -    if (sl->version.jtag_api == STLINK_JTAG_API_V1) { return(-1); }
> +    if (sl->version.jtag_api == STLINK_JTAG_API_V1) { 
> +        /* STLINK does not support getting statuses. Returns the IO operation is success */
> +        return 0;
> +    }
>  
>      unsigned char* const rdata = sl->q_buf;
>      struct stlink_libusb * const slu = sl->backend_data;
> @@ -300,7 +303,7 @@ int _stlink_usb_write_mem32(stlink_t *sl, uint32_t addr, uint16_t len) {
>  
>      if (ret == -1) { return(ret); }
>  
> -    ret = send_only(slu, 0, data, len);
> +    ret = send_only(slu, 1, data, len);
>  
>      if (ret == -1) { return(ret); }
>  
> ```

When you state "patch to develop branch", did you want me to checkout to the develop branch and attempt the same commands? If that is not what you meant, please let me know.

Otherwise, I have performed the checkout to the develop branch and performed the proper reinstallation. I am still running into the same errors mentioned in the original posting.

Ant-ON, 2020-10-26:
@WK-M I apply patch to new branch https://github.com/Ant-ON/stlink/tree/try_fix_stlinkv1. Try clone try_fix_stlinkv1 branch and build stlink tool from source.

WK-M, 2020-10-27:
Thanks for the clarification. I cloned the patch, and I'm still running into the same issues.

I'm not sure if this is a Fedora-specific issue as I am not running into the same error on my Ubuntu machine.

Nightwalker-87, 2020-12-12:
@Vascom Any idea regarding Fedora?

Vascom, 2020-12-12:
F32 use libusbx-1.0.23.
Which version at ubuntu?

WK-M, 2020-12-18:
Sorry for the late response.

From the command: `dpkg -l libusb-1.0*`, I got the following:

```
+++-======================-================-============-===================================================
ii  libusb-1.0-0:amd64     2:1.0.23-2build1 amd64        userspace USB programming library
ii  libusb-1.0-0-dev:amd64 2:1.0.23-2build1 amd64        userspace USB programming library development files

```

mosagepa, 2021-01-16:
Is this being tackled, any advance?
I didn’t see the id for this on announce for 1.6.2 still, yet of course thanks to all the efforts by developers.
I myself have same issue, on VL Discovery, I’m on the road but will post my details later on.
Regards

mosagepa, 2021-01-17:
I update.
Everything is working now for my VL Discovery.
I just had to rm -fr all libstlink* under /usr/local/lib,
Then proceed to recompile stlink from 1.6.0 tree I had in backup.
Maybe this can shed some light also:
I had noticed previously that invoking ‘stlink-download /dev/stlinkv1-1 info’ just after connecting the VL to the USB port, and then invoking ‘stlink-gui’ the Stlink-gui let me query the device and yielded the Value Line id and chipid without any problem, and then afterwards also ‘st-info —probe’ gave out the right values, but then any debug session in gdb didn’t get to load into SRAM the flash routine to erase, and it failed...
Also, if I proceed to launch ‘st-info -probe’ right after plugging the USB,it didn’t work, I always had to launch ‘stlink-download’ (even if it failed itself) for both ‘stlink-gui’ and ‘st-info —probe’ to yield the right ID values... quite strange, it seems as if the stlink-download executable makes some change in VL that gets it right (mostly) afterwards.
This experiment could maybe give some hint to the developers about what could be the issue underneath... I hope so!
Anyway, as I said the fix for me was as I mentioned, to wipe old/conflicting libs and recompile BUT use 1.6.0.
Btw, the aforementioned ‘stlink-download’ is to be found under:
https://github.com/vanbwodonk/arm-utilities/blob/master/stlink-download/stlink-download.c
(This site seems to maintain also a v2 consolidated version. Any of those two is clarifying -see comments within the C code- as of the deeply broken nature of the ST-Link/V1 scsi-based coding...)

Nightwalker-87, 2021-01-17:
Libusb-0.1.x is not used and thus not needed by this toolset.
@mosagepa This looks way to complicated to serve as a common solution.

I have a STM32F100RBT6 (VL Discovery) with STLINK-V1 as well, but unfortunately had no time to run any tests yet besides professional work...

mosagepa, 2021-01-18:
Hi, thanks for your comments.

The solution I've found to satisfy my needs, the stlink-download.c rigmarole I mentioned only just in case it gave you a hint.

I guess I can compare/ track/ debug the actual calls, parameters and execution of the current 1.6.x against the "bad" libraries and the 1.6.0 versions, couldn't be due to differing interfaces, I'm almost sure this has to be some quirk we've not yet found.

If I make some advance on this of course I'll let people now here.

mosagepa, 2021-01-25:
New update:
I've noticed that, even with the libraries "purged" as I commented above, first call to 'st-info --probe' yields out a significant (and different but consistent for any one of my 3 VL Discoveries) serial number, just after plugging them to USB port, BUT as soon as you relaunch 'st-info --probe' the model is correctly detected still as Value Line F1xxx blah blah... BUT, yields now 0x30 0x30 0x30..... i.e. all zeroes chain as "serial".
So maybe this gives me a clue to debug what is really happening here, hopefully I can trace the "bug" (or problem) down to its origin following the inner calls in the 'st-info' execution.
I am being stubborn in this just because I'd like the VL Discovery not to get into forever oblivion but to end up having proper "support" under Linux, so I guess any help in this regard can be appreciated by a bunch of people using VLs if yet a minority I know...

mosagepa, 2021-01-25:
I wish to add (and yeah I know this is not a Forum but anyway...) I had once a friend which works designing firmwares and products for automotive products with NXP chips, objecting to my dwelling and obfuscations over "End-Of-Life" products, his reasoning was, the development boards and kits have the lifespan of flies, serving only their purpose while the HW guys are figuring out the final board but the SW/FW guys should be developing the support code anyway even with the prototypes.
So when a new "line" is launched by the chip/uC manufacturers, they (the companies manufacturing real stuff) obligue to buy the development kits in order to be quicker than their competitors launching their new-line-based final product.
Meaning, any hobbyist using (or resurrecting, as I tend to do) older deprecated platforms were making fools of themselves.
I can't agree with that reasoning, and that's why I am so stubborn with this and many other abandonware HW/FW stuff. But I digress...

GadgetAngel, 2021-01-29:
I am getting the same problem with my STM32VLDISCOVERY board.  ST-LINK V1 is detected by Linux Mint 20.1 but st-util does not work or st-info does not read the chip info correctly.

mosagepa, 2021-02-02:
Exactly same approach I used, @GadgetAngel  Congratulations!
I am trying to get a bit of hold of other work occupations in order to track down what exactly are the differences between 1.6.0 and 1.6.1.

Please can you confirm if you face this also (I copy here the fragment of previous post):

_- "I've noticed that, even with the libraries "purged" as I commented above, first call to 'st-info --probe' yields out a significant (and different but consistent for any one of my 3 VL Discoveries) serial number, just after plugging them to USB port, BUT as soon as you relaunch 'st-info --probe' the model is correctly detected still as Value Line F1xxx blah blah... BUT, yields now 0x30 0x30 0x30..... i.e. all zeroes chain as "serial"._

_

igor-m, 2021-02-27:
Hi guys! Thanks for your effort!!
As I've found a couple of the VL boards in my junkbox I've tried with 1.6.1 and I've ran into the same issues as described above - with latest Lubuntu 20.04. (Btw the 1.6.1 flashes with the Discovery F407 board, but not reliable enough - sometimes it flashes ok, sometimes not..)
I've done the 1.6.0 install as described above by GadgetAngel and the flashing of my VL board now works.

GadgetAngel, 2021-02-27:
> Exactly same approach I used, @GadgetAngel Congratulations!
> I am trying to get a bit of hold of other work occupations in order to track down what exactly are the differences between 1.6.0 and [...]

I honestly cannot remember what the output was when I had the error.  

What I do know is the V1.6.0 is more reliable for me and I can use it on many different boards {VL discovery, F4 discovery, NUCLEO board}

Nightwalker-87, 2021-03-08:
As I was able to verify with the same board, this regression is still present in the current `develop` branch.

igor-m, 2021-03-09:
Btw I still get "Found 0 stlink programmers", even it flashes the binary into F100 ok..

igor-m, 2021-03-09:
me@lub:~$ st-info
st-info --version
st-info --flash
st-info --sram
st-info --descr
st-info --pagesize
st-info --chipid
st-info --serial
st-info --hla-serial
st-info --probe
me@lub:~$ st-info --flash
0x20000
me@lub:~$ st-info --version
v1.6.0
me@lub:~$ st-info --chipid
0x0420
me@lub:~$ st-info --hla-serial
""
me@lub:~$ st-info --descr
F1 Medium/Low-density Value Line device
me@lub:~$ st-info --flash
0x20000
me@lub:~$ st-info --probe
Found 0 stlink programmers
me@lub:~$

mosagepa, 2021-03-09:
Woah, what a change.
I was contributing to this very same issue, along with @GadgetAngel, but somehow @Nightwalker-87 didn't allow me to post more follow-ups to this.
I know the exact cure for this issue BUT I don't think people harassing other people's advances (even if it means going back to v1.6.0) would help anyone.
As these follow-ups show, there's enough demand for ST-Link v1 in the VL Discoveries to merit investigation.
Too sad the manners here are such that they need such evidence to fix things up even WITH people wanting to contribute.
All I know I got my *three* VL Discoveries working nicely right now (yeah, I got two more of those... come to think of it, when people got fed up of trying to fix the damn thingies, they suddenly dropped in price, I can't complain about that :)

Nightwalker-87, 2021-03-09:
aha, interesting, let me have a look at this message... That's not intended.

Nightwalker-87, 2021-03-09:
@mosagepa: Please feel free to share your findings. In fact we need a fix for the current `develop` tree. Going back in time is not a satisfiable solution, apart from locating fixes from the past.

mosagepa, 2021-03-10:
> @mosagepa: Please feel free to share your findings. In fact we need a fix for the current `develop` tree. Going back in time is not a satisfiable solution, apart from locating fixes from the past.

That's ok, I was just in a bad mood yesterday I guess. I don't know what the limitation was about, but it's now removed. Sorry for the lengthy posts, I know this is not a forum.

To focus on available solutions, I don't know if this could be fixed in v1.6.1 onward since I didn't write the code so I don't know, what could possibly have gone "wrong"?

If you read my previous posts there were an alternative command line utility in some other site, now only reachable by Wayback Machine. Within the comments in such code the guy was making clear the ST-Link v1 implementation was "deeply broken", since it relied on emulating SCSI interfaces to a "drive" (same approach used in most recent ST Discoverys e.g. F40x series, but in ST Link v1 the "faked SCSI device" was used also for communicating with the firmware, commands and all that. I can't get why on earth did they do this (ST engineers)".

It must be clearly some interaction with such "device mode", as it was the v1 series only which did this, new implementations/ interfaces of libusb+some quirk/s on v1.6.1+ are not totally "retrocompatible"...

mosagepa, 2021-03-10:
This is also to let you know I've written a bit of mixed up code which if you execute will let all you test whether your VL Discovery can be detected (fully, means: meaningful serial and all that).
I encourage all folks having problems with their VL's to please copy up the code, and I myself compiled it with this precise command line:
`g++ -o tryMosa02 tryMosa02.cpp -I /usr/local/include/libusb-1.0 -L/usr/local/lib -lusb-1.0`

For reference, my libs are as follows (your command line and mileage may vary, but that's one of our targets, to identify which configurations work and which don't...):
```
lrwxrwxrwx 1 root root     19 abr 26  2020 /usr/local/lib/libusb-0.1.so.4 -> libusb-0.1.so.4.4.4
-rwxr-xr-x 1 root root 110472 abr 26  2020 /usr/local/lib/libusb-0.1.so.4.4.4
-rw-r--r-- 1 root root 851222 feb  3 15:46 /usr/local/lib/libusb-1.0.a
-rwxr-xr-x 1 root root    980 feb  3 15:46 /usr/local/lib/libusb-1.0.la
lrwxrwxrwx 1 root root     19 feb  3 15:46 /usr/local/lib/libusb-1.0.so -> libusb-1.0.so.0.2.0
lrwxrwxrwx 1 root root     19 feb  3 15:46 /usr/local/lib/libusb-1.0.so.0 -> libusb-1.0.so.0.2.0
-rwxr-xr-x 1 root root 502144 feb  3 15:46 /usr/local/lib/libusb-1.0.so.0.2.0
-rwxr-xr-x 1 root root    795 abr 26  2020 /usr/local/lib/libusb.la.backup
lrwxrwxrwx 1 root root     21 abr 26  2020 /usr/local/lib/libusbpp-0.1.so.4 -> libusbpp-0.1.so.4.4.4
-rwxr-xr-x 1 root root 201664 abr 26  2020 /usr/local/lib/libusbpp-0.1.so.4.4.4
-rw-r--r-- 1 root root 385884 abr 26  2020 /usr/local/lib/libusbpp.a
-rwxr-xr-x 1 root root    834 abr 26  2020 /usr/local/lib/libusbpp.la
lrwxrwxrwx 1 root root     21 abr 26  2020 /usr/local/lib/libusbpp.so -> libusbpp-0.1.so.4.4.4
lrwxrwxrwx 1 root root     19 abr 26  2020 /usr/local/lib/libusb.so -> libusb-0.1.so.4.4.4

```
[tryMosa02.cpp.gz](https://github.com/stlink-org/stlink/files/6114710/tryMosa02.cpp.gz)

Nightwalker-87, 2021-03-10:
I can give you the console output for e.g. b47f083736a6f6137e317e70998501205d44265f :

```
st-info --probe
[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_READCOREID
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
Found 1 stlink programmers
  version:    V1J13
  serial:     553f6d06483f48534641213f
  hla-serial: "\x55\x3f\x6d\x06\x48\x3f\x48\x53\x46\x41\x21\x3f"
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0000
  descr:      unknown device
```

```
st-flash write /[...]/Blinker.bin 0x8000000
st-flash 1.6.1-135-gb47f083
[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_READCOREID
2021-03-10T09:32:04 ERROR common.c: Failed to read core_id
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
2021-03-10T09:32:04 WARN common.c: Invalid flash type, please check device declaration
Failed to connect to target
```

... and the device locks up with the COM LED switching to the on-state, so I assume this is related to `libusb` device handling.

Looking at the output, I find the device serial not to be the problem at this point, but as we have found out in #356, some general problems regarding codebase-compatibility seem to have silently occurred between `master` and `develop` without notice (e.g. build errors that could be detected by our CI service). We need to bring both branches to the same common base before trying to solve anything. Maybe it's about time for a new release including all other fixes from `develop` well knowing there are still bugs like this one to be solved, but we can include this in the release note and ask everybody only to work with that new release as a common base for investigation. Any other ideas on that?

mosagepa, 2021-03-10:
My current "idea" is that using plain simple libusb calls I can go on this success forever:
`mosagepa@MProLinux:~/decomp/STM32/oldLinuxSTLink/withSTLink$ st-info --probe
Found 1 stlink programmers
 serial:     393f6f064d42343816512543
 hla-serial: "\x39\x3f\x6f\x06\x4d\x42\x34\x38\x16\x51\x25\x43"
 flash:      131072 (pagesize: 1024)
 sram:       8192
 chipid:     0x0420
 descr:      F1xx Value Line
mosagepa@MProLinux:~/decomp/STM32/oldLinuxSTLink/withSTLink$ st-info --probe
Found 1 stlink programmers
 serial:     303030303030303030303031
 hla-serial: "\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x31"
 flash:      131072 (pagesize: 1024)
 sram:       8192
 chipid:     0x0420
 descr:      F1xx Value Line
mosagepa@MProLinux:~/decomp/STM32/oldLinuxSTLink/withSTLink$ 
`
and so on and on...

I appreciate your codebase maintenance workforce, but I guess something is happening in the translation so to speak between your encapsulations and the pure libusb calls. I though that libusb was the culprit, but now I guess the detail is hidden somewhere, yeah, but might be easier fixed than located :)

mosagepa, 2021-03-10:
> Found 1 stlink programmers
>   version:    V1J13
>   serial:     553f6d06483f48534641213f

Launch it a second time, see where the
`Found 1 stlink programmers
  version:    V1J13
  serial:     553f6d06483f48534641213f`
goes...

mosagepa, 2021-03-10:
You say "Looking at the output, I find the device serial not to be the problem at this point, "

But I insist on the serial issue just because it's the hint that something is not going right in the libusb calls within stlink's encapsulation.

mosagepa, 2021-03-10:
Please try to re-code st-link, especifically the --probe switch in the code. I have a feeling the problem comes with the specific sequence of libusb calls made (and maybe already made before?) within where the user calls with that argument before doing anything else with st-link.
You never know the side effects to some detection/ probing double calls to libusb. Just my two cents worth (I don't know what the code does at a deep level, just talking from memory some days ago in Febraury where I was trying to diagnose it...)

Nightwalker-87, 2021-03-10:
@mosagepa: "at this point" solely referred to the used commit. I have not systematically tested the other (especially earlier versions). I'd consider it a good idea if people who coded in this part more recently have a look at this. Quite a few parts in the codebase don't have sufficient documentation which of course does not facilitate understanding of what is happening here exactly.

mosagepa, 2021-03-10:
Please ask them to have a look at indexes accessed during probing.
If you get the indexes wrong, it refers to the wrong USB sub-device which is basically the broken USB storage part which Linux users couldn't care less about (for the VL case). So any libusb done after that goes bad too.

Nightwalker-87, 2021-03-10:
Ping @grevaillot @rewolff @petertorelli

Ant-ON, 2021-03-10:
@mosagepa @Nightwalker-87 Can you try get log after change line? 

https://github.com/stlink-org/stlink/blob/8429ac078a783ee6d75db6bd8741eba732959c54/src/st-info/info.c#L88
to
```
    sl = stlink_v1_open(DEBUG_LOG_LEVEL, 1);
```

Nightwalker-87, 2021-03-10:
error: ‘DEBUG_LOG_LEVEL’ undeclared (first use in this function)
   97 |   sl = stlink_v1_open(DEBUG_LOG_LEVEL, 1); :-1:

Ant-ON, 2021-03-10:
@Nightwalker-87 use 100 instead DEBUG_LOG_LEVEL

Nightwalker-87, 2021-03-10:
It compiles, but produces same output as posted earlier.
We should better not try and error here - one needs to understand what is happening and where to look at exactly.

I'll have to put this aside for now to deal with some other issues as long as I have the chance to fix stuff and do some maintenance for a few days. I certainly do what I can, but can't help the fact that there aren't enough developers around that address advanced tasks regularly. :disappointed:

mosagepa, 2021-03-10:
It must be LIBUSB_LOG_LEVEL_INFO maybe? (suggested to me by the compiler).

Anyway, on first execution from just connected VL I get:
`[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_READCOREID
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
Found 1 stlink programmers
  version:    V1J13
  serial:     393f6f064d42343816512543
  hla-serial: "\x39\x3f\x6f\x06\x4d\x42\x34\x38\x16\x51\x25\x43"
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0000
  descr:      unknown device
`

and that's surely the "good" serial BUT you see the chipid zero'd and something goes wrong, the lights on VL get both RED leds on  (but it's apparently still executing my previously flashed blinky on the GREEN one),
but on second and further executions afterwards not even the serial is given out right, it gets the wrong "interface", "device" or whatever personality which is the STM32 Mass Storage broken "index", with serial 00000...1:
`[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_GET_VERSION
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_ENTER
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_READCOREID
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
Found 1 stlink programmers
  version:    V1
  serial:     303030303030303030303031
  hla-serial: "\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x31"
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0000
  descr:      unknown device`

I've got myself a good TRACE of a lot of previously and after error calls to libusb from the st-link code, and I'll gladly share the details since this is giving me the feeling about the index problem. It's just that I am not fully able to explain it in technical terms, it's just that, my tracing experiments...

mosagepa, 2021-03-10:
What about this "home-made" trace I got?

**BIEN (GOOD RESULT, ON FIRST EXECUTION):
=====**
`DEBUG: I am going to call libusb_get_device_descriptor with IDs = 00001D6B:00000002...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 00001D6B:00000002...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 00001D6B:00000001...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 000005AC:00008206...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 000005AC:0000020C...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 000005AC:00001003...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 00001D6B:00000001...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 0000045E:000000CB...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 00001D6B:00000001...\
DEBUG : libusb_get_string_descriptor_ascii() with sizeof(serial) = 64... done.
  desc.iSerialNumber = 0\
  serial = \
done.\
  desc.iSerialNumber = 1\
  serial = STMicroelectronics\
done.\
  desc.iSerialNumber = 2\
  **serial = STM32 STLink**\
done.\
  desc.iSerialNumber = 3\
  **serial = 9?oMB48Q%C\**`

**MAL (BAD SUBSEQUENT RESULTS, SEE THE DIFFERING TAGS AND STRINGS ON APPARENTLY SAME CALLS):
====**
`DEBUG: I am going to call libusb_get_device_descriptor with IDs = 00001D6B:00000002...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 00001D6B:00000002...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 00001D6B:00000001...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 000005AC:00008206...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 000005AC:0000020C...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 000005AC:00001003...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 00001D6B:00000001...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 0000045E:000000CB...\
DEBUG: I am going to call libusb_get_device_descriptor with IDs = 00001D6B:00000001...\
DEBUG : libusb_get_string_descriptor_ascii() with sizeof(serial) = 64... done.\
  desc.iSerialNumber = 0\
  serial = \
done.\
  desc.iSerialNumber = 1\
  serial = STMicroelectronics\
done.\
  desc.iSerialNumber = 2\
  **serial = STM32 Mass Storage**\
done.\
  desc.iSerialNumber = 3\
  **serial = 000000000001**\
done.`

mosagepa, 2021-03-10:
Please btw tell me how can I insert CR+LFs within pasted code in order to edit my recent posts and make them more readable... ;)

Nightwalker-87, 2021-03-10:
Usually one can simply copy console output as is and paste it and use < > afterwards to define a codeblock formatting around it.
However, note that there also is www.pastebin.com for complete console outputs.

mosagepa, 2021-03-10:
```
[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_READCOREID
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
Found 1 stlink programmers
  version:    V1J13
  serial:     393f6f064d42343816512543
  hla-serial: "\x39\x3f\x6f\x06\x4d\x42\x34\x38\x16\x51\x25\x43"
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0000
  descr:      unknown device

```

petertorelli, 2021-03-10:
@Nightwalker-87 
pong: Unfortunately I do not have any L100 series boards, so I cannot reproduce the issue.

GadgetAngel, 2021-03-10:
> @petertorelli: Ok, but can you review the current state of STLINK/V1 device handling with libusb? I saw that you dealt with such problems before in the codebase. We have at least two people here with the right hardware to assist with testing and can provide feedback - this shouldn't be an issue.

I have both Mint Linux (20.1) machine and Windows 10 machine to do testing on the following boards:
1) STM32VLDISCOVERY (STM32F100RB) using ST-LINK-V1
2) STM32F4DISCOVERY (STM32F407VG) using ST-LINK/V2-A
2) NUCLEO-F103RB (STM32F103RB) using ST-LINK-V2-1
3) NUCLEO-F446RE (STM32F446RE) using ST-LINK/V2-1
4) NUCLEO-L433RC-P (STM32L433RC-P) using ST-LINK/V2-1
5) NUCLEO-F302R8 (STM32F302R8) using ST-LINK/V2-1
6) NUCLEO-F767ZI (STM32F767ZI) using ST-LINK/V2-1

I also own an ST-LINK-V2 device and ST-LINK-V3 device and various ST-LINK-V2 clones

just let me know what you want me to test.

Ant-ON, 2021-03-11:
@Nightwalker-87 I looked again at the changes when stlink V3 support was added. Here are the fixes for the bugs I found. Can you check work stlink v1 after it is applied?

```patch
@@ -301,7 +301,7 @@ int _stlink_usb_write_mem32(stlink_t *sl, uint32_t addr, uint16_t len) {

     if (ret == -1) { return(ret); }

-    ret = send_only(slu, 0, data, len);
+    ret = send_only(slu, 1, data, len);

     if (ret == -1) { return(ret); }

@@ -431,6 +431,8 @@ int _stlink_usb_status(stlink_t * sl) {
         } else {
             sl->core_stat = TARGET_UNKNOWN;
         }
+    } else {
+        sl->core_stat = TARGET_UNKNOWN;
     }

     return(0);
@@ -469,19 +471,14 @@ int _stlink_usb_enter_swd_mode(stlink_t * sl) {
     unsigned char* const cmd  = sl->c_buf;
     ssize_t size;
     unsigned char* const data = sl->q_buf;
-    const uint32_t rep_len = sl->version.stlink_v > 1 ? 2 : 0;
+    const uint32_t rep_len = sl->version.jtag_api == STLINK_JTAG_API_V1 ? 0 : 2;
     int i = fill_command(sl, SG_DXFER_FROM_DEV, rep_len);

     cmd[i++] = STLINK_DEBUG_COMMAND;
     // select correct API-Version for entering SWD mode: V1 API (0x20) or V2 API (0x30).
     cmd[i++] = sl->version.jtag_api == STLINK_JTAG_API_V1 ? STLINK_DEBUG_APIV1_ENTER : STLINK_DEBUG_APIV2_ENTER;
     cmd[i++] = STLINK_DEBUG_ENTER_SWD;
-
-    if (rep_len == 0) {
-        size = send_only(slu, 1, cmd, slu->cmd_len);
-    } else {
-        size = send_recv(slu, 1, cmd, slu->cmd_len, data, rep_len);
-    }
+    size = send_recv(slu, 1, cmd, slu->cmd_len, data, rep_len);

     if (size == -1) {
         printf("[!] send_recv STLINK_DEBUG_ENTER\n");
```

rewolff, 2021-03-11:
On Wed, Mar 10, 2021 at 08:26:19AM -0800, mosagepa wrote:
> It must be LIBUSB_LOG_LEVEL_INFO maybe? (suggested to me by the compiler).
>
> Anyway, on first execution from just connected VL I get:
> `[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
> [!] send_recv [...]

This really looks as if it was able to enumerate that the STLINK
device existes and then.... fails to communicate with the STLINK
completely.

This looks as if you might have a permission problem on the USB
device. Try running the command as root, and if you get further,
please install the correct udev rules to be able to use the device as
a normal user.

-- REW

mosagepa, 2021-03-11:
@rewolff My udev settings are ok and as I said above in another post a compilation of libusb calls mixed code which asks also for the serial works beautifully and consistenly with the ST Link v1 in the VL Discovery, so I am pretty sure privileges is not the problem. Also, @Nightwalker-87 had the same log messages and he surely doesn't have a privilege issue neither.

As I pointed out before, the problem seems to be in something that goes wrong just after an initial enquery by st-link, just in the part in which I copied out my own debug info, notice that first time it gets the correct (sub)device, which is the ST-Link programmer "device", at second and further times the same index passed gets to the dreaded "storage device" whose serial is the "0000...00001".

So the first "pass" takes the VL Discovery into some wrong "state" so to speak which also causes both red leds to light. From then on any access gets directed to the "storage device" which we know it's broken by design and we don't use for anything in Linux case, it addresses that "part" of the device when it should address the "ST Link" part of it, so to speak.

Ant-ON, 2021-03-11:
@mosagepa See the third part of my patch https://github.com/stlink-org/stlink/issues/1045#issuecomment-796475351. I think that the process of receiving information stops with the error because the application does not read the response from the STLINK_DEBUG_ENTER_SWD command.

mosagepa, 2021-03-11:
@Ant-ON Please, I highly recommend that when you have a spare time you [check this out](https://github.com/stlink-org/stlink/issues/1045#issuecomment-795143885). I think that proper comparison of the libusb calls that an `st-link --probe` command issues correlating them with my code's could give us some insight. As you can see in my code and related post, it works like a charm. So why does `st-link --probe` amount to the very same sequence of libusb calls (you could trace the flow and correlate both sequences just to pinpoint the moment when `st-link --probe` goes divergent!)

Nightwalker-87, 2021-03-11:
@Ant-ON: I can test your code with my STLink/V1 now, after merging `develop` into `master` to have a common codebase again.

I know by now that the merge itself does not make any difference to the behaviour of my device, **but** we can now be sure that there will be no more code ingress from old parts because we'll only accept new PRs that start from this commit or later. Hence I ask everybody to work with latest `develop` based on this latest commit also for any further testing and/or fixes. Thanks in advance. :wink:

Nightwalker-87, 2021-03-11:
@Ant-ON: Yes, that seems to be the key:

```
$ sudo make uninstall && sudo make clean && hash -r && sudo make install && st-info --probe
[...]
Found 1 stlink programmers
  version:    V1J13
  serial:     553f6d06483f48534641213f
  hla-serial: "\x55\x3f\x6d\x06\x48\x3f\x48\x53\x46\x41\x21\x3f"
  flash:      131072 (pagesize: 1024)
  sram:       8192
  chipid:     0x0420
  descr:      F1xx Value Line
```

Great work! Thx. :tada: 

I find both red LEDs to remain in a lit up state, but assume that this is an equal behaviour to the red LED turning blue on my STLink/V2-clone programmer when testing with my bluepill-board for other open issues.

I'll now continue to test other commands with it.
However this is a first to solve this topic here. :+1:

Nightwalker-87, 2021-03-11:
General test with st-flash:

```
$ st-flash --debug --reset --area=option read
2021-03-11T18:01:39 DEBUG common.c: *** looking up stlink version
2021-03-11T18:01:39 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-03-11T18:01:39 DEBUG common.c: stlink pid     = 0x3744
2021-03-11T18:01:39 DEBUG common.c: stlink version = 0x1
2021-03-11T18:01:39 DEBUG common.c: jtag version   = 0xd
2021-03-11T18:01:39 DEBUG common.c: swim version   = 0x0
2021-03-11T18:01:39 DEBUG common.c:     notice: the firmware doesn't support a swim interface
2021-03-11T18:01:39 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-03-11T18:01:39 DEBUG usb.c: JTAG/SWD freq set to 0
2021-03-11T18:01:39 DEBUG common.c: *** set_swdclk ***
2021-03-11T18:01:39 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-03-11T18:01:39 DEBUG common.c: *** stlink_reset ***
2021-03-11T18:01:39 WARN usb.c: NRST is not connected
2021-03-11T18:01:39 DEBUG usb.c: Using reset through SYSRESETREQ
2021-03-11T18:01:39 DEBUG common.c: *** stlink_soft_reset ***
2021-03-11T18:01:39 DEBUG common.c: *** stlink_write_debug32 0xa05f0003 to 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_write_debug32 0x01000500 to 0xe000edfc
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x03030003 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_write_debug32 0x0000001f to 0xe000ed30
2021-03-11T18:01:39 DEBUG common.c: Loading device parameters....
2021-03-11T18:01:39 DEBUG common.c: *** stlink_core_id ***
2021-03-11T18:01:39 DEBUG common.c: core_id = 0x1ba01477
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x10016420 at 0xe0042000
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0xffff0080 at 0x1ffff7e0
2021-03-11T18:01:39 INFO common.c: F1xx Value Line: 8 KiB SRAM, 128 KiB flash in at least 1 KiB pages.
2021-03-11T18:01:39 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-03-11T18:01:39 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-03-11T18:01:39 DEBUG common.c: *** stlink_reset ***
2021-03-11T18:01:39 WARN usb.c: NRST is not connected
2021-03-11T18:01:39 DEBUG usb.c: Using reset through SYSRESETREQ
2021-03-11T18:01:39 DEBUG common.c: *** stlink_soft_reset ***
2021-03-11T18:01:39 DEBUG common.c: *** stlink_write_debug32 0xa05f0003 to 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_write_debug32 0x01000500 to 0xe000edfc
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x03030003 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-03-11T18:01:39 DEBUG common.c: *** stlink_write_debug32 0x0000001f to 0xe000ed30
2021-03-11T18:01:39 DEBUG common.c: *** stlink_force_debug_mode ***
2021-03-11T18:01:39 DEBUG common.c: *** stlink_status ***
2021-03-11T18:01:39 DEBUG usb.c: core status: 00030003
2021-03-11T18:01:39 DEBUG common.c:   core status: halted
2021-03-11T18:01:39 DEBUG flash.c: @@@@ Read 0 (0) option bytes from          0
[... 3 lines cut ...]
2021-03-11T18:01:39 DEBUG common.c: *** stlink_close ***
```

Flashing:

```
$ sudo make uninstall && sudo make clean && sudo make install && st-flash write /.../Blinker.bin 0x8000000
st-flash 1.6.1-216-gcc9bedd-dirty
2021-03-11T18:04:07 WARN usb.c: NRST is not connected
[... cut ...]
2021-03-11T18:04:07 INFO common.c: Flash page at addr: 0x08000c00 erased
2021-03-11T18:04:07 INFO common.c: Finished erasing 4 pages of 1024 (0x400) bytes
2021-03-11T18:04:07 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-11T18:04:07 INFO flash_loader.c: Successfully loaded flash loader in sram
  4/  4 pages written
2021-03-11T18:04:07 INFO common.c: Starting verification of write complete
2021-03-11T18:04:07 INFO common.c: Flash written and verified! jolly good!
```

However I find that no LED is flashing... :disappointed:

igor-m, 2021-03-11:
I've built the latest on Ubuntu 18.04 running in latest VM_Virtualbox (git clone, make clean, make release, make install). I had to do sudo ldconfig in addition as I got error with missing libusb.. something.
I get
```
[!] send_recv read reply failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_READCOREID
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
Found 1 stlink programmers
  version:    V1J13
  serial:     553f6f06483f48536145153f
  hla-serial: "\x55\x3f\x6f\x06\x48\x3f\x48\x53\x61\x45\x15\x3f"
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0000
  descr:      unknown device
```

Ant-ON, 2021-03-12:
@igor-m Try apply patch to develop branch from https://github.com/stlink-org/stlink/issues/1045#issuecomment-796475351
@Nightwalker-87 The firmware is being flashed with verification. In theory, it should be work normally. You can read the firmware back and compare them if in doubt about correct operation.

igor-m, 2021-03-12:
With applied patch (the patch lines at 431 are not valid anymore, it seems):
```
~/st-flash/stlink$ st-info --probe
Found 1 stlink programmers
  version:    V1J13
  serial:     553f6f06483f48536145153f
  hla-serial: "\x55\x3f\x6f\x06\x48\x3f\x48\x53\x61\x45\x15\x3f"
  flash:      131072 (pagesize: 1024)
  sram:       8192
  chipid:     0x0420
  descr:      F1xx Value Line

~/st-flash/stlink$ sudo st-flash write /tmp/arduino_build_760260/Fade.ino.bin 0x08000000
st-flash 1.6.1-216-gcc9bedd-dirty
2021-03-12T08:14:35 WARN usb.c: NRST is not connected
2021-03-12T08:14:35 INFO common.c: F1xx Value Line: 8 KiB SRAM, 128 KiB flash in at least 1 KiB pages.
file /tmp/arduino_build_760260/Fade.ino.bin md5 checksum: e9b51d95f4bb16984758f44491e4c237, stlink checksum: 0x0013fb42
2021-03-12T08:14:35 INFO common.c: Attempting to write 13708 (0x358c) bytes to stm32 address: 134217728 (0x8000000)
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08000000 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08000400 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08000800 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08000c00 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08001000 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08001400 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08001800 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08001c00 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08002000 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08002400 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08002800 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08002c00 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08003000 erased
2021-03-12T08:14:35 INFO common.c: Flash page at addr: 0x08003400 erased
2021-03-12T08:14:35 INFO common.c: Finished erasing 14 pages of 1024 (0x400) bytes
2021-03-12T08:14:35 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-12T08:14:35 INFO flash_loader.c: Successfully loaded flash loader in sram
 14/ 14 pages written
2021-03-12T08:14:36 INFO common.c: Starting verification of write complete
2021-03-12T08:14:36 INFO common.c: Flash written and verified! jolly good!
~/st-flash/stlink$ 
```
..and the LED fades 👍

igor-m, 2021-03-12:
@Nightwalker-87: the leds on the VL board are called LED_BLUE and LED_GREEN - double-check in your arduino Blink source..

Nightwalker-87, 2021-03-12:
On my STM32VL Discovery Board the status LEDs in the front part are called LD1 and LD2 and both are red ones.
Further there are LD3 and LD4 at the other end of the board, but I've not seen those lit up yet.

igor-m, 2021-03-12:
LD3 and LD4 are the leds you may let blink with the Blink demo (arduino with stm32 support installed). The leds are called LED_BLUE and LED_GREEN inside arduino IDE then. The default LED in the Blink sketch is named "9" thus you will not see the led blinking..

PR #1104 (2021-03-14): Fixed support for STLINK/V1 programmer
Correction of the regression that occurred while implementing the STLINK/V3 support.

(Fixes  #1045)
closed 2021-03-14 by Nightwalker-87