stlink-org/stlink #1047: [Nucleo Stm32F767zi]: WARN common.c: unknown chip id! 0x5fa0004

chris1seto, 2020-10-10:
- Programmer/board type: [Nucleo Stm32f767 unmodified, right out of the box, tried 2 devices] 
- Operating system and version: [Ubuntu 20.04.1 LTS] (e.g Linux, Mac OS X, Windows)
- **Stlink tools version** and/or git commit hash: [e3a2e7bc5c25ef7fbf1b0130efba746f8f24ff45
 fresh clone from git, st-flash 1.6.1-100-ge3a2e7b] 
- Stlink commandline tool name: [st-flash --format ihex write file.hex] 
- Target chip (and board if applicable): [STM32F767zi]

Command: st-flash --format ihex write file.hex
Output: 
st-flash 1.6.1-100-ge3a2e7b
2020-10-10T11:51:02 WARN common.c: unknown chip id! 0x5fa0004
Failed to connect to target

I know both nucleos are good. I regularly program them with STLinkV2 tool and Stm32CubeProgrammer

Any ideas? Thanks!

chris1seto, 2020-10-10:
Oops. Dupe of https://github.com/stlink-org/stlink/issues/1040

Turns out the stlink tool doesn't download firmware from the net, it uses whatever it was bundled with. I was many versions behind!

Nightwalker-87, 2020-10-10:
@chris1seto don't close tickets even if they are a duplicate.
They will be closed automatically, if the related issue has been finally solved.

Nightwalker-87, 2020-10-21:
Reviewing of this issue leads to the conclusion that a firmware update has been applied with the official ST Link FW Updater:
https://www.st.com/en/development-tools/stsw-link007.html
This is also the procedure we propose when running into issues with this toolset before opening a new issue.
As we do not provide any support for firmware updates in this toolset nor the ability to update the latter,
one should contact the ST Microelectronics Support for related issues.

closed 2020-10-21 by Nightwalker-87

chris1seto, 2020-10-21:
@Nightwalker-87 yes, sorry. I had run an update but I had an old version of the updater. I assumed it went out onto the internet to get the update and that I had the latest, but the stlink binaries are embedded in the app. Fwiw, 39 was the latest version which worked