stlink-org/stlink #1048: Understand the change an executed `cargo flash` introduces in reference to this stlink toolset.

stappersg, 2020-10-10:
_Originally posted by @Nightwalker-87 in https://github.com/stlink-org/stlink/issues/1007#issuecomment-704571402_

> However it would be helpful to understand the change an executed `cargo flash` introduces in reference to this stlink toolset.

Right now I see no other way to help with this, then creating a fresh issue for it.

stappersg, 2020-10-10:
In an attempt to add value.

`cargo` is the build tool for _rust_.    (Something like `maven` for _java_)

`cargo flash` uses source code provided by the project  https://probe.rs

The _stlink_ parts of _probe-rs_ are in https://github.com/probe-rs/probe-rs/tree/master/probe-rs/src/probe/stlink

stappersg, 2021-02-12:
Closing the issue that was opened four months ago by me.  Because I think the project is beter of with this issue being closed.

Nightwalker-87, 2021-02-12:
Please leave closing of issues to automation and/or administration within this project.

Nightwalker-87, 2021-03-09:
Failed contribution - Issue has been closed without anything to complete.

closed 2021-03-09 by Nightwalker-87