stlink-org/stlink #1050: STM32 STLink 1.6.1-1 won't install

doctek, 2020-10-14:
Thank you for giving feedback to the stlink project.

**NOTICE: Please read and follow instructions in #906 before submitting a ticket. This feature request will be deleted without notice when not enough information is provided! So please ensure that all fields are filled out.**

- [ x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: STLinkv2.1 on Nucleo-L053R8
- Operating system and version: Ubuntu 16.04? LTS
- **Stlink tools version** and/or git commit hash: v1.6.1-1 from stlink-1.6.1-1_amd64.deb
- Stlink commandline tool name: sudo apt install ./stlink-1.6.1-1_amd64.deb
- Target chip (and board if applicable): N/A

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output and OUTPUT/ERROR of the commandline tool(s)
Trying to install STlink from https://github.com/stlink-org/stlink. Downloaded stlink-1.6.1-1_amd64.deb and tried to install with

sudo apt install ./stlink-1.6.1-1_amd64.deb

from the directory the deb is in.

Here is the error that resulted.: (lots deleted that went just fine.)

    The following NEW packages will be installed:
    stlink
    0 upgraded, 1 newly installed, 0 to remove and 258 not upgraded.
    8 not fully installed or removed.
    Need to get 0 B/165 kB of archives.
    After this operation, 846 kB of additional disk space will be used.
    Get:1 /home/jim/Downloads/stlink-1.6.1-1_amd64.deb stlink amd64 1.6.1-1 [165 kB]
    (Reading database ... 515962 files and directories currently installed.)
    Preparing to unpack .../stlink-1.6.1-1_amd64.deb ...
    Unpacking stlink (1.6.1-1) ...
    dpkg: error processing archive /home/jim/Downloads/stlink-1.6.1-1_amd64.deb (--unpack):
    trying to overwrite '/etc/udev/rules.d/49-stlinkv2-1.rules', which is also in package st-stlink-udev-rules 1.0.0
    dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
    Errors were encountered while processing:
    /home/jim/Downloads/stlink-1.6.1-1_amd64.deb
    E: Sub-process /usr/bin/dpkg returned an error code (1)

Needless to say, the offending dev rule is NOT in /etc/udev/rules.d.

    ls /etc/udev/rules.d
    25-vc500.rules 49-node.rules 70-snap.core.rules
    49-arduino.rules 49-stlinkv2.rules 70-snap.obs-studio.rules
    49-cameo.rules 49-stlinkv3.rules 80-canon_mfp2.rules
    49-ftdi.rules 49-teensy.rules 99-jlink.rules
    49-megaCNC.rules 51-android.rules
    49-mega.rules 70-persistent-net.rules 

I'm hung out to dry until this is resolved. Any help much appreciated.

Expected/description:

I expected an install of stlink.

Thank you for your support.

ps: I hope I have provided the information you need. I tried to find any similar bug and could not find such. I am sorry to bother you with this, but I can't get around it and I really want to use your version of the stlink tool.

The stlink project maintainers

Nightwalker-87, 2020-10-17:
What is package `st-stlink-udev-rules 1.0.0`? Where does that come from?

doctek, 2020-10-17:
Are you asking me? I hope not, cause I have no idea.

I did solve my immediate problem of needing your stlink package by 
building from source following your instructions for Linux. Worked 
perfectly and package all runs.

                  -jim

Nightwalker-87, 2020-10-17:
@doctek: Good that you found a working solution. However we schould find out if there is a problem with the package.
Was there a previous version installed or any package that derived from the ubuntu/debian-repo?

doctek, 2020-10-18:
I don't recall a previous installation, but I may not be a reliable 
witness! I could do apt list --installed and pipe it to a file if you 
would like to see it.

Nightwalker-87, 2020-10-18:
Yes, that would be great, thx. I recommend to use https://pastebin.com/ for this.
It would be just fine to limit the paste expiration to 1 week - there is no need to preserve it for any longer.

doctek, 2020-10-18:
It's all in here:

https://pastebin.com/xhKzg9iH

Nightwalker-87, 2020-10-18:
You problem lies here:

```
st-stlink-server/now 1.2.0-3 amd64 [installed,local]
st-stlink-udev-rules/now 1.0.0 all [installed,local]
st-stm32cubeide-1.0.2/now 1.0.2-3566-20190716-0927-1 amd64 [installed,local]
```
You should at least uninstall the second package, better would be to wipe all three and retry.
Does the second package depend on `st-stm32cubeide` ?

doctek, 2020-10-19:
I'm not sure how to proceed. Since I have successfully installed your 
stlink package from source, should I try to install the .deb version 
also? What if I remove the stm32cube package - but I want to use it? I'm 
sorry for not really knowing what I should do here. Can you give me a 
bit more info?

Thanks!

         -jim

Nightwalker-87, 2020-10-19:
You should run

```
sudo apt purge st-stlink-server
sudo apt purge st-stlink-udev-rules
sudo apt purge st-stm32cubeide-1.0.2
```
and then retry to install our package following your original approach.
I assume installation will proceed then.
Afterwards you can try to reinstall stm32cube. Should there be any problems with this, you wil not be able to use both simultaneously, I guess.

doctek, 2020-10-20:
OK. I did the instructions as you suggested. Then I tried the deb 
package install and it went with no errors! I think that's good, but I 
don't know how to tell if the stlink tools from the deb are now running 
or if I'm still running the ones from the install from source that I did.

Can you tell me how to quickly check? Then I'll know that the deb can be 
successfully installed.

Thank you for all the help. I seem to have at least one set of stlink 
tools running properly and that's what I needed.

                -jim

Nightwalker-87, 2020-10-20:
Run `sudo apt list *stlink*` to see what is still there from the Debian software repo related to the toolset.
Check the list and run the folloing command for every found package:
`sudo apt purge <name_of_found_stlink-package>`
Afterwards you can be quite sure that there are no remnants left and try to install from source again by following your original approach. I expect the installation to succeed under this condition.

stappersg, 2020-10-20:
> Trying to install STlink from https://github.com/stlink-org/stlink. Downloaded stlink-1.6.1-1_amd64.deb and tried to install with
>
> sudo apt install ./stlink-1.6.1-1_amd64.deb
>
> from the directory the deb is in.

:-)

Use
```bash
sudo apt install stlink
```
To install `stlink` from an _apt archive_.

Use
```bash
sudo dpkg --install ./stlink-1.6.1-1_amd64.deb
```
to install a downloaded or a self build  `stlink-1.6.1-1_amd64.deb`.

Nightwalker-87, 2020-10-20:
@stappersg: I think he is aware of that, let him do the proposed clean-up first. :wink:

doctek, 2020-10-21:
Running the first command (with stlink in quotes, not stars), I got the 
response:

stlink/now 1.6.1-1 amd64 [installed,local]

Looks good to me! All the stlink tools work and my man files still exist.

So I've verified that the install from source works and that the .deb 
package can be installed also. I think I'm done! Nightwalker-87, I thank 
you for your patience and your expert assistance. Keep up the great work 
on this project!

                     -jim

stappersg, 2020-10-21:
Here the person that came late to the party
and did only add "when apt and when dpkg"

On Tue, Oct 20, 2020 at 05:08:05PM -0700, Jim Larson wrote:
> Looks good to me! All the stlink tools work and my man files still exist.
>
> So I've verified that the install from source works and that the .deb
> package can be installed also. I think I'm done! [...]

Yes, the challenge of
how to express the great value of libre software.

Try closing this issue

Nightwalker-87, 2020-10-21:
@doctek Good to hear that! Thanks for the feedback. :+1: 
As this topic has found a good solution, I am closing this issue now.

closed 2020-10-21 by Nightwalker-87

doctek, 2020-10-21:
Again, thanks for your excellent assistance with this. And thank you for 
closing it.