stlink-org/stlink #1054: [STM32 device name]: [_$YourTitle_]

joseph-cb, 2020-10-21:
Thank you for giving feedback to the stlink project.

**NOTICE: Please read and follow instructions in #906 before submitting a ticket. This feature request will be deleted without notice when not enough information is provided! So please ensure that all fields are filled out.**

- [ ] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: [ST-LINK/V2.1] (e.g Stlink /v1, /v2, /v2-clone, /v2-1)
- Operating system and version: [Linux] (e.g Linux, Mac OS X, Windows)
- **Stlink tools version** and/or git commit hash: [baab8ca Updated ISSUE_TEMPLATES] (e.g v1.1.0/git-c722056)
- Stlink commandline tool name: [st-flash] (e.g `st-info`, `st-flash`, `st-util`)
- Target chip (and board if applicable): [L452RE] (e.g STM32F402VG)

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:
I run into the error shown in [#981 ](https://github.com/stlink-org/stlink/issues/981) I follow the solution by increasing WAIT_ROUNDS to extremely large value.

```
-#define WAIT_ROUNDS 100
+//#define WAIT_ROUNDS 100
+#define WAIT_ROUNDS 10000
```
Now, I find that flashing will work one time, fail next, work again, fail again. i.e. alternate between succeeding and failing. This strange behavior is very consistent.

The following screenshot captures one success followed by a failure followed by another success. Same linux binary and same target STM32 binary were used in these three invocations.

Furthermore, I use `./st-util` and a remote gdb to flash the elf file produced by the same code. `st-util`+ remote gdb can consistently flash my binary reliably. The binaries (elf and bin) were compiled using STM32CubeIDE 1.4.0. I also tested Release build and Debug build. Same strange behavior is observed.

What will be the next step for debugging `st-flash`? I can use `st-util` in the meantime, but we will need `st-flash` when we hit production.

```
pi@pi-0c:~/ws/stlink/build/Release/bin $ ./st-flash write ~/ws/uC/release_STM32T_UFBGA.bin 0x08000000
st-flash 1.6.1-1-gbaab8ca-dirty
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/020: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/022: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
2020-10-21T02:47:49 INFO common.c: L45x/46x: 128 KiB SRAM, 512 KiB flash in at least 2 KiB pages.
file /home/pi/ws/uC/release_STM32T_UFBGA.bin md5 checksum: 13a5891bde141a2a46b35d3da95bfbe6, stlink checksum: 0x00343a39
2020-10-21T02:47:49 INFO common.c: Attempting to write 32784 (0x8010) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Page:0x0 Size:0x800 2020-10-21T02:47:49 INFO common.c: Flash page at addr: 0x08000000 erased
EraseFlash - Page:0x1 Size:0x800 2020-10-21T02:47:49 INFO common.c: Flash page at addr: 0x08000800 erased
EraseFlash - Page:0x2 Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08001000 erased
EraseFlash - Page:0x3 Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08001800 erased
EraseFlash - Page:0x4 Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08002000 erased
EraseFlash - Page:0x5 Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08002800 erased
EraseFlash - Page:0x6 Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08003000 erased
EraseFlash - Page:0x7 Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08003800 erased
EraseFlash - Page:0x8 Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08004000 erased
EraseFlash - Page:0x9 Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08004800 erased
EraseFlash - Page:0xa Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08005000 erased
EraseFlash - Page:0xb Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08005800 erased
EraseFlash - Page:0xc Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08006000 erased
EraseFlash - Page:0xd Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08006800 erased
EraseFlash - Page:0xe Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08007000 erased
EraseFlash - Page:0xf Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08007800 erased
EraseFlash - Page:0x10 Size:0x800 2020-10-21T02:47:50 INFO common.c: Flash page at addr: 0x08008000 erased
2020-10-21T02:47:50 INFO common.c: Finished erasing 17 pages of 2048 (0x800) bytes
2020-10-21T02:47:50 INFO common.c: Starting Flash write for F2/F4/L4
2020-10-21T02:47:50 INFO flash_loader.c: Successfully loaded flash loader in sram
size: 32768
size: 16
2020-10-21T02:47:51 INFO common.c: Starting verification of write complete
2020-10-21T02:47:51 INFO common.c: Flash written and verified! jolly good!
pi@pi-0c:~/ws/stlink/build/Release/bin $ ./st-flash write ~/ws/uC/release_STM32T_UFBGA.bin 0x08000000
st-flash 1.6.1-1-gbaab8ca-dirty
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/058: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[... 50 lines cut ...]
EraseFlash - Page:0xa Size:0x800 2020-10-21T02:48:13 INFO common.c: Flash page at addr: 0x08005000 erased
EraseFlash - Page:0xb Size:0x800 2020-10-21T02:48:13 INFO common.c: Flash page at addr: 0x08005800 erased
EraseFlash - Page:0xc Size:0x800 2020-10-21T02:48:13 INFO common.c: Flash page at addr: 0x08006000 erased
EraseFlash - Page:0xd Size:0x800 2020-10-21T02:48:13 INFO common.c: Flash page at addr: 0x08006800 erased
EraseFlash - Page:0xe Size:0x800 2020-10-21T02:48:13 INFO common.c: Flash page at addr: 0x08007000 erased
EraseFlash - Page:0xf Size:0x800 2020-10-21T02:48:13 INFO common.c: Flash page at addr: 0x08007800 erased
EraseFlash - Page:0x10 Size:0x800 2020-10-21T02:48:13 INFO common.c: Flash page at addr: 0x08008000 erased
2020-10-21T02:48:13 INFO common.c: Finished erasing 17 pages of 2048 (0x800) bytes
2020-10-21T02:48:13 INFO common.c: Starting Flash write for F2/F4/L4
2020-10-21T02:48:13 INFO flash_loader.c: Successfully loaded flash loader in sram
size: 32768
size: 16
2020-10-21T02:48:14 INFO common.c: Starting verification of write complete
2020-10-21T02:48:14 INFO common.c: Flash written and verified! jolly good!
pi@pi-0c:~/ws/stlink/build/Release/bin $ 

```

Expected/description:

`short description of the expected value`

Thank you for your support.

The stlink project maintainers

joseph-cb, 2020-10-21:
Use a different Rpi, another dev board. Apply fix of #981. `make clean; make debug`.  Invoke with `./st-flash --debug write ~/ws/uC/release_STM32T_UFBGA.bin 0x08000000`

Observed the same symptom. Fail once then succeed once, forever.
```
2020-10-21T16:09:37 DEBUG common.c:   core status: running
2020-10-21T16:09:37 DEBUG common.c: *** stlink_status ***
2020-10-21T16:09:37 DEBUG usb.c: core status: 01090001
2020-10-21T16:09:37 DEBUG common.c:   core status: running
2020-10-21T16:09:37 DEBUG common.c: *** stlink_status ***
2020-10-21T16:09:37 DEBUG usb.c: core status: 01090001
2020-10-21T16:09:37 DEBUG common.c:   core status: running
2020-10-21T16:09:37 ERROR flash_loader.c: flash loader run error
2020-10-21T16:09:37 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
2020-10-21T16:09:37 DEBUG common.c: *** stlink_read_debug32 ffffffff is 0x8000000
```

```
002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c: 

2020-10-21T16:10:38 DEBUG common.c: *** stlink_read_mem32 ***
2020-10-21T16:10:38 DEBUG common.c: data_len = 16 0x10
2020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c:  002020-10-21T16:10:38 DEBUG common.c: 

2020-10-21T16:10:38 INFO common.c: Flash written and verified! jolly good!
2020-10-21T16:10:38 DEBUG common.c: *** stlink_read_debug32 20028000 is 0x8000000
```

joseph-cb, 2020-10-21:
`./st-flash --reset write ~/ws/uC/release_STM32T_UFBGA.bin 0x08000000` will make st-flash succeed consistently. Please mention this option in the Tutorial. Better yet, make `--reset` on by default.

joseph-cb, 2020-10-24:
I declared victory too early. The flash tool always says " Flash written and verified! jolly good!" after I added `--reset` but the actual behavior is NOT. The actual behavior is still succeed once then failed once, cycling like that consistently for me.

On the other hand, I git clone openocd source tree and compiled my openocd. I can use it to reliably flash my L452RE
`src/openocd -s tcl -f interface/stlink.cfg -f  target/slow_stm32l4x.cfg -c "program /home/pi/ws/uC/release_STM32T_UFBGA.bin verify reset exit 0x08000000"`

openocd does not give error message and I check behavior after flashing with openocd. My L452RE is reflashed and boot up every time.

Nightwalker-87, 2020-10-24:
The observation that `openocd` is able to flash reliably does help this tool any further.
We can't make any use of it and it's implementation.

sambazley, 2021-02-19:
I think I'm running into the same problem on my F429ZI Nucleo using the on board ST Link, running 4bfaab0. One attempt to flash will output
```
ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
```
and the next will output
```
Flash written and verified! jolly good!
```

However, the first attempt did actually successfully flash and reset. On the second attempt, the STM32 resets and begins to execute the new code, but for some reason crashes after a few lines. I've tested this with this bash line `export X=$((X+1)); CFLAGS="-DX=\\\"$X\\\"" make clean flash; echo "X=$X"` which increments a number, recompiles the binary, and flashes the STM32 using `st-flash write stm32.bin 0x08000000`

[first.txt](https://github.com/stlink-org/stlink/files/6006820/first.txt)
[second.txt](https://github.com/stlink-org/stlink/files/6006822/second.txt)

Ant-ON, 2021-03-19:
@joseph-cb 
@sambazley 
Can you try https://github.com/Ant-ON/stlink/tree/flash_loader_wait_ms? I'm trying to improve a work of bootloaders...

sambazley, 2021-03-20:
I tested it on an STM32F072 which is having the same issue. Every other time I try to flash, I get this:
```
2021-03-20T17:11:21 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-03-20T17:11:21 ERROR flash_loader.c: Write error
2021-03-20T17:11:21 WARN flash_loader.c: Loader state: R2 0x800 R15 0xFFFFFFFE
2021-03-20T17:11:21 WARN flash_loader.c: MCU state: DHCSR 0x30003 DFSR 0x9 CFSR 0x0 HFSR 0x0
2021-03-20T17:11:21 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
```

I can test it on the F429 as well if you want.

sambazley, 2021-03-20:
Sorry, ignore my last comment. Adding `--reset` makes it work consistently on the F072. I tested again on the F429 using both mainline st-flash and v1.6.1, but for some reason I can't reproduce the issue I was having before.

Ant-ON, 2021-03-20:
@sambazley Yes, I test it with --reset. On my assumptions (which I fixed) this failure occurred due to the configured interrupts. In particular, SysTick. I tested this solution on F1 and F072 with the ST examples for the discovery board.

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87