stlink-org/stlink #1063: STM32H743 on a Nucleo-144 board: "unknown chip id"

codedump, 2020-11-04:
Hello everyone,

this being my first time writing here let me begin with: thanks for the effort! :-)

I'm new to both STM32 programming and to ```stlink``` utility, so please bear
with me. I'm fairly proficient in C, have some experience with other micro
controllers and lots of experience with Linux.

Now to my issue. Not sure if this is a bug, a feature request, or none of the
above. Feel free to reclassify if necessary.

I'm trying to connect to a ```STM32H743``` chip on a Nucleo-144 prototyping
board. I'm using the vanilla ```stlink``` installation from Fedora (on a Fedora
Silverblue distro), apparently ```stlink``` version is ```1.6.1```. Symptoms are
as follows:
```
    $ lsusb
    ...
    Bus 001 Device 025: ID 0483:374e STMicroelectronics STLINK-V3
    ...

    $ ls -l /dev/stlink*
    lrwxrwxrwx. 1 root root  3  3. Nov 15:48 /dev/stlinkv3e_ -> sdb
    lrwxrwxrwx. 1 root root 11  3. Nov 15:48 /dev/stlinkv3e_0 -> bsg/2:0:0:0
    lrwxrwxrwx. 1 root root  3  3. Nov 15:48 /dev/stlinkv3e_1 -> sg1
    lrwxrwxrwx. 1 root root 15  3. Nov 15:48 /dev/stlinkv3e_3 -> bus/usb/001/025

    $ st-info --probe
    found 1 stlink programmers
    serial:     303032413030334633343338353130433334333133393339
    hla-serial: "\x30\x30\x32\x41\x30\x30\x33\x46\x33\x34\x33\x38\x35\x31\x30\x43\x33\x34\x33\x31\x33\x39\x33\x39"
    flash:      0 (pagesize: 0)
    sram:       0
    chipid:     0x0000
    descr:      unknown device

    $ st-flash read dummy.bin 0 0xffff
    st-flash 1.6.1
    2020-11-04T10:32:12 INFO usb.c: Unable to match requested speed 1800 kHz, using 1000 kHz
    2020-11-04T10:32:13 WARN common.c: unknown chip id! 0x5fa0004
    Failed to connect to target
```

So apparently the device is visible, but ```stlink``` doesn't know what to
do with it. I know that no STM32H* devices are on the supported list
in ```stlink``` documentation.

My questions:
  - Is it because ST-LINK-V3, or because of the chip?
  - Is there any chance of getting it to work in the short term?
  - Is there anything I can do? (I'm not afraid of compiling ```stlink``` by hand -- I just didn't to it yet, as the version numbers 1.6.1 vs current 1.6.2 don't sound like there would be a substantial change).
  - Is it possibly only a matter of entering chip ID numbers in a magic table and hoping for the best? :-D

Thanks & Cheers,
Florin.

timothytylee, 2020-11-04:
Hi, I've added support for STM32H7 to the master branch, but there are lingering JTAG reset problems plaguing the chip detection.

Your best bet is to try https://github.com/Ant-ON/stlink/commits/try_h7_debug from @Ant-ON , which works well with my NUCLEO-H7432ZI2 devboard.

Ant-ON, 2020-11-04:
@codedump This https://github.com/Ant-ON/stlink/tree/504858566be82af996566795c41684b701b52ce1 shoud work well. Next commits is breaking a code compileble (i'm work on it)

codedump, 2020-11-05:
@Ant-ON Thanks, that seems to work:

    $ ./st-info --probe
    Found 1 stlink programmers
    serial:     303032413030334633343338353130433334333133393339
    hla-serial: "\x30\x30\x32\x41\x30\x30\x33\x46\x33\x34\x33\x38\x35\x31\x30\x43\x33\x34\x33\x31\x33\x39\x33\x39"
    flash:      2097152 (pagesize: 131072)
    sram:       131072
    chipid:     0x0450
    descr:      H742/743/753

    $ ./st-flash read test.foo 0 0xffff
    st-flash 1.6.1-112-g5048585
    2020-11-05T16:17:34 INFO usb.c: Unable to match requested speed 1800 kHz, using 1000 kHz
    2020-11-05T16:17:34 INFO common.c: H742/743/753: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
    2020-11-05T16:17:34 INFO common.c: read from address 0000000000 size 65535

:-)

Cheers,
Florin.

codedump, 2020-11-06:
Oh, and for the record, just in case some other poor soul stumbles over this thread: mine is also a ```NUCLEO-H743ZI2``` board.

closed 2020-12-05 by Nightwalker-87