stlink-org/stlink #1065: STM32F103CB:  All the Register values are wrongly shown when running info registers

hannanmumtaz, 2020-11-07:
- Programmer/board type: Stlink v2
- Operating system and version: Windows
- **Stlink tools version** and/or git commit hash: v1.6.1
- Stlink commandline tool name: st-util
- Target chip (and board if applicable): STM32F103CB

When command info registers is run on gdb i get the wrong output.

I was trying to set register r7 to 0xDEADBEEF  using assembly 

```
.syntax unified
.cpu cortex-m3
.fpu softvfp
.thumb

.global vtable
.global reset_handler

.type vtable, %object
vtable:
    .word _estack
    .word reset_handler
.size vtable, .-vtable

.type reset_handler, %function
reset_handler:

    LDR  r0, =_estack
    MOV  sp, r0
    LDR  r7, =0xDEADBEEF
.size reset_handler, .-reset_handler
```

Commandline-Output:

[image]

```
OUTPUT/ERROR of the commandline tool(s)
```
all register values are shown as r# + 1 so instead of r7 being 0xDEADBEEF r8 is shown with that value also even the stack pointer value is shown infront of lr register.

Expected/description:

r7 should be shown with correct value in registers.
below is the output when using openocd

all registers are correctly displayed when using openocd server.

[image]

Thanks in advance.

Ant-ON, 2020-11-09:
@hannanmumtaz Can you try the develop branch? This should be fixed by #1027

hannanmumtaz, 2020-11-10:
Hi, i tried the develop branch, i see now the registers values are correct, so the original issue is resovled.

But, i noticed some other issues in develop branch maybe i have to create separate ticket for these.

1. calling load in gdb automatically starts execution , previously that was not that case.
2. if a program is already running on the mcu, attaching to it via target extended-remote doesn't stop execution and program keeps running, not the case in v1.6.1 or openocd.
3. calling starti restarts the program but program keeps running in background and prints a random break location which is incorrect as program is never halted after restart.
4. calling step in gdb hangs the gdb and has to be stopped and restarted.

Ant-ON, 2020-11-10:
@hannanmumtaz The first problem I solved in this branch: https://github.com/Ant-ON/stlink/tree/try_h7_debug I will try to look at and solve other problems. It's better to create a separate issue.

Ant-ON, 2020-12-29:
@Nightwalker-87 The main problem is solved. Issues were not created for other problems. I think the current issue can be closed.

Nightwalker-87, 2021-03-11:
@hannanmumtaz: If still relevant: Please check currently open tickets for the other topics and open new issues, if necessary.

@Ant-ON: Can you confirm that the mentioned PR fixes this issue?

Ant-ON, 2021-03-12:
@Nightwalker-87 Yes, the problem described in the first post has been fixed

Nightwalker-87, 2021-03-12:
Fixed in PR #1027.

closed 2021-03-12 by Nightwalker-87