stlink-org/stlink #1077: Build with GCC 11 fail

Vascom, 2020-12-09:
I can't build stlink 1.6.1 with GCC 11 compiler. Fedora GNU/Linux Rawhide.
Errors are:

```
/builddir/build/BUILD/stlink-1.6.1/src/st-util/gdb-server.c:90:22: error: 'stlink_open_usb' accessing 64 bytes in a region of size 28 [-Werror=stringop-overflow=]
   90 |                 sl = stlink_open_usb(st->logging_level, st->reset, serialnumber, 0);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/builddir/build/BUILD/stlink-1.6.1/src/st-util/gdb-server.c: In function 'serve':
/builddir/build/BUILD/stlink-1.6.1/src/st-util/gdb-server.c:90:22: note: referencing argument 3 of type 'char *'
In file included from /builddir/build/BUILD/stlink-1.6.1/include/stlink.h:297,
                 from /builddir/build/BUILD/stlink-1.6.1/src/st-util/gdb-server.c:27:
/builddir/build/BUILD/stlink-1.6.1/src/usb.h:71:15: note: in a call to function 'stlink_open_usb'
   71 |     stlink_t *stlink_open_usb(enum ugly_loglevel verbose, int reset, char serial[STLINK_SERIAL_MAX_SIZE], int freq);
      |               ^~~~~~~~~~~~~~~
```
```
In function 'do_connect',
    inlined from 'do_connect' at /builddir/build/BUILD/stlink-1.6.1/src/st-util/gdb-server.c:87:18,
    inlined from 'main' at /builddir/build/BUILD/stlink-1.6.1/src/st-util/gdb-server.c:212:10:
/builddir/build/BUILD/stlink-1.6.1/src/st-util/gdb-server.c:90:22: error: 'stlink_open_usb' accessing 64 bytes in a region of size 28 [-Werror=stringop-overflow=]
   90 |                 sl = stlink_open_usb(st->logging_level, st->reset, serialnumber, 0);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Please help me correct it.

Vascom, 2020-12-09:
May be `STLINK_SERIAL_MAX_SIZE` define should be set to `28`?

Ant-ON, 2020-12-09:
You may fix `src/st-util/gdb-server.c`:
```C
static char serialnumber[28] = {0};
```
to
```C
static char serialnumber[STLINK_SERIAL_MAX_SIZE] = {0};
```

Vascom, 2020-12-09:
I prefer see fix in upstream.

Ant-ON, 2020-12-09:
@Vascom I too. But to fix it in upstream, you need to check works of it.

Vascom, 2020-12-09:
Yes, it works.

Template not care about compiling problems.

Nightwalker-87, 2020-12-09:
The template is meant to be used and to be filled with content which also includes a description. So please consider this next time.

JeffreyALaw, 2020-12-09:
> You may fix `src/st-util/gdb-server.c`:
> 
> ```c
> static char serialnumber[28] = {0};
> ```
> 
> to
> 
> ```c
> static char serialnumber[STLINK_SERIAL_MAX_SIZE] = {0};
> ```

I nearly suggested that (I originally found the issue testing gcc-11 with Fedora).  My concern with that was this loop in gdb-server.c:
               for (size_t k = 0; j >= 0 && k < sizeof(serialnumber); ++k, j -= 2) {
                    char buffer[3] = {0};
                    memcpy(buffer, optarg + j, 2);
                    serialnumber[length - k] = (uint8_t)strtol(buffer, NULL, 16);

Which would be affected by changing the size of "serialnumber".   Without knowing the intent or the code in general I didn't feel comfortable recommending changing the size of "serialnumber".

And to give folks a bit of background.  gcc-11 has added diagnostics to detect out of bounds accesses for array arguments.  In simplest terms if a function declares an argument as an array with a fixed length, then GCC assumes the entire array could potentially be referenced.

stlink_open_usb declares its 3rd argument as a 64 byte character array, yet "serialnumber" is just 28 bytes and GCC naturally warns because it thinks there's a potential out-of-bounds of the "serialnumber" argument by stlink_open_usb.

commit 95e9c36a (2020-12-12): General Project Update

- Mention support for VS Code (Closes #1057)
- Compilation with GCC 11 fails (Closes #1077)
- Updated list of contributors
- Updated CHANGELOG.md
closed 2020-12-12 by Nightwalker-87