stlink-org/stlink #1082: GD32F103: Failed to unlock flash, flash_loader.c: write error

geocine, 2020-12-22:
*Details*

- [X ] I made serious effort to avoid creating duplicate or nearly similar issue
- Programmer/board type: Stlink v2 Clone
- Operating system and version: MacOS
- **Stlink tools version** and/or git commit hash: 1.6.1
- Stlink commandline tool name: `st-flash`
- Target chip (and board if applicable): GD32F103
   ```
   Found 1 stlink programmers
   serial:     0a240f002b135937334d4e00
   hla-serial: "\x0a\x24\x0f\x00\x2b\x13\x59\x37\x33\x4d\x4e\x00"
   flash:      1064960 (pagesize: 1024)
   sram:       20480
   chipid:     0x0410
   descr:      F1xx Medium-density
   ```

*Description*

1. Unlock using `open-ocd` 
3. flash `generic_boot20_pc13.bin` it was ok but I was not able to enter bootloader mode.
4. flash `naze32_boot20.bin` it was ok but I was not able to enter bootloader mode.
5. flash `gd32f1_generic_boot20_pc13.bin`  it was ok but I was not able to enter bootloader mode.
6. flash `gd32f1_frankenmaple.bin` it was ok but I was not able to enter bootloader mode.
7. flash `#3` again , retrying but I cannot flash anymore with this error shown
    ```sh
    2020-12-22T22:07:26 WARN common.c: Failed to unlock flash!
    2020-12-22T22:07:26 ERROR flash_loader.c: write error, count == 3758157296
    2020-12-22T22:07:26 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
    stlink_fwrite_flash() == -1
    ```

*Commandline-Output:*

```sh
st-flash 1.6.1
2020-12-22T22:07:26 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 1040 KiB flash in at least 1 KiB pages.
file generic_boot20_pc13.bin md5 checksum: 333c30605e739ce9bedee5999fdaf81b, stlink checksum: 0x0008e534
2020-12-22T22:07:26 INFO common.c: Attempting to write 7172 (0x1c04) bytes to stm32 address: 134217728 (0x8000000)
2020-12-22T22:07:26 INFO common.c: Flash page at addr: 0x08000000 erased
2020-12-22T22:07:26 WARN common.c: Failed to unlock flash!
2020-12-22T22:07:26 INFO common.c: Flash page at addr: 0x08000400 erased
2020-12-22T22:07:26 WARN common.c: Failed to unlock flash!
2020-12-22T22:07:26 INFO common.c: Flash page at addr: 0x08000800 erased
2020-12-22T22:07:26 WARN common.c: Failed to unlock flash!
2020-12-22T22:07:26 INFO common.c: Flash page at addr: 0x08000c00 erased
2020-12-22T22:07:26 WARN common.c: Failed to unlock flash!
2020-12-22T22:07:26 INFO common.c: Flash page at addr: 0x08001000 erased
2020-12-22T22:07:26 WARN common.c: Failed to unlock flash!
2020-12-22T22:07:26 INFO common.c: Flash page at addr: 0x08001400 erased
2020-12-22T22:07:26 WARN common.c: Failed to unlock flash!
2020-12-22T22:07:26 INFO common.c: Flash page at addr: 0x08001800 erased
2020-12-22T22:07:26 WARN common.c: Failed to unlock flash!
2020-12-22T22:07:26 INFO common.c: Flash page at addr: 0x08001c00 erased
2020-12-22T22:07:26 INFO common.c: Finished erasing 8 pages of 1024 (0x400) bytes
2020-12-22T22:07:26 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2020-12-22T22:07:26 INFO flash_loader.c: Successfully loaded flash loader in sram
2020-12-22T22:07:26 WARN common.c: Failed to unlock flash!
2020-12-22T22:07:26 ERROR flash_loader.c: write error, count == 3758157296
2020-12-22T22:07:26 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
```

*Expected/description:*

I expect that it should be able to flash

*What I've tried so far*

1. Unlock again using `open-ocd`
    Command
    ```sh
    openocd -f interface/stlink-v2.cfg \ 
     -f target/stm32f1x.cfg \
     -c "init; reset halt; stm32f1x unlock 0; reset halt; exit"
    ```
    Output
    ```
    Open On-Chip Debugger 0.10.0
    Licensed under GNU GPL v2
    For bug reports, read
	    http://openocd.org/doc/doxygen/bugs.html
    Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
    Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
    adapter speed: 1000 kHz
    adapter_nsrst_delay: 100
    none separate
    Info : Unable to match requested speed 1000 kHz, using 950 kHz
    Info : Unable to match requested speed 1000 kHz, using 950 kHz
    Info : clock speed 950 kHz
    Info : STLINK v2 JTAG v31 API v2 SWIM v7 VID 0x0483 PID 0x3748
    Info : using stlink api v2
    Info : Target voltage: 3.222134
    Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
    target halted due to debug-request, current mode: Thread
    xPSR: 0x01000000 pc: 0x080000f0 msp: 0x20002800
    Info : device id = 0x13030410
    Info : flash size = 128kbytes
    target halted due to breakpoint, current mode: Thread
    xPSR: 0x61000000 pc: 0x2000003a msp: 0x20002800
    stm32x unlocked.
    INFO: a reset or power cycle is required for the new settings to take effect.
    target halted due to debug-request, current mode: Thread
    xPSR: 0x01000000 pc: 0x080000f0 msp: 0x20002800
    ```
2. I tried to flash again, and still get the error above.

Nightwalker-87, 2020-12-22:
Please also take note of #769 in this context.

Ant-ON, 2020-12-23:
@geocine There is no need to unlock through OpenOСD. After resetting the microcontroller, the flash register lock appears again. I looked at the documentation for GD32F103. Unlocking is similar to STM32F103. `st-flash` must do it correctly.

Specifically, which microcontroller is used? In the OpenOCD logs, the size of the flash is detected as 128k, and `st-flash` is detected as 1024k.

ps I would recommend testing the flashing on the `st-flash` compiled from source of `develop` branch.

geocine, 2020-12-23:
@Ant-ON thanks I have tried this version `v1.6.1-199-g104940e` and it is working perfectly. My microcontroller is `GD32F103CBT6` . Thank you so much for the very quick response, I am closing this issue now.

closed 2020-12-23 by geocine