stlink-org/stlink #1083: [STM32 device name]: STM32G0 not find and not suported

Krasnov-Max, 2020-12-27:
Thank you for giving feedback to the stlink project.

**NOTICE: Please read and follow instructions in #906 before submitting a ticket. This feature request will be deleted without notice when not enough information is provided! So please ensure that all fields are filled out.**

- [ ] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: [Stlink V2] Nucleo-F401RE
- Operating system and version: [Ubuntu 20.04] 
- **Stlink tools version** and/or git commit hash: [v1.6.1]
- Stlink commandline tool name: [st-info], st-flash 
- Target chip (and board if applicable): [STM32G031J4Mx]

unknown chip id! 0x374b

Commandline-Output:
> st-flash --debug erase
st-flash 1.6.1
2020-12-27T23:03:03 DEBUG common.c: *** looking up stlink version
2020-12-27T23:03:03 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-12-27T23:03:03 DEBUG common.c: stlink pid     = 0x374b
2020-12-27T23:03:03 DEBUG common.c: stlink version = 0x2
2020-12-27T23:03:03 DEBUG common.c: jtag version   = 0x24
2020-12-27T23:03:03 DEBUG common.c: swim version   = 0x1a
2020-12-27T23:03:03 DEBUG common.c: *** looking up stlink version
2020-12-27T23:03:03 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-12-27T23:03:03 DEBUG common.c: stlink pid     = 0x374b
2020-12-27T23:03:03 DEBUG common.c: stlink version = 0x2
2020-12-27T23:03:03 DEBUG common.c: jtag version   = 0x24
2020-12-27T23:03:03 DEBUG common.c: swim version   = 0x1a
2020-12-27T23:03:03 DEBUG common.c: stlink current mode: mass
2020-12-27T23:03:03 DEBUG usb.c: JTAG/SWD freq set to 0
2020-12-27T23:03:03 DEBUG common.c: *** set_swdclk ***
2020-12-27T23:03:03 DEBUG common.c: stlink current mode: mass
2020-12-27T23:03:03 DEBUG common.c: *** stlink_enter_swd_mode ***
2020-12-27T23:03:03 DEBUG common.c: *** stlink_jtag_reset ***
2020-12-27T23:03:03 DEBUG common.c: *** stlink_reset ***
2020-12-27T23:03:03 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-12-27T23:03:03 DEBUG common.c: Loading device parameters....
2020-12-27T23:03:03 DEBUG common.c: *** stlink_core_id ***
2020-12-27T23:03:03 DEBUG common.c: core_id = 0x0000374b
2020-12-27T23:03:03 DEBUG common.c: *** stlink_read_debug32 374b is 0xe0042000
2020-12-27T23:03:03 WARN common.c: unknown chip id! 0x374b
Failed to connect to target

st-info --probe
Found 1 stlink programmers
 serial:     303637304646353235373530383737323637313535333434
 hla-serial: "\x30\x36\x37\x30\x46\x46\x35\x32\x35\x37\x35\x30\x38\x37\x37\x32\x36\x37\x31\x35\x35\x33\x34\x34"
 flash:      0 (pagesize: 0)
 sram:       0
 chipid:     0x074b

Ant-ON, 2020-12-29:
@Krasnov-Max I checked the NUCLEO-G031K8 board for G0 support. Everything is working. There may be some problems with the reset. Can you try `st-flash` builded from source of `develop` branch? It may also be that the target board have firmware with remapped swd pin. In this case see https://github.com/stlink-org/stlink/issues/1012#issuecomment-751650320

Krasnov-Max, 2020-12-31:
This new mcu. This mcu not reset pin. so8 packege. connect to programmator four line swdclk, swdio, vdd, gnd

Ant-ON, 2021-01-02:
@Krasnov-Max In version v1.6.1 there are problems with resetting mcu without (or without connect) reset pin. Can you try `st-flash` builded from source of `develop` branch?

Nightwalker-87, 2021-03-12:
Ping @Krasnov-Max

Nightwalker-87, 2021-03-25:
Ah, this appears to be the topic with SWD lacking a `NRST` reset pin. At least it seems related.

Nightwalker-87, 2021-03-27:
@Ant-ON I think there will be no further response from the author. His account has no link to this project any more.

Nightwalker-87, 2021-03-28:
Based on https://github.com/stlink-org/stlink/issues/1083#issuecomment-752013261 and https://github.com/stlink-org/stlink/issues/1083#issuecomment-753481954 appropriate advice has been given to satisfiable resolve this topic.
Thus we can now safely close this issue as no related bug appears to be present in the toolset.

closed 2021-03-28 by Nightwalker-87