stlink-org/stlink #1084: st-flash terminates unexpectedly when trying to read the chip [STM32F103C8T6]: [hobbyist]

asafsirin, 2020-12-30:
Hi 

The st-flash unexpectedly terminates when trying to read from the device in subject.

- Programmer/board type: st-link v2-clone
- Windows 7 sp1
- st-link 1.6.1
- st-flash
- Target chip (and board if applicable): [enter here] (e.g STM32F103C8T6)

st-info runs without problem:
```
C:\Users\user1>st-info --probe
Found 1 stlink programmers
 serial:     400030001000004150514e4e
 hla-serial: "\x40\x00\x30\x00\x10\x00\x00\x41\x50\x51\x4e\x4e"
 flash:      65536 (pagesize: 1024)
 sram:       20480
 chipid:     0x0410
 descr:      F1xx Medium-density

```

st-flash program returns with error:
```
C:\Users\user1>st-flash --debug --reset --area=option read > option_byte
2020-12-30T09:27:06 DEBUG common.c: *** looking up stlink version
2020-12-30T09:27:06 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-12-30T09:27:06 DEBUG common.c: stlink pid     = 0x3748
2020-12-30T09:27:06 DEBUG common.c: stlink version = 0x2
2020-12-30T09:27:06 DEBUG common.c: jtag version   = 0x1d
2020-12-30T09:27:06 DEBUG common.c: swim version   = 0x7
2020-12-30T09:27:06 DEBUG common.c: *** looking up stlink version
2020-12-30T09:27:06 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2020-12-30T09:27:06 DEBUG common.c: stlink pid     = 0x3748
2020-12-30T09:27:06 DEBUG common.c: stlink version = 0x2
2020-12-30T09:27:06 DEBUG common.c: jtag version   = 0x1d
2020-12-30T09:27:06 DEBUG common.c: swim version   = 0x7
2020-12-30T09:27:06 DEBUG common.c: stlink current mode: mass
2020-12-30T09:27:06 DEBUG usb.c: JTAG/SWD freq set to 0
2020-12-30T09:27:06 DEBUG common.c: *** set_swdclk ***
2020-12-30T09:27:06 DEBUG common.c: stlink current mode: mass
2020-12-30T09:27:06 DEBUG common.c: *** stlink_enter_swd_mode ***
2020-12-30T09:27:06 DEBUG common.c: *** stlink_jtag_reset ***
2020-12-30T09:27:06 DEBUG common.c: *** stlink_reset ***
2020-12-30T09:27:06 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2020-12-30T09:27:06 DEBUG common.c: Loading device parameters....
2020-12-30T09:27:06 DEBUG common.c: *** stlink_core_id ***
2020-12-30T09:27:06 DEBUG common.c: core_id = 0x1ba01477
2020-12-30T09:27:06 DEBUG common.c: *** stlink_read_debug32 20036410 is 0xe0042000
2020-12-30T09:27:06 DEBUG common.c: *** stlink_read_debug32 ffff0040 is 0x1ffff7e0
2020-12-30T09:27:06 INFO common.c:
C:\Users\user1>st-flash --version
v1.6.1

```

ST-Link drivers were installed from latest stmcubprogrammer v2.5.0. 
The cubeprogrammer doesn't recognize the st-link so it wasn't possible to establish connection.
To workaround the problem programmer was uninstalled and drivers were deleted, and ST's ST-Link utility was installed (and its drivers too) then.  
Installed st-link utility v4.6.0.0, driver version - v5.1.2.0. Nothing changed on st-flash side. 
Note that the st-link utility could read the chip without problem. Didn't test write yet. 

Previous versions of st-link (older than 1.6.1) were not tried. 

Regards
/A

asafsirin, 2020-12-30:
st-linkv2-clone firmware revision: 
```
C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility>ST-LINK_CLI.exe -c
STM32 ST-LINK CLI v3.6.0.0
STM32 ST-LINK Command Line Interface

ST-LINK SN: 400030001000004150514E4E
ST-LINK Firmware version: V2J29S7
Connected via SWD.
SWD Frequency = 4000K.
Target voltage = 3.2 V
Connection mode: Normal
Reset mode: Hardware reset
Device ID: 0x410
Device flash Size: 64 Kbytes
Device family: STM32F10xx Medium-density
```

asafsirin, 2020-12-30:
Upgraded the v2-clone firmware using ST's ST-Link utility to V2J37S7. Fault is still there.

Nightwalker-87, 2021-03-10:
I couldn't reproduce the described issue in this environment:

- Release v1.6.1 (commit 6aa08a659a701feb9c08a4cb1628b92645f41309)
- STM32F103C8T6 ("Bluepill") and
- a STLink/V2-clone programmer (Firmware version: V2.J37.S7)
- Linux (Debian Bullseye [Testing])

```
st-flash --debug --reset --area=option read > option_byte
2021-03-11T00:00:15 DEBUG common.c: *** looking up stlink version
2021-03-11T00:00:15 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-03-11T00:00:15 DEBUG common.c: stlink pid     = 0x3748
2021-03-11T00:00:15 DEBUG common.c: stlink version = 0x2
2021-03-11T00:00:15 DEBUG common.c: jtag version   = 0x25
2021-03-11T00:00:15 DEBUG common.c: swim version   = 0x7
2021-03-11T00:00:15 DEBUG common.c: stlink current mode: mass
2021-03-11T00:00:15 DEBUG usb.c: JTAG/SWD freq set to 0
2021-03-11T00:00:15 DEBUG common.c: *** set_swdclk ***
2021-03-11T00:00:15 DEBUG common.c: stlink current mode: mass
2021-03-11T00:00:15 DEBUG common.c: *** stlink_enter_swd_mode ***
2021-03-11T00:00:15 DEBUG common.c: *** stlink_jtag_reset ***
2021-03-11T00:00:15 DEBUG common.c: *** stlink_reset ***
2021-03-11T00:00:15 WARN usb.c: NRST is not connected
2021-03-11T00:00:15 DEBUG usb.c: Using reset through SYSRESETREQ
2021-03-11T00:00:15 DEBUG common.c: *** stlink_soft_reset ***
2021-03-11T00:00:15 DEBUG common.c: *** stlink_write_debug32 0xa05f0003 to 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_write_debug32 0x01000500 to 0xe000edfc
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0x01030003 at 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0x03030003 at 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_write_debug32 0x0000001f to 0xe000ed30
2021-03-11T00:00:15 DEBUG common.c: Loading device parameters....
2021-03-11T00:00:15 DEBUG common.c: *** stlink_core_id ***
2021-03-11T00:00:15 DEBUG common.c: core_id = 0x2ba01477
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0x20036410 at 0xe0042000
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0xffff0040 at 0x1ffff7e0
2021-03-11T00:00:15 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
2021-03-11T00:00:15 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-03-11T00:00:15 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-03-11T00:00:15 DEBUG common.c: *** stlink_jtag_reset ***
2021-03-11T00:00:15 DEBUG common.c: *** stlink_reset ***
2021-03-11T00:00:15 WARN usb.c: NRST is not connected
2021-03-11T00:00:15 DEBUG usb.c: Using reset through SYSRESETREQ
2021-03-11T00:00:15 DEBUG common.c: *** stlink_soft_reset ***
2021-03-11T00:00:15 DEBUG common.c: *** stlink_write_debug32 0xa05f0003 to 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_write_debug32 0x01000500 to 0xe000edfc
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0x03030003 at 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_write_debug32 0x0000001f to 0xe000ed30
2021-03-11T00:00:15 DEBUG common.c: *** stlink_force_debug_mode ***
2021-03-11T00:00:15 DEBUG common.c: *** stlink_status ***
2021-03-11T00:00:15 DEBUG usb.c: core status: 00030003
2021-03-11T00:00:15 DEBUG common.c:   core status: halted
2021-03-11T00:00:15 DEBUG flash.c: @@@@ Read 0 (0) option bytes from          0
2021-03-11T00:00:15 ERROR common.c: Option bytes read is currently not supported for connected chip
2021-03-11T00:00:15 DEBUG common.c: *** stlink_exit_debug_mode ***
2021-03-11T00:00:15 DEBUG common.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2021-03-11T00:00:15 DEBUG common.c: *** stlink_close ***
```

Can somebody run the same procedure on a Windows system again?

Nightwalker-87, 2021-03-25:
@Ant-ON As you were in tough with this scope: Is there any indication for a bug here?

Ant-ON, 2021-03-25:
@Nightwalker-87 There may be some problems related with Windows. I work in Linux and have not noticed this behavior

Nightwalker-87, 2021-03-26:
@slyshykO Can you try to reproduce it on Windows?

slyshykO, 2021-03-26:
Ok, will see

slyshykO, 2021-03-27:
@Nightwalker-87 
I've tested the latest develop on Win10. Build with MinGW-32 and msvc-19 x86.
Both build run without unexpected terminations.

Nightwalker-87, 2021-03-27:
Did you use the same command: `st-flash --debug --reset --area=option read > option_byte` ? Can you post your output ?
BTW: One should note that option byte reading is currently not implemented for the F0 & F1 series.

slyshykO, 2021-03-27:
```sh
>st-info --probe
Found 1 stlink programmers
  version:    V2J30S19
  serial:     066AFF575152836687074835
  flash:      65536 (pagesize: 1024)
  sram:       20480
  chipid:     0x0410
  descr:      F1xx Medium-density
```

```sh
>st-flash --debug --reset --area=option read
st-flash 1.6.1-247-gcfff13c
2021-03-27T12:31:37 DEBUG common.c: *** looking up stlink version
2021-03-27T12:31:37 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-03-27T12:31:37 DEBUG common.c: stlink pid     = 0x374b
2021-03-27T12:31:37 DEBUG common.c: stlink version = 0x2
2021-03-27T12:31:37 DEBUG common.c: jtag version   = 0x1e
2021-03-27T12:31:37 DEBUG common.c: swim version   = 0x13
2021-03-27T12:31:37 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-03-27T12:31:37 DEBUG usb.c: JTAG/SWD freq set to 0
2021-03-27T12:31:37 DEBUG common.c: *** set_swdclk ***
2021-03-27T12:31:37 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-03-27T12:31:37 DEBUG common.c: Loading device parameters....
2021-03-27T12:31:37 DEBUG common.c: *** stlink_core_id ***
2021-03-27T12:31:37 DEBUG common.c: core_id = 0x1ba01477
2021-03-27T12:31:37 DEBUG common.c: *** stlink_read_debug32 0x411fc231 at 0xe000ed00
2021-03-27T12:31:38 DEBUG common.c: *** stlink_read_debug32 0x20036410 at 0xe0042000
2021-03-27T12:31:38 DEBUG common.c: *** stlink_read_debug32 0xffff0040 at 0x1ffff7e0
2021-03-27T12:31:38 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
2021-03-27T12:31:38 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-03-27T12:31:38 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-03-27T12:31:38 DEBUG common.c: *** stlink_jtag_reset ***
2021-03-27T12:31:38 DEBUG common.c: *** stlink_reset ***
2021-03-27T12:31:38 WARN usb.c: NRST is not connected
2021-03-27T12:31:38 DEBUG usb.c: Using reset through SYSRESETREQ
2021-03-27T12:31:38 DEBUG common.c: *** stlink_soft_reset ***
2021-03-27T12:31:38 DEBUG common.c: *** stlink_write_debug32 0xa05f0003 to 0xe000edf0
2021-03-27T12:31:38 DEBUG common.c: *** stlink_write_debug32 0x01000500 to 0xe000edfc
2021-03-27T12:31:38 DEBUG common.c: *** stlink_read_debug32 0x01030003 at 0xe000edf0
2021-03-27T12:31:38 DEBUG common.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2021-03-27T12:31:38 DEBUG common.c: *** stlink_read_debug32 0x03010001 at 0xe000edf0
2021-03-27T12:31:38 DEBUG common.c: *** stlink_read_debug32 0x01010001 at 0xe000edf0
2021-03-27T12:31:38 DEBUG common.c: *** stlink_write_debug32 0x0000001f to 0xe000ed30
2021-03-27T12:31:38 DEBUG common.c: *** stlink_force_debug_mode ***
2021-03-27T12:31:38 DEBUG common.c: *** stlink_status ***
2021-03-27T12:31:38 DEBUG usb.c: core status: 01030003
2021-03-27T12:31:38 DEBUG common.c:   core status: halted
2021-03-27T12:31:38 DEBUG flash.c: @@@@ Read 0 (0) option bytes from          0
2021-03-27T12:31:38 ERROR common.c: Option bytes read is currently not supported for connected chip
could not read option bytes (-1)
2021-03-27T12:31:38 DEBUG common.c: *** stlink_exit_debug_mode ***
2021-03-27T12:31:38 DEBUG common.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2021-03-27T12:31:38 DEBUG common.c: *** stlink_close ***
```

Nightwalker-87, 2021-03-27:
Yeah, it is exactly the same - thx :wink:

PR #1112 (2021-05-10): Support for writing option bytes on STM32F0/F1/F3
Implementation to write option bytes for STM32F0/F1/F3.

(Closes #346, Closes #458, Closes #808, Closes #1084)
closed 2021-05-10 by Nightwalker-87