stlink-org/stlink #1088: STM32H743ZI: GDB `info all-registers` failing

eosti, 2021-01-20:
- [x] I made serious effort to avoid creating duplicate or nearly similar issue

- Programmer/board type: [STLink V3] 
- Operating system and version: [macOS 11.1 and Raspberry Pi OS 01-2021]
- **Stlink tools version** and/or git commit hash: [v1.6.1-201-g4bfaab0]
- Stlink commandline tool name: [st-util] 
- Target chip (and board if applicable): [STM32H743ZI (on NUCLEO-H743ZI2 dev kit)] 

I'm trying to read the registers of the H7 with `arm-none-eabi-gdb` using `info all-registers`. The Nucleo is connected to the debugging server via USB and the onboard STLink v3. This behavior is the same when st-util is running on a Raspberry Pi or on MacOS. 

With st-util, I get the following error after running `info all-registers`:

`arm-none-eabi-gdb` output:

```
(gdb) target extended-remote localhost:4242
(gdb) info all-registers
r0             0x0                 0
r1             0x0                 0
r2             0x0                 0
r3             0x0                 0
r4             0x0                 0
r5             0x0                 0
r6             0x0                 0
r7             0x2407ffdc          604504028
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0x0                 0
sp             0x2407ffdc          0x2407ffdc
lr             0xfffffff9          4294967289
pc             0x8000e0a           0x8000e0a <HardFault_Handler+4>
Could not fetch register "f0"; remote failure reply 'E00'
```

`st-util -v99` output (relevant section):

```
2021-01-19T21:09:34 DEBUG common.c: *** stlink_read_all_regs ***
2021-01-19T21:09:34 DEBUG gdb-server.c: send: 00000000000000000000000000000000000000000000000000000000dcff07240000000000000000000000000000000000000000dcff0724f9ffffff0a0e0008
2021-01-19T21:09:34 DEBUG gdb-server.c: recv: p10
2021-01-19T21:09:34 DEBUG gdb-server.c: p packet: stlink_read_unsupported_reg failed with id 16
2021-01-19T21:09:34 DEBUG gdb-server.c: send: E00
```

The GDB command `info registers` functions as expected, it's the `all-registers` part that seems to be tripping stlink up. 

OpenOCD is able to run `info all-registers` (using OpenOCD 0.11.0-rc1+dev-gbd1adcf-dirty and the board/st_nucleo_h743zi.cfg profile):

```
(gdb) target extended-remote localhost:3333
(gdb) info all-registers
r0             0x1888a             100490
r1             0x1                 1
r2             0x12d               301
r3             0x1888a             100490
r4             0x0                 0
r5             0x0                 0
r6             0x0                 0
r7             0x2407ffdc          604504028
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0x0                 0
sp             0x2407ffdc          0x2407ffdc
lr             0x8001111           134222097
pc             0x80010d8           0x80010d8 <HAL_GetTick+12>
xPSR           0x21000000          553648128
d0             0                   (raw 0x0000000000000000)
d1             0                   (raw 0x0000000000000000)
d2             0                   (raw 0x0000000000000000)
d3             0                   (raw 0x0000000000000000)
d4             0                   (raw 0x0000000000000000)
d5             0                   (raw 0x0000000000000000)
d6             0                   (raw 0x0000000000000000)
d7             -nan(0xfffff00000000) (raw 0xffffffff00000000)
d8             0                   (raw 0x0000000000000000)
d9             0                   (raw 0x0000000000000000)
d10            0                   (raw 0x0000000000000000)
d11            0                   (raw 0x0000000000000000)
d12            0                   (raw 0x0000000000000000)
d13            0                   (raw 0x0000000000000000)
d14            0                   (raw 0x0000000000000000)
d15            -nan(0xfffff00000000) (raw 0xffffffff00000000)
fpscr          0x0                 0
msp            0x2407ffdc          0x2407ffdc
psp            0x0                 0x0
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
s0             0                   (raw 0x00000000)
s1             0                   (raw 0x00000000)
s2             0                   (raw 0x00000000)
s3             0                   (raw 0x00000000)
s4             0                   (raw 0x00000000)
s5             0                   (raw 0x00000000)
s6             0                   (raw 0x00000000)
s7             0                   (raw 0x00000000)
s8             0                   (raw 0x00000000)
s9             0                   (raw 0x00000000)
s10            0                   (raw 0x00000000)
s11            0                   (raw 0x00000000)
s12            0                   (raw 0x00000000)
s13            0                   (raw 0x00000000)
s14            0                   (raw 0x00000000)
s15            -nan(0x7fffff)      (raw 0xffffffff)
s16            0                   (raw 0x00000000)
s17            0                   (raw 0x00000000)
s18            0                   (raw 0x00000000)
s19            0                   (raw 0x00000000)
s20            0                   (raw 0x00000000)
s21            0                   (raw 0x00000000)
s22            0                   (raw 0x00000000)
s23            0                   (raw 0x00000000)
s24            0                   (raw 0x00000000)
s25            0                   (raw 0x00000000)
s26            0                   (raw 0x00000000)
s27            0                   (raw 0x00000000)
s28            0                   (raw 0x00000000)
s29            0                   (raw 0x00000000)
s30            0                   (raw 0x00000000)
s31            -nan(0x7fffff)      (raw 0xffffffff)
```

The register values for OpenOCD are different due to being in different points in the code; the point is that there are a bunch of registers that st-util can't read.

Thanks!

Ant-ON, 2021-03-19:
Can you try the `develop` branch? This bug shoud be fixed by #1027

eosti, 2021-03-19:
I updated my develop branch, so now `st-util` is using `v1.6.1-225-g125cb2d` but the issue still persists. It looks like the PR you linked was merged to the develop branch in September, so I think I've already been testing with that patch.

Ant-ON, 2021-03-20:
@eosti Sorry. I didn't pay enough attention to the description. I fixed the problem. Can you try this branch: https://github.com/Ant-ON/stlink/tree/gdb_features ?

eosti, 2021-03-20:
Yay, that looks like that fixed it. I can now see all the registers, without any errors. Thank you so much!

PR #1109 (2021-03-20): GDB: Fixed problems with target description
Adds a description of registers for all targets (#1088). Fixes an issue with initializing memory segments. They were initializing with an incorrect pattern (#1013).

(Closes #1013, Closes #1088)
closed 2021-03-20 by Nightwalker-87