stlink-org/stlink #1090: STM32WLx5 Support

super-cow-powers, 2021-02-01:
Currently the WLx5 family are not supported. 
I would like to open a PR to merge WLx5 support based on the work done for the WB family of chips.

wolsty7, 2021-07-14:
I have the NUCLEO-WL55JC1 dev board now and am developing an application based on the STM32WL55JC with that arriving in a few weeks so I'm keen to help with testing where I can.

Nightwalker-87, 2022-01-03:
@super-cow-powers @wolsty7 Can we finally proceed here?

Nightwalker-87, 2022-01-12:
Please provide the full device-specific parametres as well as the ID of the respective ST reference manual.
This would allow to complete the generated chip file for this device type which currently serves as a template only.

wolsty7, 2022-01-18:
@Nightwalker-87 
I tested the NUCLEO-WL55JC1 dev board on fe48a98cb1802732edb117fe60cd9b2fa0689145 and it flashed perfectly

Nightwalker-87, 2022-01-18:
Device specific parametres for the STM32WLEx series were updated in commit e62b9e1f2a1b6dea7e9ccb053c14c2c6ad0b57d7.

PR #1216 (2022-01-28): [refactoring] General maintenance for code structure
List of changes and improvements:

- Revised device-specific parametres in the .chip files
- Aligned naming for .chip files and defines
- Unified formatting for .chip files
- Removed function `dump_a_chip` and old device database from `common.c`
- Moved-out STM32-specific defines to `stm32.h`
- Updated MCU core-ids
- Removed additional `extern "C"` linkage specs

(Closes #903) (Closes #1090) (Closes #1199) (Closes #1212)
closed 2022-01-28 by Nightwalker-87