stlink-org/stlink #1092: STM32F767ZI: v1.6.1 F7 flashing not supported anymore...

domingguss, 2021-02-19:
Thank you for giving feedback to the stlink project.

**NOTICE: Please read and follow instructions in #906 before submitting a ticket. This feature request will be deleted without notice when not enough information is provided! So please ensure that all fields are filled out.**

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to fill out each of the following items appropriate to your specific problem:

- Programmer/board type: STLINK v2 clone
- Operating system and version: Linux (debian) in docker image
- **Stlink tools version** and/or git commit hash: `v1.6.1` (though it worked in v1.5.0 and v1.6.0)
- Stlink commandline tool name: `st-flash`
- Target chip (and board if applicable): STM32F767ZI

Futher we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline-Output:
```
$ echo "flashing SENIOR (254)..."
flashing SENIOR (254)...
$ st-flash --serial 303637354646353735323531373137383637313433343535 write Senior.bin 0x8010000
2021-02-19T22:25:51 INFO common.c: F76xxx: 512 KiB SRAM, 2048 KiB flash in at least 2 KiB pages.
st-flash 1.6.1
file Senior.bin md5 checksum: 5642b66d9385252e3205ba2b8d18496, stlink checksum: 0x0351e002
2021-02-19T22:25:51 INFO common.c: Attempting to write 524889 (0x80259) bytes to stm32 address: 134283264 (0x8010000)
2021-02-19T22:25:51 WARN common.c: unaligned len 0x80259 -- padding with zero
EraseFlash - Sector:0x2 Size:0x8000 2021-02-19T22:25:51 INFO common.c: Flash page at addr: 0x08010000 erased
EraseFlash - Sector:0x3 Size:0x8000 2021-02-19T22:25:51 INFO common.c: Flash page at addr: 0x08018000 erased
EraseFlash - Sector:0x4 Size:0x20000 2021-02-19T22:25:53 INFO common.c: Flash page at addr: 0x08020000 erased
EraseFlash - Sector:0x5 Size:0x40000 2021-02-19T22:25:55 INFO common.c: Flash page at addr: 0x08040000 erased
EraseFlash - Sector:0x6 Size:0x40000 2021-02-19T22:25:57 INFO common.c: Flash page at addr: 0x08080000 erased
2021-02-19T22:25:57 INFO common.c: Finished erasing 5 pages of 262144 (0x40000) bytes
2021-02-19T22:25:57 INFO common.c: Starting Flash write for F2/F4/F7/L4
2021-02-19T22:25:57 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-02-19T22:25:57 INFO common.c: enabling 32-bit flash writes
2021-02-19T22:25:57 INFO common.c: Go to Thumb mode
2021-02-19T22:25:58 ERROR flash_loader.c: Flash loader run error (R2 0x00000000 R15 0x00000000 DHCSR 0x00010001 DFSR 0x0000000B)
2021-02-19T22:25:58 ERROR common.c: stlink_flash_loader_run(0x8010000) failed! == -1
2021-02-19T22:25:58 INFO common.c: Go to Thumb mode
stlink_fwrite_flash() == -1
```

Expected/description:

Expected to work as in v1.5.1, see below. 
I'm now diffing the logs, maybe you haven't tested it with this rare 2MB Flash version we are using?

Because I also need to flash F3s which now have a different bug (verification failed) that might be fixed in the latest version, I wanted to try the latest version, but this brings me further from home.. Kinda stuck now...

```
$ echo "flashing SENIOR (254)..."
flashing SENIOR (254)...
$ st-flash --serial 303637354646353735323531373137383637313433343535 write Senior.bin 0x8010000
2021-02-19T14:48:45 INFO common.c: Loading device parameters....
st-flash 1.5.1-50-g3690de9
2021-02-19T14:48:45 INFO common.c: Device connected is: F76xxx device, id 0x10016451
2021-02-19T14:48:45 INFO common.c: SRAM size: 0x80000 bytes (512 KiB), Flash: 0x200000 bytes (2048 KiB) in pages of 2048 bytes
2021-02-19T14:48:45 INFO common.c: Attempting to write 524889 (0x80259) bytes to stm32 address: 134283264 (0x8010000)
2021-02-19T14:48:45 WARN common.c: unaligned len 0x80259 -- padding with zero
Flash page at addr: 0x08080000 erased
2021-02-19T14:48:51 INFO common.c: Finished erasing 5 pages of 262144 (0x40000) bytes
2021-02-19T14:48:51 INFO common.c: Starting Flash write for F2/F4/L4
2021-02-19T14:48:51 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-02-19T14:48:59 INFO common.c: Starting verification of write complete
enabling 32-bit flash writes
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 602
2021-02-19T14:49:04 INFO common.c: Flash written and verified! jolly good!
$ sleep 4
[..]
```

Ant-ON, 2021-02-20:
@domingguss As I can see from the log the develop branch is being used. I'm right?
Can you rebuild source with increased `WAIT_ROUNDS` in `src/stlink-lib/flash_loader.c`?

domingguss, 2021-02-20:
@Ant-ON hi, I tried  `30,100,300, 1000, 3000` all without luck... Or should I increase it significantly more?

i took the liberty to log the used value together with the printf(STLINK_VERSION) as u can see, to make sure it was my forked and changed version.

```
$ echo "flashing SENIOR (254)..."
flashing SENIOR (254)...
$ st-flash --serial 303637354646353735323531373137383637313433343535 write Senior.bin 0x8010000
st-flash 1.6.1 (WAIT_ROUNDS=3000)
2021-02-20T22:18:31 INFO common.c: F76xxx: 512 KiB SRAM, 2048 KiB flash in at least 2 KiB pages.
2021-02-20T22:18:31 INFO common.c: Attempting to write 524889 (0x80259) bytes to stm32 address: 134283264 (0x8010000)
2021-02-20T22:18:31 WARN common.c: unaligned len 0x80259 -- padding with zero
file Senior.bin md5 checksum: 5642b66d9385252e3205ba2b8d18496, stlink checksum: 0x0351e002
EraseFlash - Sector:0x2 Size:0x8000 2021-02-20T22:18:31 INFO common.c: Flash page at addr: 0x08010000 erased
EraseFlash - Sector:0x3 Size:0x8000 2021-02-20T22:18:31 INFO common.c: Flash page at addr: 0x08018000 erased
EraseFlash - Sector:0x4 Size:0x20000 2021-02-20T22:18:32 INFO common.c: Flash page at addr: 0x08020000 erased
EraseFlash - Sector:0x5 Size:0x40000 2021-02-20T22:18:34 INFO common.c: Flash page at addr: 0x08040000 erased
EraseFlash - Sector:0x6 Size:0x40000 2021-02-20T22:18:36 INFO common.c: Flash page at addr: 0x08080000 erased
2021-02-20T22:18:36 INFO common.c: Finished erasing 5 pages of 262144 (0x40000) bytes
2021-02-20T22:18:36 INFO common.c: Starting Flash write for F2/F4/F7/L4
2021-02-20T22:18:36 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-02-20T22:18:36 INFO common.c: enabling 32-bit flash writes
2021-02-20T22:19:09 ERROR flash_loader.c: Flash loader run error (R2 0x00000000 R15 0x00000000 DHCSR 0x00090001 DFSR 0x0000000B)
2021-02-20T22:19:09 ERROR common.c: stlink_flash_loader_run(0x8010000) failed! == -1
2021-02-20T22:19:09 INFO common.c: Go to Thumb mode
stlink_fwrite_flash() == -1
Uploading artifacts...
```

Ant-ON, 2021-02-21:
@dobromyslov No, no need significant increase.
As I can see from the DFSR status registers the processor is in the "Domain fault on Page" error. 
In the last log, he is also in the S_LOCKUP state.
Perhaps the firmware has time to configure something in the microcontroller. Can you try `--connect-under-reset` option?

domingguss, 2021-02-21:
unfortunately, also no success:
```
$ echo "flashing SENIOR (254)..."
flashing SENIOR (254)...
$ st-flash --connect-under-reset --serial 303637354646353735323531373137383637313433343535 write Senior.bin 0x8010000
st-flash 1.6.1
2021-02-21T07:42:20 INFO common.c: F76xxx: 512 KiB SRAM, 2048 KiB flash in at least 2 KiB pages.
file Senior.bin md5 checksum: d1f2e2ec7cd989c560c5cb4465251dfb, stlink checksum: 0x0351df05
2021-02-21T07:42:20 INFO common.c: Attempting to write 524889 (0x80259) bytes to stm32 address: 134283264 (0x8010000)
2021-02-21T07:42:20 WARN common.c: unaligned len 0x80259 -- padding with zero
EraseFlash - Sector:0x2 Size:0x8000 2021-02-21T07:42:20 INFO common.c: Flash page at addr: 0x08010000 erased
EraseFlash - Sector:0x3 Size:0x8000 2021-02-21T07:42:21 INFO common.c: Flash page at addr: 0x08018000 erased
EraseFlash - Sector:0x4 Size:0x20000 2021-02-21T07:42:21 INFO common.c: Flash page at addr: 0x08020000 erased
EraseFlash - Sector:0x5 Size:0x40000 2021-02-21T07:42:23 INFO common.c: Flash page at addr: 0x08040000 erased
EraseFlash - Sector:0x6 Size:0x40000 2021-02-21T07:42:26 INFO common.c: Flash page at addr: 0x08080000 erased
2021-02-21T07:42:26 INFO common.c: Finished erasing 5 pages of 262144 (0x40000) bytes
2021-02-21T07:42:26 INFO common.c: Starting Flash write for F2/F4/F7/L4
2021-02-21T07:42:26 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-02-21T07:42:26 INFO common.c: enabling 32-bit flash writes
2021-02-21T07:42:26 ERROR flash_loader.c: Flash loader run error (R2 0x00000000 R15 0x00000000 DHCSR 0x00010001 DFSR 0x0000000B)
2021-02-21T07:42:26 ERROR common.c: stlink_flash_loader_run(0x8010000) failed! == -1
2021-02-21T07:42:26 INFO common.c: Go to Thumb mode
stlink_fwrite_flash() == -1
```

Ant-ON, 2021-02-24:
@domingguss Your MCU firmware (already flashed) may be using an MPU or something similar. You may try add disable MPU to flash_loader.c. Something this before 'Setup core':
```
stlink_write_reg(sl, 0xE000ED94 /*MPU_CTRL*/, 0);     // disable MPU
```
Complete erasing of the microcontroller before flashing can also help.

Ant-ON, 2021-03-26:
@Nightwalker-87 Most likely it closes by #1113

Nightwalker-87, 2021-03-26:
Closed by #1113.

closed 2021-03-26 by Nightwalker-87