stlink-org/stlink #1096: [feature] Support for STM32U5-series

Nightwalker-87, 2021-03-06:
Currently the U5 family is not supported.
I would like to open a PR to merge WLx5 support based on the work done for the WB family of chips.
The only real issue is that the WL55 and H7 have the same core_id.

    [x] Programmer/board type: STlink/v2, STlink/v2-onboard, STlink/v3
    [x] Operating system: Linux, Mac OS X, Windows (with specific version)
    [x] Stlink tools version and/or git commit hash: v1.6.1
    [x] Stlink commandline tool name: st-info, st-flash, st-util
    [x] Target chip (and optional board): STM32U5-series

tarek-bochkati, 2021-03-06:
what does this mean ? I'm not aware of such thing !
> The only real issue is that the WL55 and H7 have the same core_id

Nightwalker-87, 2021-03-06:
There were unintended leftovers from another ticket which served as a template by mistake. I've removed it.
The U5 series has been announced by ST recently: https://www.st.com/en/microcontrollers-microprocessors/stm32u5-series.html

tarek-bochkati, 2021-03-26:
indeed, I already added support of this device in OpenOCD
http://openocd.zylin.com/#/c/6108/
the flash delta's are in the commit msg ...

Nightwalker-87, 2021-03-26:
Just copied the content to have it here directly:

> Author:     Tarek BOCHKATI <tarek.bouchkati@gmail.com>
> AuthorDate: 2021-03-16 16:10:59 +0100

> flash/stm32l4x: add support of STM32U57x/U58x

> this device flash registers are quite similar to STM32L5 with this changes :
> - flash size is up to 2MB
> - 2MB variants are always dual bank
> - 1MB and 512KB variants could be dual bank (contiguous addressing) depending on DUALBANK bit(21)
> - flash data width is 16 bytes (quad-word)

Nightwalker-87, 2021-05-14:
Related to #1005.

Nightwalker-87, 2022-01-03:
A first contribution to this topic is a (preliminary) chip-id file for the STM32U5 series in branch `chipid_cleanup`.

dhollos, 2022-05-13:
Hi, is there any update on this? The device seems to be still unsupported..

```
Failed to parse flash type or unrecognized flash type
2022-05-13T13:44:04 WARN usb.c: skipping ST device : 0x483:0x5740)
Failed to enter SWD mode
Found 1 stlink programmers
  version:    V3
  serial:     0040001A3438510D34313939
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x000

detected chip_id parametres

# Device Type: unknown
# Reference Manual: RM0000
#
chip_id 0x0
flash_type 0
flash_size_reg 0x0
flash_pagesize 0x0
sram_size 0x0
bootrom_base 0x0
bootrom_size 0x0
option_base 0x0
option_size 0x0
flags 0

  dev-type:   unknown

$ st-flash --version
v1.7.0-186-gc4762e6

```

Nightwalker-87, 2023-01-01:
@tarek-bochkati:
The `.chip` config-file for the STM32U5 series, holding all device-specific configuration, is now complete.
It will be pushed to our `nightly` branch very soon to allow for a review.
We also seem to have most of the library functions for handling this device type.
Many of them appear to be equal or similar to the STM32L5 series.
However we now need additional know-how and contribution to address this issue...

Nightwalker-87, 2023-01-08:
We need people to test the changes and relevant features.

Nightwalker-87, 2023-03-18:
Hello everybody.
Getting in touch with staff on the ST-Microelectronics booth at the Embedded World 2023 resulted in a Nucleo STM32U575ZI-Q board as a give-away. :tada:
Of course one of the first subsequent ideas was to proceed here, now being able to do some testing myself.
I managed to get the board working with the toolset after fixing a wrong setting in the related chip-id file. Finally here is the result I get:

```
st-info --probe
Found 1 stlink programmers
  version:    V3J10
  serial:     001500093656501620363341
  flash:      2097152 (pagesize: 8192)
  sram:       804864
  chipid:     0x482
  dev-type:   STM32U5x5
```

```
st-flash write .../test.bin 0x8000000
st-flash 1.7.0-253-g186e38a
2023-03-19T00:02:52 INFO common.c: STM32U5x5: 786 KiB SRAM, 2048 KiB flash in at least 8 KiB pages.
file .../test.bin md5 checksum: 5761fe3885e985c3021dd8aceb38db9, stlink checksum: 0x00f78dcc
2023-03-19T00:02:52 INFO common_flash.c: Attempting to write 65536 (0x10000) bytes to stm32 address: 134217728 (0x8000000)
-> Flash page at 0x8000000 erased (size: 0x2000)
-> Flash page at 0x8002000 erased (size: 0x2000)
-> Flash page at 0x8004000 erased (size: 0x2000)
-> Flash page at 0x8006000 erased (size: 0x2000)
-> Flash page at 0x8008000 erased (size: 0x2000)
-> Flash page at 0x800a000 erased (size: 0x2000)
-> Flash page at 0x800c000 erased (size: 0x2000)
-> Flash page at 0x800e000 erased (size: 0x2000)

2023-03-19T00:02:52 INFO flashloader.c: Starting Flash write for WB/G0/G4/L5/U5
  8/  8 pages written
2023-03-19T00:02:58 INFO common_flash.c: Starting verification of write complete
2023-03-19T00:02:58 INFO common_flash.c: Flash written and verified! jolly good!
```

So maybe there will also be some progress at #1005 soon, as both MCU families do share some settings.

PR #1300 (2023-04-06): Initial support for STM32 L5 & U5 devices and minor changes
- Initial support for STM32 L5 & U5 devices (Closes #1005) (Closes #1096) (Reference #1247)
- Updated CHANGELOG.md
- Updating windows signal handling to not terminate the application before clean-up.
- Log message improvements for st-flash
- Fix for option bytes on STM32G031G8 (Closes #1194)
closed 2023-04-06 by Nightwalker-87