stlink-org/stlink #1105: [STM32F103CBT6]: Flash verification fails when writing

HexF, 2021-03-16:
Thank you for giving feedback to the stlink project.

**NOTICE: Please read and follow instructions in #906 before submitting a ticket.
This bug report will be deleted without notice when not enough information is provided! So please ensure that all fields are filled out.**

- [x] I made serious effort to avoid creating duplicate or nearly similar issue
This issue extends on issue #1068, which was closed and locked so I can't add to that - hence a new issue.

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to select the check boxes below
and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: STLink v2 Clone and STLink v2
- [x] Operating system an version: Linux (NixOS 20.09)
- [x] **Stlink tools version** and/or git commit hash: v1.6.0 ([nixpkg](https://github.com/NixOS/nixpkgs/blob/master/pkgs/development/tools/misc/stlink/default.nix))
- [x] Stlink commandline tool name: st-flash
- [x] Target chip (and board, if applicable): STM32F103CBT6 (Custom PCB)

```
nix-shell:~/Projects/brainware]$ sudo st-flash --reset write build/src/brainware.bin 0x08000000
st-flash 1.6.0
2021-03-16T18:47:27 INFO common.c: Loading device parameters....
2021-03-16T18:47:27 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2021-03-16T18:47:27 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 1024 bytes
2021-03-16T18:47:27 INFO common.c: Attempting to write 15088 (0x3af0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08003800 erased
2021-03-16T18:47:27 INFO common.c: Finished erasing 15 pages of 1024 (0x400) bytes
2021-03-16T18:47:27 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2021-03-16T18:47:27 INFO flash_loader.c: Successfully loaded flash loader in sram
 15/15 pages written
2021-03-16T18:47:28 INFO common.c: Starting verification of write complete
2021-03-16T18:47:28 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
```

The flashing of the processor seems to fail at this point, as in the aforementioned issue.
One technicality that I have noted is that people are assuming the F103C8 and F103CB chips are the same, and infact they are not.
I have tested the exact same command with a C8 chip, and it worked perfectly fine, but upgrading to a CB chip (which I have had ST confirm are genuine and not counterfeits)

I loaded up a Windows PC with the STLink Utility from ST and it flashed the chip perfectly fine. I was then able to confirm that the flash reads back using `st-flash` on my Linux machine fine, and it matched byte-for-byte the built firmware.
However, if I ignore the verification failing on Linux, and run the same `read` command - I get completely random data.

Any pointers on what to do?

Nightwalker-87, 2021-03-16:
@HexF: Please use the latest version of the toolset (currently 1.6.1) when reporting bugs.

HexF, 2021-03-16:
Using 9956bdeaab034033c3959751460e6cffe464df52 (latest develop commit) I get:
```
nix-shell:~/Projects.local/brainware]$ sudo st-flash --reset write build/src/brainware.bin 0x08000000
st-flash 1.6.1
2021-03-16T22:34:31 WARN usb.c: NRST is not connected
2021-03-16T22:34:31 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 128 KiB flash in at least 1 KiB pages.
2021-03-16T22:34:31 WARN usb.c: NRST is not connected
file build/src/brainware.bin md5 checksum: 53116e8ca4448facb7c4895364041f7, stlink checksum: 0x00179d53
2021-03-16T22:34:31 INFO common.c: Attempting to write 15088 (0x3af0) bytes to stm32 address: 134217728 (0x8000000)
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08000000 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08000400 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08000800 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08000c00 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08001000 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08001400 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08001800 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08001c00 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08002000 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08002400 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08002800 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08002c00 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08003000 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08003400 erased
2021-03-16T22:34:31 INFO common.c: Flash page at addr: 0x08003800 erased
2021-03-16T22:34:31 INFO common.c: Finished erasing 15 pages of 1024 (0x400) bytes
2021-03-16T22:34:31 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-16T22:34:31 INFO flash_loader.c: Successfully loaded flash loader in sram
 15/ 15 pages written
2021-03-16T22:34:32 INFO common.c: Starting verification of write complete
2021-03-16T22:34:32 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
```
Basically the same, just more text

Ant-ON, 2021-03-19:
Can you try https://github.com/Ant-ON/stlink/tree/flash_loader_wait_ms? I'm trying to improve a work of bootloaders...

HexF, 2021-03-22:
@Ant-ON That seems to have fixed it:
```
[nix-shell:~/Projects.local/brainware]$ st-flash --reset write build/src/brainware.bin 0x08000000
st-flash 1.6.1
2021-03-22T18:26:14 WARN usb.c: NRST is not connected
2021-03-22T18:26:14 INFO common.c: F1xx Medium-density: 20 KiB SRAM, 128 KiB flash in at least 1 KiB pages.
2021-03-22T18:26:14 WARN usb.c: NRST is not connected
file build/src/brainware.bin md5 checksum: fe4bf5eb73e1a0fc4bfdcf37ff4a651f, stlink checksum: 0x00179d59
2021-03-22T18:26:14 INFO common.c: Attempting to write 15088 (0x3af0) bytes to stm32 address: 134217728 (0x8000000)
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08000000 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08000400 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08000800 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08000c00 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08001000 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08001400 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08001800 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08001c00 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08002000 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08002400 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08002800 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08002c00 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08003000 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08003400 erased
2021-03-22T18:26:14 INFO common.c: Flash page at addr: 0x08003800 erased
2021-03-22T18:26:14 INFO common.c: Finished erasing 15 pages of 1024 (0x400) bytes
2021-03-22T18:26:14 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-22T18:26:14 INFO flash_loader.c: Successfully loaded flash loader in sram
 15/ 15 pages written
2021-03-22T18:26:15 INFO common.c: Starting verification of write complete
2021-03-22T18:26:15 INFO common.c: Flash written and verified! jolly good!
2021-03-22T18:26:15 WARN usb.c: NRST is not connected
```

Nightwalker-87, 2021-03-22:
@HexF Can you test that in a sequence as well to verify this is not a certain coincidence?

HexF, 2021-03-25:
Yep, running my `ninja -C build flash` worked perfectly fine

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87