stlink-org/stlink #1108: [feature] Support non-cotinuous Intel IHEX files

digitalentity, 2021-03-20:
Not sure if it should be a bug or a feature request. From looking at the code it seems WAI, but from user perspective `st-flash` is horribly broken when dealing with non-cotinuous Intel IHEX files.

**Use-case:**

I'm flashing an Intel IHEX where some of the flash sectors are excluded (Flash Sector 1 to be exact). That sector is used to emulate EEPROM for the firmware.

Correctness of the HEX was verified with `objdump -s`:

```
build/storchfc_1.0.0.hex:     file format ihex

Contents of section .sec1:
 8000000 00000120 4d630108 8f350108 9d350108  ... Mc...5...5..
<snip>
 80001e0 99630108 99630108 99630108 99630108  .c...c...c...c..
 80001f0 00000000 99630108                    .....c..        
Contents of section .sec2:
 8010000 81f00041 02e000bf 83f00043 30b54fea  ...A.......C0.O.
 8010010 41044fea 430594ea 050f08bf 90ea020f  A.O.C...........
<snip>
```

As is clearly seen, there is a gap starting at 0x080001f0 and ending at 0x0800FFFF (encompassing the entire Sector 1).

**Expected behavior:**

The Sector 1 is not erased and it's content survives flashing.

**Actual behavior:**

The `st-flash` happily erases everything between min and max addresses on the ihex file wiping also the emulated EEPROM.

- [X] Programmer/board type: STLink/V2-clone
- [X] Operating system an version: Linux
- [X] **Stlink tools version** and/or git commit hash: v1.6.1
- [X] Stlink commandline tool name: `st-flash`
- [X] Target chip (and board, if applicable): STM32F745

Commandline-Output:

```
st-flash --reset --format ihex write ./build/storchfc_1.0.0.hex
st-flash 1.6.1
2021-03-20T10:59:57 INFO common.c: F76xxx: 512 KiB SRAM, 2048 KiB flash in at least 2 KiB pages.
2021-03-20T10:59:57 INFO common.c: Attempting to write 95524 (0x17524) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Sector:0x0 Size:0x8000 2021-03-20T10:59:57 INFO common.c: Flash page at addr: 0x08000000 erased
EraseFlash - Sector:0x1 Size:0x8000 2021-03-20T10:59:58 INFO common.c: Flash page at addr: 0x08008000 erased
EraseFlash - Sector:0x2 Size:0x8000 2021-03-20T10:59:58 INFO common.c: Flash page at addr: 0x08010000 erased
2021-03-20T10:59:58 INFO common.c: Finished erasing 3 pages of 32768 (0x8000) bytes
2021-03-20T10:59:58 INFO common.c: Starting Flash write for F2/F4/L4
2021-03-20T10:59:58 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 32768
size: 32768
size: 29988
2021-03-20T10:59:59 INFO common.c: Starting verification of write complete
2021-03-20T11:00:00 INFO common.c: Flash written and verified! jolly good!
```

Nightwalker-87, 2021-08-17:
@Ant-ON Can we discuss whether this is a bug or a missing feature?
I'm not so sure about this any more.

Nightwalker-87, 2021-08-22:
Here we should discuss and point out how the observed behaviour could be addressed technically.
Detecting a spare area may not be that trivial...

Nightwalker-87, 2023-09-04:
Closing this feature request due to inactivity.

closed 2023-09-04 by Nightwalker-87