stlink-org/stlink #1115: [STM32L4]: Can't write flash to L451 or L496

lmpynix, 2021-03-26:
- [X] Programmer/board type: STLink/V2 onboard (used both attached and detached)
- [X] Operating system an version: Linux (Manjaro latest)
- [X] **Stlink tools version** and/or git commit hash: v1.6.1-247-gcfff13c
- [X] Stlink commandline tool name: `st-util`
- [X] Target chip (and board, if applicable): STM32L496ZG (NUCLEO-L496ZG), STM32L451VE (Custom)

Commandline-Output:

(With GDB input of `load` when attached to the GDB server, with built ELF file selected)

```
st-util
2021-03-26T11:32:24 INFO common.c: L496x/L4A6x: 256 KiB SRAM, 1024 KiB flash in at least 2 KiB pages.
2021-03-26T11:32:24 WARN usb.c: NRST is not connected
2021-03-26T11:32:24 INFO gdb-server.c: Listening at *:4242...
2021-03-26T11:32:48 WARN usb.c: NRST is not connected
2021-03-26T11:32:48 INFO gdb-server.c: Found 6 hw breakpoint registers
2021-03-26T11:32:48 INFO gdb-server.c: GDB connected.
2021-03-26T11:32:50 WARN usb.c: NRST is not connected
2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: block 08000000 -> e000
2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08000000
EraseFlash - Page:0x0 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08000800
EraseFlash - Page:0x1 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08001000
EraseFlash - Page:0x2 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08001800
EraseFlash - Page:0x3 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08002000
EraseFlash - Page:0x4 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08002800
EraseFlash - Page:0x5 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08003000
EraseFlash - Page:0x6 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08003800
EraseFlash - Page:0x7 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08004000
EraseFlash - Page:0x8 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08004800
EraseFlash - Page:0x9 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08005000
EraseFlash - Page:0xa Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08005800
EraseFlash - Page:0xb Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08006000
EraseFlash - Page:0xc Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08006800
EraseFlash - Page:0xd Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08007000
EraseFlash - Page:0xe Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08007800
EraseFlash - Page:0xf Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08008000
EraseFlash - Page:0x10 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08008800
EraseFlash - Page:0x11 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08009000
EraseFlash - Page:0x12 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 08009800
EraseFlash - Page:0x13 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 0800a000
EraseFlash - Page:0x14 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 0800a800
EraseFlash - Page:0x15 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 0800b000
EraseFlash - Page:0x16 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 0800b800
EraseFlash - Page:0x17 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 0800c000
EraseFlash - Page:0x18 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 0800c800
EraseFlash - Page:0x19 Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 0800d000
EraseFlash - Page:0x1a Size:0x800 2021-03-26T11:32:50 INFO gdb-server.c: flash_erase: page 0800d800
EraseFlash - Page:0x1b Size:0x800 2021-03-26T11:32:50 INFO common.c: Starting Flash write for F2/F4/F7/L4
2021-03-26T11:32:50 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-03-26T11:32:50 INFO gdb-server.c: flash_do: block 08000000 -> e000
2021-03-26T11:32:50 INFO gdb-server.c: flash_do: page 08000000
2021-03-26T11:32:51 ERROR flash_loader.c: Flash loader run error
2021-03-26T11:32:51 WARN flash_loader.c: Loader state: R2 0x0 R15 0x0
2021-03-26T11:32:51 WARN flash_loader.c: MCU state: DHCSR 0x1010009 DFSR 0x1 CFSR 0x0 HFSR 0x0
2021-03-26T11:32:51 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
```

Output from `st-info --probe`:

```
Found 1 stlink programmers
  version:    V2J37S26
  serial:     066EFF544851717867216044
  flash:      1048576 (pagesize: 2048)
  sram:       262144
  chipid:     0x0461
  descr:      L496x/L4A6x
```

Expected/description:

Flash loads successfully.

I have been having trouble for a while trying to load flash on a custom L451-based board using an STLink v2 programmer detached from a Nucleo board.  I saw today that come commits had been made to `develop` that could help, so I tried again with the latest commit, but it's still seemingly not able to write to flash.  I happened to have a brand new, never been used NUCLEO-L496 board handy, and I was able to reproduce the issue on that board as well (command line output above is from the Nucleo, however it is pretty much identical to the custom L451 board's case).  It is interesting that the `NRST is not connected` warning occurs both on the custom board and on the Nucleo, when both have NRST connected and pulled up correctly.

For reference, the custom board is able to be programmed and debugged perfectly fine using the same electrical interface with a Black Magic Probe, so it is unlikely that there's anything wrong with the chip or the application circuit.  I have also heard from a co-worker that they see this same problem with an STM32L0 on a custom board, using a similar OS and programmer.

Reading the flash back using `stlink-gui` seems to indicate that the first few words of the flash are being written, but nothing else.  After a full erase of the chip performed and verified by the official ST flash utility on Windows, then running `st-util` under Linux as above, the entire flash was blank except for the first two words.

Interestingly, rolling back to v1.6.0 also has the same issues, when I distinctly remember being able to use `stlink` for this application before.  If it helps, the packages my up-to-date Manjaro installation has for `libusb` are `libusb-1.0.24-2` and `libusb-compat 0.1.7-1`.

Thank you for maintaining this awesome program!

Ant-ON, 2021-03-26:
@lmpynix In some Lx issues, they wrote that erasing memory does not work correctly. Can you do a full erase in Windows by stlink utility and flash by `st-flash`?

Nightwalker-87, 2021-03-26:
@lmpynix Please try to work with the latest `develop` we are approaching a new release.
In this context it would not make sense to go back to the last version as many issues have been fixed in the meanwhile.

Ant-ON, 2021-03-26:
@lmpynix I got one idea. Can you try patch `src/stlink-lib/flash_loader.c` and retry flash?
```c
fl->buf_addr = fl->loader_addr + (uint32_t)size;
```
to
```c
// Alignment of the buffer position by 8 bytes
fl->buf_addr = fl->loader_addr + (uint32_t)(size + 8 - size%8);
```

Ant-ON, 2021-03-27:
@lmpynix I debugged the L4 flash loader on G4 target. I fixed some problems. Can you try https://github.com/Ant-ON/stlink/tree/flash_loader_fix ?

Nightwalker-87, 2021-04-03:
Ping @lmpynix

erikarn, 2021-04-13:
Hi! I've hit the same issue on:

```
adrianchadd@adrianchadd-mbp bin % ./st-info --probe
Found 1 stlink programmers
  version:    V2J17S4
  serial:     XXXXXX
  flash:      262144 (pagesize: 2048)
  sram:       49152
  chipid:     0x0435
  descr:      L43x/L44x
```

.. Ant-ON 's branch did fix the problem on this L433!

Ant-ON, 2021-04-14:
@erikarn This is already a deprecated branch. I have modified the flash loading process and merged changes. Can you test develop branch (https://github.com/stlink-org/stlink/tree/develop)?

erikarn, 2021-04-14:
hi! It didn't work! I already checked out and built develop.

So something else is missing!

-adrian

Ant-ON, 2021-04-14:
@erikarn Does the flashing via st-flash also not work? Can you provide a log with error?

erikarn, 2021-04-14:
i'll do it tomorrow; iirc it was stuck on the core constantly reporting
alive

>

erikarn, 2021-04-14:
oh! you merged in changes last night!

I just checked out cb551d50f879ec21e3d54c75041bffe506e830e5 and verified it works! thanks!

Nightwalker-87, 2021-04-14:
Fixed by #1126.

closed 2021-04-14 by Nightwalker-87