stlink-org/stlink #1117: STM32F101VG: st-flash reports " ERROR common.c: unsupported flash method, abort"

nbkolchin, 2021-03-27:
- [X] I made serious effort to avoid creating duplicate or nearly similar issue

- [X] Programmer/board type: STLink/V2
- [X] Operating system an version: Linux
- [X] **Stlink tools version** and/or git commit hash: v1.6.1
- [X] Stlink commandline tool name: st-flash
- [X] Target chip (and board, if applicable): STM32F101VG (custom board)

Commandline:

```
st-flash --reset --flash=512k --format=ihex write app.hex
```

Commandline-Output:

```
st-flash 1.6.1
2021-03-27T19:09:43 INFO common.c: F1xx XL-density: 96 KiB SRAM, 1024 KiB flash in at least 2 KiB pages.
Forcing flash size: --flash=0x00080000
2021-03-27T19:09:43 INFO common.c: Attempting to write 415132 (0x6559c) bytes to stm32 address: 134275072 (0x800e000)
2021-03-27T19:09:43 ERROR common.c: unsupported flash method, abort2021-03-27T19:09:43 ERROR common.c: unsupported flash method, ab
ort2021-03-27T19:09:43 ERROR common.c: unsupported flash method, abort2021-03-27T19:09:43 ERROR common.c: unsupported flash method,
 abort2021-03-27T19:09:43 ERROR common.c: unsupported flash method, abort2021-03-27T19:09:43 ERROR common.c: unsupported flash meth
od, abort2021-03-27T19:09:43 ERROR common.c: unsupported flash method, abort2021-03-27T19:09:43 ERROR common.c: unsupported flash m
ethod, abort2021-03-27T19:09:43 ERROR common.c: unsupported flash method, abort2021-03-27T19:09:43 ERROR common.c: unsupported flas
h method, abort2021-03-27T19:09:43 ERROR common.c: unsupported flash method, abort2021-03-27T19:09:43 ERROR common.c: unsupported f
lash method, abort2021-03-27T19:09:43 ERROR common.c: unsupported flash method, abort2021-03-27T19:09:43 ERROR common.c: unsupporte
d flash method, abort2021-03-27T19:09:43 ERROR common.c: unsupported flash method, abort2021-03-27T19:09:43 ERROR common.c: unsuppo
rted flash method, abort2021-03-27T19:09:43 ERROR common.c: unsupported flash method, abort2021-03-27T19:09:43 ERROR common.c: unsu
pported flash method, abort2021-03-27T19:09:43 ERROR common.c: unsupported flash method, abort2021-03-27T19:09:43 ERROR common.c: u
nsupported flash method, abort....
```

Expected/description: Correct operation.

Also, I've noticed that st-util reports incorrect RAM size:

```
$ st-util
st-util
2021-03-27T19:12:46 INFO common.c: F1xx XL-density: 96 KiB SRAM, 1024 KiB flash in at least 2 KiB pages.
```

STM32F101VG has only 80Kib.

P.S. OpenOCD also fails with "error waiting for target flash write algorithm". Original ST Windows utilities work without problems.

Ant-ON, 2021-03-27:
@nbkolchin I fixed typo in code. Can you try https://github.com/Ant-ON/stlink/tree/f1_xl_fix_typo ?

The exact size of SRAM is difficult to obtain. 96 KiB is maximum size for the F1 XL serices.

nbkolchin, 2021-03-27:
It fails with another error after flash erase. 

```
$ ./st-flash --reset --format ihex write nuttx.hex
st-flash 1.6.1-247-gcfff13c
2021-03-27T22:06:30 INFO common.c: F1xx XL-density: 96 KiB SRAM, 1024 KiB flash in at least 2 KiB pages.
2021-03-27T22:06:30 WARN usb.c: NRST is not connected
2021-03-27T22:06:30 INFO common.c: Attempting to write 386212 (0x5e4a4) bytes to stm32 address: 134275072 (0x800e000)
2021-03-27T22:06:31 INFO common.c: Flash page at addr: 0x0800e000 erased
2021-03-27T22:06:31 INFO common.c: Flash page at addr: 0x0800e800 erased
2021-03-27T22:06:31 INFO common.c: Flash page at addr: 0x0800f000 erased
2021-03-27T22:06:31 INFO common.c: Flash page at addr: 0x0800f800 erased
2021-03-27T22:06:31 INFO common.c: Flash page at addr: 0x08010000 erased
2021-03-27T22:06:31 INFO common.c: Flash page at addr: 0x08010800 erased
2021-03-27T22:06:31 INFO common.c: Flash page at addr: 0x08011000 erased
2021-03-27T22:06:31 INFO common.c: Flash page at addr: 0x08011800 erased
2021-03-27T22:06:31 INFO common.c: Flash page at addr: 0x08012000 erased
2021-03-27T22:06:31 INFO common.c: Flash page at addr: 0x08012800 erased
2021-03-27T22:06:31 INFO common.c: Flash page at addr: 0x08013000 erased
...
2021-03-27T22:06:42 INFO common.c: Flash page at addr: 0x0806a800 erased
2021-03-27T22:06:42 INFO common.c: Flash page at addr: 0x0806b000 erased
2021-03-27T22:06:42 INFO common.c: Flash page at addr: 0x0806b800 erased
2021-03-27T22:06:42 INFO common.c: Flash page at addr: 0x0806c000 erased
2021-03-27T22:06:42 INFO common.c: Finished erasing 189 pages of 2048 (0x800) bytes
2021-03-27T22:06:42 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-27T22:06:42 INFO flash_loader.c: Successfully loaded flash loader in sram
154/189 pages written2021-03-27T22:06:54 ERROR flash_loader.c: Flash loader run error
2021-03-27T22:06:54 WARN flash_loader.c: Loader state: R2 0x21000000 R15 0x21000000
2021-03-27T22:06:54 WARN flash_loader.c: MCU state: DHCSR 0x1010009 DFSR 0x0 CFSR 0x0 HFSR 0x0
2021-03-27T22:06:54 ERROR common.c: stlink_flash_loader_run(0x805b000) failed! == -1
stlink_fwrite_flash() == -1
```

ST Link original utilities report :

```
      -------------------------------------------------------------------
                        STM32CubeProgrammer v2.7.0
      -------------------------------------------------------------------

ST-LINK SN  : 55FF68066685485518281367
ST-LINK FW  : V2J29S7
Board       : --
Voltage     : 3.27V
SWD freq    : 4000 KHz
Connect mode: Normal
Reset mode  : Software reset
Device ID   : 0x430
Revision ID : Rev A
Device name : STM32F101/F103 XL-density
Flash size  : 1 MBytes
Device type : MCU
Device CPU  : Cortex-M3
```

Maybe this will help somehow.

Ant-ON, 2021-03-27:
@nbkolchin 154 pages were written without errors ... Then, based on the contents of the registers, something damages the flash loader code in SRAM. PC register points to an address outside the flash loader (0x21000000). The fault registers are empty.

Does the firmware use the DMA to write any values to the SRAM?

Also you can run the commands to check this:
```console
./st-flash --reset erase
./st-flash --reset --format ihex write nuttx.hex
```

nbkolchin, 2021-03-27:
> Does the firmware use the DMA to write any values to the SRAM?

Yes.

I've tried suggested commands:

```
$ ./st-flash --reset erase
st-flash 1.6.1-247-gcfff13c
2021-03-27T22:54:17 INFO common.c: F1xx XL-density: 96 KiB SRAM, 1024 KiB flash in at least 2 KiB pages.
2021-03-27T22:54:17 WARN usb.c: NRST is not connected
Mass erasing
2021-03-27T22:54:17 WARN usb.c: NRST is not connected
$ ./st-flash --reset --format ihex write nuttx.hex
st-flash 1.6.1-247-gcfff13c
2021-03-27T22:54:42 INFO common.c: F1xx XL-density: 96 KiB SRAM, 1024 KiB flash in at least 2 KiB pages.
2021-03-27T22:54:42 WARN usb.c: NRST is not connected
2021-03-27T22:54:43 INFO common.c: Attempting to write 386212 (0x5e4a4) bytes to stm32 address: 134275072 (0x800e000)
2021-03-27T22:54:43 INFO common.c: Flash page at addr: 0x0800e000 erased
2021-03-27T22:54:43 INFO common.c: Flash page at addr: 0x0800e800 erased
2021-03-27T22:54:43 INFO common.c: Flash page at addr: 0x0800f000 erased
2021-03-27T22:54:43 INFO common.c: Flash page at addr: 0x0800f800 erased
2021-03-27T22:54:43 INFO common.c: Flash page at addr: 0x08010000 erased
2021-03-27T22:54:43 INFO common.c: Flash page at addr: 0x08010800 erased
...
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x08016800 erased
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x08017000 erased
2021-03-27T22:54:44 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x08017800 erased
2021-03-27T22:54:44 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x08018000 erased
2021-03-27T22:54:44 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x08018800 erased
2021-03-27T22:54:44 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x08019000 erased
2021-03-27T22:54:44 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x08019800 erased
2021-03-27T22:54:44 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x0801a000 erased
2021-03-27T22:54:44 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x0801a800 erased
2021-03-27T22:54:44 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x0801b000 erased
2021-03-27T22:54:44 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x0801b800 erased
2021-03-27T22:54:44 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x0801c000 erased
2021-03-27T22:54:44 INFO common.c: Flash page at addr: 0x0801c800 erased
...
2021-03-27T22:54:45 INFO common.c: Flash page at addr: 0x08022000 erased
2021-03-27T22:54:45 INFO common.c: Flash page at addr: 0x08022800 erased
2021-03-27T22:54:45 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:45 INFO common.c: Flash page at addr: 0x08023000 erased
2021-03-27T22:54:45 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:45 INFO common.c: Flash page at addr: 0x08023800 erased
2021-03-27T22:54:45 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:45 INFO common.c: Flash page at addr: 0x08024000 erased
2021-03-27T22:54:45 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:45 INFO common.c: Flash page at addr: 0x08024800 erased
2021-03-27T22:54:45 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:45 INFO common.c: Flash page at addr: 0x08025000 erased
2021-03-27T22:54:45 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:45 INFO common.c: Flash page at addr: 0x08025800 erased
2021-03-27T22:54:45 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:45 INFO common.c: Flash page at addr: 0x08026000 erased
2021-03-27T22:54:45 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:45 INFO common.c: Flash page at addr: 0x08026800 erased
2021-03-27T22:54:45 WARN common.c: Failed to unlock flash!
2021-03-27T22:54:45 INFO common.c: Flash page at addr: 0x08027000 erased
2021-03-27T22:54:45 WARN common.c: Failed to unlock flash!
[... 8 lines cut ...]
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x08064800 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x08065000 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x08065800 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x08066000 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x08066800 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x08067000 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x08067800 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x08068000 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x08068800 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x08069000 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x08069800 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x0806a000 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x0806a800 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x0806b000 erased
2021-03-27T22:54:53 INFO common.c: Flash page at addr: 0x0806b800 erased
2021-03-27T22:54:54 INFO common.c: Flash page at addr: 0x0806c000 erased
2021-03-27T22:54:54 INFO common.c: Finished erasing 189 pages of 2048 (0x800) bytes
2021-03-27T22:54:54 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-27T22:54:54 INFO flash_loader.c: Successfully loaded flash loader in sram
 24/189 pages written2021-03-27T22:54:57 ERROR flash_loader.c: Write error
2021-03-27T22:54:57 WARN flash_loader.c: Loader state: R2 0x800 R15 0xFFFFFFFE
2021-03-27T22:54:57 WARN flash_loader.c: MCU state: DHCSR 0x3000B DFSR 0x8 CFSR 0x1 HFSR 0x40000000
2021-03-27T22:54:57 ERROR common.c: stlink_flash_loader_run(0x801a000) failed! == -1
stlink_fwrite_flash() == -1
```

Ant-ON, 2021-03-27:
@nbkolchin And everything goes fail again. Hmm... I have merged F1 XL fixes with minor bug fixes in the flash loader. Can you check erase - write again?
https://github.com/Ant-ON/stlink/commits/flash_loader_fix

nbkolchin, 2021-03-27:
We have progress:

```
...
2021-03-27T23:24:31 INFO common.c: Flash page at addr: 0x08072800 erased
2021-03-27T23:24:31 INFO common.c: Flash page at addr: 0x08073000 erased
2021-03-27T23:24:31 INFO common.c: Finished erasing 203 pages of 2048 (0x800) bytes
2021-03-27T23:24:31 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-27T23:24:31 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-03-27T23:24:31 INFO flash_loader.c: Clear DFSR
203/203 pages written
2021-03-27T23:24:46 INFO common.c: Starting verification of write complete
2021-03-27T23:24:46 ERROR common.c: Verification of flash failed at offset: 0
```

I've tried running st-flash several times and always got this result.

Ant-ON, 2021-03-27:
@nbkolchin I'm glad the problem has been resolved. Thanks for your quick testing! 
I'll do PR later and the fixes will be in the main branch. I'm waiting now for information about the state of the problem with L4...

nbkolchin, 2021-03-27:
The problem wasn't resolved. Verify failed and data wasn't written to flash (board doesn't boot).

Nightwalker-87, 2021-03-27:
No, it's fine. You can continue here. - As you can see I just noticed that there still was this missing bit, before you replied.

Ant-ON, 2021-03-27:
@nbkolchin Oh, I'm already inattentive by the evening. No, need. Can you read data from MCU and run something
```
./st-flash read 0x08000000 32
xxd -l 32 flash.bin
```

Nightwalker-87, 2021-03-27:
@Ant-ON Don't tell that we are spending too much time on this project. :rofl:

nbkolchin, 2021-03-27:
`st-flash read` worked flawlessly and matched real data (that was written with original ST-Link utilities).

Ant-ON, 2021-03-27:
@nbkolchin I'm interested in: What is written instead of the firmware. 
I realized one flaw in the checks of flash loader results. Flash loader doesn't react to possible errors on it run

nbkolchin, 2021-03-27:
I've tried to run `st-flash --reset erase && st-flash --reset --format ihex write nuttx.hex` and suddenly got back flash errors:

```
2021-03-28T00:52:16 INFO common.c: Flash page at addr: 0x08067800 erased
2021-03-28T00:52:16 INFO common.c: Flash page at addr: 0x08068000 erased
2021-03-28T00:52:16 WARN common.c: Failed to unlock flash!
2021-03-28T00:52:17 INFO common.c: Flash page at addr: 0x08068800 erased
2021-03-28T00:52:17 INFO common.c: Flash page at addr: 0x08069000 erased
2021-03-28T00:52:17 INFO common.c: Flash page at addr: 0x08069800 erased
2021-03-28T00:52:17 INFO common.c: Flash page at addr: 0x0806a000 erased
2021-03-28T00:52:17 INFO common.c: Flash page at addr: 0x0806a800 erased
2021-03-28T00:52:17 INFO common.c: Flash page at addr: 0x0806b000 erased
2021-03-28T00:52:17 INFO common.c: Flash page at addr: 0x0806b800 erased
2021-03-28T00:52:17 INFO common.c: Flash page at addr: 0x0806c000 erased
2021-03-28T00:52:17 INFO common.c: Finished erasing 189 pages of 2048 (0x800) bytes
2021-03-28T00:52:17 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-28T00:52:17 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-03-28T00:52:17 INFO flash_loader.c: Clear DFSR
2021-03-28T00:52:17 INFO flash_loader.c: Clear CFSR
2021-03-28T00:52:17 INFO flash_loader.c: Clear HFSR
  8/189 pages written2021-03-28T00:52:18 WARN common.c: Failed to unlock flash!
 11/189 pages written2021-03-28T00:52:18 ERROR flash_loader.c: Write error
2021-03-28T00:52:18 WARN flash_loader.c: Loader state: R2 0x52 R15 0xFFFFFFFE
2021-03-28T00:52:18 WARN flash_loader.c: MCU state: DHCSR 0x3000B DFSR 0x8 CFSR 0x1 HFSR 0x40000000
2021-03-28T00:52:18 ERROR common.c: stlink_flash_loader_run(0x8013800) failed! == -1
stlink_fwrite_flash() == -1
```

I've double checked that source code is from `flash_loader_fix` brach.

Nightwalker-87, 2021-03-27:
You obviously locked your MCU and need to unlock it first before continuing with attempts to flash it.

nbkolchin, 2021-03-27:
No. `STM32_Programmer_CLI` flashes without errors.

```
$ STM32_Programmer_CLI -c port=swd -w nuttx/nuttx/nuttx.hex -v -s
      -------------------------------------------------------------------
                        STM32CubeProgrammer v2.7.0
      -------------------------------------------------------------------

ST-LINK SN  : 55FF68066685485518281367
ST-LINK FW  : V2J29S7
Board       : --
Voltage     : 3.27V
SWD freq    : 4000 KHz
Connect mode: Normal
Reset mode  : Software reset
Device ID   : 0x430
Revision ID : Rev A                                                                                                                   Device name : STM32F101/F103 XL-density
Flash size  : 1 MBytes
Device type : MCU
Device CPU  : Cortex-M3

Memory Programming ...
Opening and parsing file: nuttx.hex
  File          : nuttx.hex
  Size          : 386300 Bytes
  Address       : 0x0800E000

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [28 216]
Download in Progress:
[==================================================] 100%

File download complete
Time elapsed during download operation: 00:00:18.635

Verifying ...

Read progress:
[==================================================] 100%

Download verified successfully

RUNNING Program ...
  Address:      : 0x8000000
Application is running, Please Hold on...
Start operation achieved successfully
```

Ant-ON, 2021-03-29:
@nbkolchin I have added more fixes and improvements to the https://github.com/Ant-ON/stlink/commits/flash_loader_fix . I think it should work now. Can you check again?

nbkolchin, 2021-03-29:
```
2021-03-29T23:03:14 INFO common.c: Flash page at addr: 0x0806a800 erased
2021-03-29T23:03:14 INFO common.c: Flash page at addr: 0x0806b000 erased
2021-03-29T23:03:14 INFO common.c: Flash page at addr: 0x0806b800 erased
2021-03-29T23:03:14 INFO common.c: Flash page at addr: 0x0806c000 erased
2021-03-29T23:03:14 INFO common.c: Finished erasing 189 pages of 2048 (0x800) bytes
2021-03-29T23:03:14 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-29T23:03:14 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-03-29T23:03:14 INFO flash_loader.c: Clear DFSR
  6/189 pages written2021-03-29T23:03:15 ERROR flash_loader.c: Flash loader run error
2021-03-29T23:03:15 WARN flash_loader.c: Loader state: R2 0x61000000 R15 0x61000000
2021-03-29T23:03:15 WARN flash_loader.c: MCU state: DHCSR 0x1010009 DFSR 0x0 CFSR 0x0 HFSR 0x0
2021-03-29T23:03:15 ERROR common.c: stlink_flash_loader_run(0x8011000) failed! == -1
2021-03-29T23:03:15 INFO common.c: Go to Thumb mode
stlink_fwrite_flash() == -1
```

P.S. There was an error in current sources:

```
FAILED: src/stlink-gui/CMakeFiles/stlink-gui-local.dir/gui.c.o
/usr/bin/cc -DSTLINK_HAVE_SYS_MMAN_H -DSTLINK_HAVE_SYS_TIME_H -DSTLINK_HAVE_UNISTD_H -DSTLINK_UI_DIR=\"/mnt/work/company/proposals/ivolgin/telemetron/sources/openocd/stlink/.build/bin\" -I/usr/include/libusb-1.0 -I../inc -Iinc -I../src -I../src/st-flash -I../src/stlink-lib -isystem /usr/include/gtk-3.0 -isystem /usr/include/pango-1.0 -isystem /usr/include/glib-2.0 -isystem /usr/lib/glib-2.0/include -isystem /usr/include/harfbuzz -isystem /usr/include/freetype2 -isystem /usr/include/libpng16 -isystem /usr/include/libmount -isystem /usr/include/blkid -isystem /usr/include/fribidi -isystem /usr/include/cairo -isystem /usr/include/lzo -isystem /usr/include/pixman-1 -isystem /usr/include/gdk-pixbuf-2.0 -isystem /usr/include/gio-unix-2.0 -isystem /usr/include/cloudproviders -isystem /usr/include/atk-1.0 -isystem /usr/include/at-spi2-atk/2.0 -isystem /usr/include/dbus-1.0 -isystem /usr/lib/dbus-1.0/include -isystem /usr/include/at-spi-2.0 -std=gnu11 -std=gnu18 -Wall -Wextra -Wshadow -D_FORTIFY_SOURCE=2 -fstrict-aliasing -Wundef -Wformat -Wformat-security -Wmaybe-uninitialized -Wimplicit-function-declaration -Wredundant-decls -fPIC -O2 -Werror -MD -MT src/stlink-gui/CMakeFiles/stlink-gui-local.dir/gui.c.o -MF src/stlink-gui/CMakeFiles/stlink-gui-local.dir/gui.c.o.d -o src/stlink-gui/CMakeFiles/stlink-gui-local.dir/gui.c.o -c ../src/stlink-gui/gui.c
../src/stlink-gui/gui.c: In function ‘connect_button_cb’:
../src/stlink-gui/gui.c:492:10: error: unused variable ‘i’ [-Werror=unused-variable]
  492 |     gint i;
```

I've commented variable out for correct compilation.

nbkolchin, 2021-03-29:
```
$ ./st-flash --connect-under-reset --reset --format ihex write nuttx.hex
st-flash 1.6.1-250-g7c6448c
2021-03-29T23:53:02 INFO common.c: F1xx XL-density: 96 KiB SRAM, 1024 KiB flash in at least 2 KiB pages.
2021-03-29T23:53:02 WARN usb.c: NRST is not connected
2021-03-29T23:53:02 INFO common.c: Attempting to write 386300 (0x5e4fc) bytes to stm32 address: 134275072 (0x800e000)
2021-03-29T23:53:02 INFO common.c: Flash page at addr: 0x0800e000 erased
2021-03-29T23:53:02 INFO common.c: Flash page at addr: 0x0800e800 erased
2021-03-29T23:53:15 INFO common.c: Flash page at addr: 0x0806b800 erased
2021-03-29T23:53:15 INFO common.c: Flash page at addr: 0x0806c000 erased
2021-03-29T23:53:15 INFO common.c: Finished erasing 189 pages of 2048 (0x800) bytes
2021-03-29T23:53:15 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL
2021-03-29T23:53:15 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-03-29T23:53:15 INFO flash_loader.c: Clear DFSR
  6/189 pages written2021-03-29T23:53:16 ERROR flash_loader.c: Flash loader run error
2021-03-29T23:53:16 WARN flash_loader.c: Loader state: R2 0x61000000 R15 0x61000000
2021-03-29T23:53:16 WARN flash_loader.c: MCU state: DHCSR 0x1010009 DFSR 0x0 CFSR 0x0 HFSR 0x0
2021-03-29T23:53:16 ERROR common.c: stlink_flash_loader_run(0x8011000) failed! == -1
2021-03-29T23:53:16 INFO common.c: Go to Thumb mode
stlink_fwrite_flash() == -1
```

Yes, I have IWDG configured.

nbkolchin, 2021-03-29:
Yes, but this will require some time.

Ant-ON, 2021-03-30:
@nbkolchin I hope we understood each other correctly. I meant IWDG configured by option bytes ("hardware watchdog").

Ant-ON, 2021-04-04:
@nbkolchin I checked the behavior of the application with the IWDG enabled by hardware. It is very similar to the observed situation (interrupting the flashing in random places and the contents of the registers). I added IWDG reset during flashing. Everything should work correctly now. 
If you can, then check out the https://github.com/Ant-ON/stlink/commits/flash_loader_fix branch.

PR #1124 (2021-04-13): Improvements and fixes of the flash loaders, unification of the reset function
STM32L4 flash loader: fixed waiting for busy
STM32F1 flash loader: fixed setting of the PG flag
STM32Lx flash loader: fixed size of the flash loader
Added disabling of the DMA channels and watchdog timers while flash loader are running.
All possible reset options are combined into one function. When using the JTAG reset, the use of the pulse command has been removed. Instead, separate pull-down and pull-up commands are used.
Added target change message to the gdb server.

(Closes #244, Closes #382, Closes #705, Clos [... cut]
closed 2021-04-13 by Nightwalker-87