stlink-org/stlink #1121: STM32F446RE: st-info returns different serial number than stm32cube-programmer

ApiumJacob, 2021-03-31:
**NOTICE: Please read and follow instructions in #906 before submitting a ticket.
This bug report will be deleted without notice when not enough information is provided! So please ensure that all fields are filled out.**

- [X] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers and other contributors to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [X] Programmer/board type: STLINK/V2
- [X] Operating system an version: Linux 5.4.0-70-generic # 78~18.04.1-Ubuntu SMP  x86_64 x86_64 x86_64 GNU/Linux
- [X] **stlink tools version** and/or git commit hash: v1.6.1-69-gbe0157f
- [X] stlink commandline tool name: st-info --probe / st-info --serial
- [X] Target chip (and board, if applicable): STM32F446RE6

Detected problem

st-info returns different results from STM32CUBE-Programmer

Commandline output:

```
$ st-info --serial
363f68064246363849421157
$ st-info --hla-serial
"\x36\x3f\x68\x06\x42\x46\x36\x38\x49\x42\x11\x57"
$ st-info --probe
Found 1 stlink programmers
 serial:     363f68064246363849421157
 hla-serial: "\x36\x3f\x68\x06\x42\x46\x36\x38\x49\x42\x11\x57"
 flash:      4194304 (pagesize: 131072)
 sram:       131072
 chipid:     0x0421
 descr:      F446
```

Expected/description:

[image]

The third nibble is different in the serial number read from st-info vs stm32cube-programmer.

Ant-ON, 2021-04-01:
@ApiumJacob Can you test `develop` branch? It should be fixed by #1106

Nightwalker-87, 2021-04-03:
Ping @ApiumJacob

ApiumJacob, 2021-04-04:
Yeah, I will try it out. Could take some time because I need to figure out
how to build the tools.

Jacob

Nightwalker-87, 2021-04-04:
Take a look at `/doc/tutorial.md` for further help on this topic.

Nightwalker-87, 2021-04-11:
Ping @ApiumJacob

Nightwalker-87, 2021-04-14:
Related to #417 and #494.

tarek-bochkati, 2021-04-15:
this "could" be fixed in develop by #1106
we need @ApiumJacob to recheck using develop branch
(I can't help on how to build, in linux it compiles w/o any effort.)

Nightwalker-87, 2021-04-23:
I don't have this board type either. Anyone else? @Ant-ON @slyshykO maybe?

Nightwalker-87, 2021-04-24:
There are strong indications that this issue is fixed in #1106 along with no further response from the original author.
Against this background we can safely close this issue as resolved.

closed 2021-04-24 by Nightwalker-87