stlink-org/stlink #1125: [STM32L072]: Different st-util behaviour v1.6.1 vs develop:latest

vkleban, 2021-04-12:
Hi guys, 

- [ ] Operating system an version: macOS 11.2.3 
- [ ] Target chip (and board, if applicable): STM32L072
- [ ] Programmer/board type: B-L072Z-LRWAN1 (STLINK /V2-onboard)

I've installed `st-util` (v.1.6.1) on my macOS 11.2.3 using `homebrew` and it works fine for me except the `--freq` parameter that is not supported.

Here is a normal output:
```
2021-04-12T18:55:35 DEBUG common.c: *** looking up stlink version
2021-04-12T18:55:35 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-04-12T18:55:35 DEBUG common.c: stlink pid     = 0x374b
2021-04-12T18:55:35 DEBUG common.c: stlink version = 0x2
2021-04-12T18:55:35 DEBUG common.c: jtag version   = 0x1c
2021-04-12T18:55:35 DEBUG common.c: swim version   = 0x12
2021-04-12T18:55:35 DEBUG common.c: *** looking up stlink version
2021-04-12T18:55:35 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-04-12T18:55:35 DEBUG common.c: stlink pid     = 0x374b
2021-04-12T18:55:35 DEBUG common.c: stlink version = 0x2
2021-04-12T18:55:35 DEBUG common.c: jtag version   = 0x1c
2021-04-12T18:55:35 DEBUG common.c: swim version   = 0x12
2021-04-12T18:55:35 DEBUG common.c: stlink current mode: mass
2021-04-12T18:55:35 DEBUG usb.c: JTAG/SWD freq set to 0
2021-04-12T18:55:35 DEBUG common.c: *** set_swdclk ***
2021-04-12T18:55:35 DEBUG common.c: stlink current mode: mass
2021-04-12T18:55:35 DEBUG common.c: *** stlink_enter_swd_mode ***
2021-04-12T18:55:35 DEBUG common.c: *** stlink_jtag_reset ***
2021-04-12T18:55:35 DEBUG common.c: *** stlink_reset ***
2021-04-12T18:55:35 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2021-04-12T18:55:35 DEBUG common.c: Loading device parameters....
2021-04-12T18:55:35 DEBUG common.c: *** stlink_core_id ***
2021-04-12T18:55:35 DEBUG common.c: core_id = 0x0bc11477
2021-04-12T18:55:35 DEBUG common.c: *** stlink_read_debug32 0 is 0xe0042000
2021-04-12T18:55:35 DEBUG common.c: *** stlink_read_debug32 20086447 is 0x40015800
2021-04-12T18:55:35 DEBUG common.c: *** stlink_read_debug32 39000c0 is 0x1ff8007c
2021-04-12T18:55:35 INFO common.c: L0xx Category 5: 20 KiB SRAM, 192 KiB flash in at least 128 byte pages.
2021-04-12T18:55:35 DEBUG common.c: *** stlink_reset ***
2021-04-12T18:55:35 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2021-04-12T18:55:35 DEBUG gdb-server.c: Chip ID is 0x00000447, Core ID is 0xbc11477.
2021-04-12T18:55:35 INFO gdb-server.c: Listening at *:4242...
```

In order to make `--freq` available I've decided to build from sources (latest develop branch).
Unfortunately, built binaries demonstrate different behaviour from v.1.6.1 ones and do not work.

```
2021-04-12T18:56:11 DEBUG common.c: *** looking up stlink version
2021-04-12T18:56:11 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-04-12T18:56:11 DEBUG common.c: stlink pid     = 0x374b
2021-04-12T18:56:11 DEBUG common.c: stlink version = 0x2
2021-04-12T18:56:11 DEBUG common.c: jtag version   = 0x1c
2021-04-12T18:56:11 DEBUG common.c: swim version   = 0x12
2021-04-12T18:56:11 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-04-12T18:56:11 DEBUG usb.c: JTAG/SWD freq set to 0
2021-04-12T18:56:11 DEBUG common.c: *** set_swdclk ***
2021-04-12T18:56:11 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-04-12T18:56:11 DEBUG common.c: *** stlink_jtag_reset ***
2021-04-12T18:56:11 DEBUG common.c: *** stlink_reset ***
2021-04-12T18:56:12 DEBUG common.c: Loading device parameters....
2021-04-12T18:56:12 DEBUG common.c: *** stlink_core_id ***
2021-04-12T18:56:12 DEBUG common.c: core_id = 0x0bc11477
2021-04-12T18:56:12 DEBUG common.c: *** stlink_read_debug32 0x410cc601 at 0xe000ed00
2021-04-12T18:56:12 DEBUG common.c: *** stlink_read_debug32 0000000000 at 0xe0042000
2021-04-12T18:56:12 ERROR common.c: Could not find chip id!
2021-04-12T18:56:12 WARN common.c: Invalid flash type, please check device declaration
2021-04-12T18:56:12 ERROR gdb-server.c: Unsupported Target (Chip ID is 0000000000, Core ID is 0x0bc11477).
```

I've checked it multiple times and can reproduce.
Am I doing something wrong?

Ant-ON, 2021-04-12:
@vkleban I missed the Cortex-M0+ core when implementing the chipid detection. Can you check https://github.com/Ant-ON/stlink/tree/cm0p_fix ?

vkleban, 2021-04-13:
@Ant-ON I confirm that the issue is resolved now. Thank you!

PR #1126 (2021-04-13): Fixed reading of chip ID on Cortex-M0+ core
The Сortex-M0+ core was missing in the chip ID reading algorithm

(Closes #1125)
closed 2021-04-13 by Nightwalker-87