stlink-org/stlink #1128: [STM32L471]: Cannot set option byte

seeseemelk, 2021-04-15:
When attempting to unset the `BFB2` bit from `FLASH_OPTCR` register, I get the following error:
```
$ st-flash --area=option read
st-flash 1.6.1-109-g781557d
2021-04-15T15:00:02 INFO common.c: L4xx: 96 KiB SRAM, 512 KiB flash in at least 2 KiB pages.
fffff8aa
$ st-flash --area=option write 0xFFEFF8AA
st-flash 1.6.1-109-g781557d
2021-04-15T15:00:13 INFO common.c: L4xx: 96 KiB SRAM, 512 KiB flash in at least 2 KiB pages.
attempting to set option byte to 0, abort.
```
It seems that the st-flash utility is not quite correctly parsing its command line arguments, preventing the modification of special registers.

seeseemelk, 2021-04-15:
This issue has been fixed by eff18d81d9c64ef9eaf4972d800de09be13ff141, but as of writing this isn't yet in a released version which is why I encountered it.

closed 2021-04-15 by seeseemelk

Nightwalker-87, 2021-04-15:
@seeseemelk: Please follow #906 before opening new tickets in future.

Nightwalker-87, 2021-04-15:
Fixed by #1102.