stlink-org/stlink #1134: [STM32F407]: STLINK-V3 fails to detect target after FW update

wavexx, 2021-04-26:
- [x] I made serious effort to avoid creating duplicate or nearly similar issue
- [x] Programmer/board type: STLINK-V3SET
- [x] Operating system an version: Linux (debian unstable)
- [x] **stlink tools version** and/or git commit hash: v1.7.0/eeaef981a5dcf6bec6d9b6cada398abcf34b8737 (current master)
- [x] stlink commandline tool name: st-info --probe
- [x] Target chip (and board, if applicable): F4xx (device id 0x413)

I'm using an STLINK-V3SET connected to an STM32F407. I had the "brilliant" idea to update the FW that shipped on the programmer from "V3J5S1" to "V3J7M2B4S1" (the latest version available on st.com). After the update, I can still use STM32CubeProgrammer, but st-info (or any other tool) fails to detect the target. I tried other FW versions as available on st, such as V3J5M2B3S1 which seems the closest to the original, but it still fails.

Output from the old FW version:

```
Found 1 stlink programmers
  version:    V3J5S1
  serial:     0042003C3038510334333935
  flash:      1048576 (pagesize: 16384)
  sram:       196608
  chipid:     0x0413
  descr:      F4xx
```

after the update:

```
% st-info --probe
Found 1 stlink programmers
  version:    V0
  serial:     004C00293438510B34313939
  flash:      0 (pagesize: 0)
  sram:       24
  chipid:     0x03e8
```

(although here chipid and sram are frequently reported as 0).

st-util -v99 output using FW V3J5S1:
```
st-util
2021-04-27T00:12:39 DEBUG common.c: *** looking up stlink version
2021-04-27T00:12:39 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-04-27T00:12:39 DEBUG common.c: stlink pid     = 0x374f
2021-04-27T00:12:39 DEBUG common.c: stlink version = 0x3
2021-04-27T00:12:39 DEBUG common.c: jtag version   = 0x7
2021-04-27T00:12:39 DEBUG common.c: swim version   = 0x1
2021-04-27T00:12:39 DEBUG common.c: *** looking up stlink version
2021-04-27T00:12:39 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-04-27T00:12:39 DEBUG common.c: stlink pid     = 0x374f
2021-04-27T00:12:39 DEBUG common.c: stlink version = 0x3
2021-04-27T00:12:39 DEBUG common.c: jtag version   = 0x7
2021-04-27T00:12:39 DEBUG common.c: swim version   = 0x1
2021-04-27T00:12:39 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-04-27T00:12:39 DEBUG usb.c: JTAG/SWD freq set to 0
2021-04-27T00:12:39 DEBUG common.c: *** set_swdclk ***
2021-04-27T00:12:39 INFO usb.c: Unable to match requested speed 1800 kHz, using 1000 kHz
2021-04-27T00:12:39 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-04-27T00:12:39 DEBUG common.c: *** stlink_jtag_reset ***
2021-04-27T00:12:39 DEBUG common.c: *** stlink_reset ***
2021-04-27T00:12:39 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2021-04-27T00:12:39 DEBUG common.c: Loading device parameters....
2021-04-27T00:12:39 DEBUG common.c: *** stlink_core_id ***
2021-04-27T00:12:39 DEBUG common.c: core_id = 0x2ba01477
2021-04-27T00:12:39 DEBUG common.c: *** stlink_read_debug32 0 is 0xe0042000
2021-04-27T00:12:39 DEBUG common.c: *** stlink_read_debug32 0 is 0x40015800
2021-04-27T01:13:58 WARN common.c: Invalid flash type, please check device declaration
2021-04-27T01:13:58 ERROR gdb-server.c: Unsupported Target (Chip ID is 0000000000, Core ID is 0x2ba01477).
```

same using  V3J7M2B4S1:
```
st-util
2021-04-26T18:15:37 DEBUG common.c: *** looking up stlink version
2021-04-26T18:15:37 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-04-26T18:15:37 DEBUG common.c: stlink pid     = 0x374f
2021-04-26T18:15:37 DEBUG common.c: stlink version = 0x3
2021-04-26T18:15:37 DEBUG common.c: jtag version   = 0x5
2021-04-26T18:15:37 DEBUG common.c: swim version   = 0x1
2021-04-26T18:15:37 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-04-26T18:15:37 DEBUG usb.c: JTAG/SWD freq set to 0
2021-04-26T18:15:37 DEBUG common.c: *** set_swdclk ***
2021-04-26T18:15:37 INFO usb.c: Unable to match requested speed 1800 kHz, using 1000 kHz
2021-04-26T18:15:37 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-04-26T18:15:37 DEBUG common.c: *** stlink_reset ***
2021-04-26T18:15:37 DEBUG common.c: *** stlink_read_debug32 0x01010000 at 0xe000edf0
2021-04-26T18:15:37 DEBUG common.c: *** stlink_jtag_reset ***
2021-04-26T18:15:37 DEBUG common.c: *** stlink_jtag_reset ***
2021-04-26T18:15:37 DEBUG common.c: *** stlink_read_debug32 0x03010000 at 0xe000edf0
2021-04-26T18:15:37 DEBUG common.c: *** stlink_read_debug32 0x01010000 at 0xe000edf0
2021-04-26T18:15:37 DEBUG common.c: Loading device parameters....
2021-04-26T18:15:37 DEBUG common.c: *** stlink_core_id ***
2021-04-26T18:15:37 DEBUG common.c: core_id = 0x2ba01477
2021-04-26T18:15:37 DEBUG common.c: *** stlink_read_debug32 0x410fc241 at 0xe000ed00
2021-04-26T18:15:37 DEBUG common.c: *** stlink_read_debug32 0x100f6413 at 0xe0042000
2021-04-26T18:15:37 DEBUG common.c: *** stlink_read_debug32 0x0400c000 at 0x1fff7a20
2021-04-26T18:15:37 INFO common.c: F4xx: 192 KiB SRAM, 1024 KiB flash in at least 16 KiB pages.
2021-04-26T18:15:37 DEBUG gdb-server.c: Chip ID is 0x00000413, Core ID is 0x2ba01477.
2021-04-26T18:15:37 INFO gdb-server.c: Listening at *:4242...
```

It it possible to make it work also using the latest FW version?

Alternatively, is it possible to somehow dump the original FW from another V3?

wavexx, 2021-04-26:
To add some additional info, openocd 0.11.0 doesn't seem to have any issue with FW V3J7M2B4S1 (target is detected, I can flash/debug without issues). Please let me know if I can help in any way.

Ant-ON, 2021-04-27:
@wavexx I have updated my st-link and it is correctly detected and works. 
I look at the issue logs. It also seems strange to me that stlink with new FW is detected correct in the log:
```bash
...
2021-04-27T01:23:38 DEBUG common.c: stlink version = **0x3**
2021-04-27T01:23:38 DEBUG common.c: jtag version   = **0x7**
2021-04-27T01:23:38 DEBUG common.c: swim version   = **0x1**
...
```
But in the probe version is missing
```bash
  version:    **V0**
```
Are there any errors in the logs?

wavexx, 2021-04-28:
Ok, that's good to know at least, so that might indicate some other odd issue on my side. Not with the device at this moment. Any more detailed info into where to look? By logs you mean dmesg or something else?

Ant-ON, 2021-04-30:
@wavexx I mean the output of the application to the console

wavexx, 2021-04-30:
Turn out, this was probably a kernel issue involving the acm driver (https://bugzilla.kernel.org/show_bug.cgi?id=212725) as I happened to update during a reboot... I've since rebuilt the kernel with the relevant patch as I noticed that I also had issues with other MCUs.

With that the current version of st-info works as intended also with the  V3J7M2B4S1 FW (reported simply as V3J7S1 now). I cannot explain why STM32CubeProgrammer or openocd didn't seem to be affected in the same though.

Nightwalker-87, 2021-04-30:
Good to see that the problem appears to be resolved.
@wavexx: Is there anything else you came across that is strongly related to this issue?

wavexx, 2021-04-30:
No. Unless you might want me to dig deeper for some reason, I'd close this.

Nightwalker-87, 2021-04-30:
The provided reference should be fine. :wink: 
As the observed behaviour obviously goes back to a (local) kernel issue, I'm closing this now as resolved.

closed 2021-04-30 by Nightwalker-87