stlink-org/stlink #1138: [STM32L072CZ]: issue programming

wolsty7, 2021-05-07:
- Programmer/board type: STLINK V3 set
- Operating system an version: raspberry pi OS buster
- **stlink tools version** and/or git commit hash: 1.7.0-4-gdb9219b
- stlink commandline tool name: st-flash
- Target chip (and board, if applicable): STM32L072CZ (CMWX1ZZABZ-091 module, B-L072Z-LRWAN1 board (and custom board)

Commandline output:

```
pi@raspberrypi:~ $ st-flash --debug --reset --freq=8M write /home/pi/Downloads/v0.bin 0x8000000
st-flash 1.7.0-4-gdb9219b
2021-05-07T12:15:09 DEBUG common.c: *** looking up stlink version
2021-05-07T12:15:09 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-05-07T12:15:09 DEBUG common.c: stlink pid     = 0x374f
2021-05-07T12:15:09 DEBUG common.c: stlink version = 0x3
2021-05-07T12:15:09 DEBUG common.c: jtag version   = 0x7
2021-05-07T12:15:09 DEBUG common.c: swim version   = 0x1
2021-05-07T12:15:09 DEBUG common.c: stlink current mode: mass
2021-05-07T12:15:09 DEBUG usb.c: JTAG/SWD freq set to 8000
2021-05-07T12:15:09 DEBUG common.c: *** set_swdclk ***
2021-05-07T12:15:09 DEBUG common.c: stlink current mode: mass
2021-05-07T12:15:09 DEBUG common.c: *** stlink_enter_swd_mode ***
2021-05-07T12:15:09 DEBUG common.c: Loading device parameters....
2021-05-07T12:15:09 DEBUG common.c: *** stlink_core_id ***
2021-05-07T12:15:09 DEBUG common.c: core_id = 0x0bc11477
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x410cc601 at 0xe000ed00
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x20086447 at 0x40015800
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x038f00c0 at 0x1ff8007c
2021-05-07T12:15:09 INFO common.c: L0xx Category 5: 20 KiB SRAM, 192 KiB flash in at least 128 byte pages.
2021-05-07T12:15:09 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-05-07T12:15:09 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-05-07T12:15:09 DEBUG common.c: *** stlink_force_debug_mode ***
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0000000000 at 0x40015808
2021-05-07T12:15:09 DEBUG common.c: *** stlink_write_debug32 0x00001800 to 0x40015808
2021-05-07T12:15:09 DEBUG common.c: *** stlink_status ***
2021-05-07T12:15:09 DEBUG usb.c: core status: 01030003
2021-05-07T12:15:09 DEBUG common.c:   core status: halted
file /home/pi/Downloads/v0.bin md5 checksum: 7059ebcb3e9bde16901ce938fc6498e, stlink checksum: 0x00b99cac
2021-05-07T12:15:09 INFO common.c: Attempting to write 139392 (0x22080) bytes to stm32 address: 134217728 (0x8000000)
2021-05-07T12:15:09 DEBUG common.c: *** stlink_core_id ***
2021-05-07T12:15:09 DEBUG common.c: core_id = 0x0bc11477
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x0000010c at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_write_debug32 0x00013f00 to 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000007 at 0x40022004
2021-05-07T12:15:09 DEBUG common.c: *** stlink_write_debug32 0x89abcdef to 0x4002200c
2021-05-07T12:15:09 DEBUG common.c: *** stlink_write_debug32 0x02030405 to 0x4002200c
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000006 at 0x40022004
2021-05-07T12:15:09 DEBUG common.c: *** stlink_write_debug32 0x8c9daebf to 0x40022010
2021-05-07T12:15:09 DEBUG common.c: *** stlink_write_debug32 0x13141516 to 0x40022010
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000004 at 0x40022004
2021-05-07T12:15:09 DEBUG common.c: *** stlink_write_debug32 0x0000020c to 0x40022004
2021-05-07T12:15:09 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0x08000000
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x0000000e at 0x40022018
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x0000020c at 0x40022004
2021-05-07T12:15:09 DEBUG common.c: *** stlink_write_debug32 0x0000020f to 0x40022004
2021-05-07T12:15:09 DEBUG common.c: *** stlink_read_debug32 0x0000000e at 0x40022018
[... 297 lines cut ...]
2021-05-07T12:15:34 DEBUG common.c: data_len = 2 0x2
 80 00
2021-05-07T12:15:34 DEBUG common.c: *** stlink_run ***
2021-05-07T12:15:34 DEBUG common.c: *** stlink_read_reg
2021-05-07T12:15:34 DEBUG common.c:  (16) ***
2021-05-07T12:15:34 DEBUG common.c: data_len = 8 0x8
 80 00 00 00 00 00 00 f0
2021-05-07T12:15:34 DEBUG usb.c: r_idx (16) = 0xf0000000
2021-05-07T12:15:34 INFO common.c: Go to Thumb mode
2021-05-07T12:15:34 DEBUG common.c: *** stlink_write_reg
2021-05-07T12:15:34 DEBUG common.c: data_len = 2 0x2
 80 00
stlink_fwrite_flash() == -1
2021-05-07T12:15:34 DEBUG common.c: *** stlink_exit_debug_mode ***
2021-05-07T12:15:34 DEBUG common.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2021-05-07T12:15:34 DEBUG common.c: *** stlink_close ***
```

Expected/description:

`
Programs STM32L151 based board correctly.
On the STM32L072 it gets stuck after the erase section and doesn't continue till the NRST button is pressed.
STM32L072 board programs correctly with correctly in stm32cube programmer on windows.
`

Ant-ON, 2021-05-22:
@wolsty7 Why do you additionally press the reset button during programming? It is not necessary.
The DHCSR status indicates an unrecoverable exception occurred (S_LOCKUP bit). I have no idea why this is happening yet.

wolsty7, 2021-05-24:
See my updated description (I inserted the where I pressed NRST in the wrong section)

It gets stuck on the line:

> stlink_read_debug32 0x0000000f at 0x40022018

until I press nrst button at which point the program ends.

Ant-ON, 2021-05-24:
@wolsty7 As I understand it, the blank target may have problems with writing. Writing flash by word (in use in the 1.7.0 release) can take a long time. I added half page write fallback for stm32L0/L1. 
Can you check out https://github.com/Ant-ON/stlink/tree/l0_soft_loader_fix branch?

wolsty7, 2021-05-25:
that seems to do better but it fails at the end, at least it doesn't get stuck like 1.7.0

`
pi@raspberrypi:~/git/stlinktest/stlink $ st-flash --debug --connect-under-reset --freq=8M write /home/pi/Downloads/v0.bin 0x8000000
st-flash 1.6.1-326-gb93f8b2
2021-05-25T19:56:39 DEBUG common.c: *** looking up stlink version
2021-05-25T19:56:39 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-05-25T19:56:39 DEBUG common.c: stlink pid     = 0x374f
2021-05-25T19:56:39 DEBUG common.c: stlink version = 0x3
2021-05-25T19:56:39 DEBUG common.c: jtag version   = 0x7
2021-05-25T19:56:39 DEBUG common.c: swim version   = 0x1
2021-05-25T19:56:39 DEBUG common.c: stlink current mode: mass
2021-05-25T19:56:39 DEBUG usb.c: JTAG/SWD freq set to 8000
2021-05-25T19:56:39 DEBUG common.c: *** stlink_enter_swd_mode ***
2021-05-25T19:56:39 DEBUG common.c: *** stlink_jtag_reset 0 ***
2021-05-25T19:56:39 DEBUG common.c: *** stlink_jtag_reset 1 ***
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x03030003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_soft_reset (halt) ***
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0xa05f0003 to 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0x01000501 to 0xe000edfc
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0x00000008 to 0xe000ed30
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x03030003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0xe000ed30
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0x0000001f to 0xe000ed30
2021-05-25T19:56:39 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-05-25T19:56:39 DEBUG common.c: Loading device parameters....
2021-05-25T19:56:39 DEBUG common.c: *** stlink_core_id ***
2021-05-25T19:56:39 DEBUG common.c: core_id = 0x0bc11477
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x410cc601 at 0xe000ed00
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x20086447 at 0x40015800
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x038f00c0 at 0x1ff8007c
2021-05-25T19:56:39 INFO common.c: L0xx Category 5: 20 KiB SRAM, 192 KiB flash in at least 128 byte pages.
2021-05-25T19:56:39 DEBUG common.c: *** stlink_force_debug_mode ***
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0000000000 at 0x40015808
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0x00001800 to 0x40015808
2021-05-25T19:56:39 DEBUG common.c: *** stlink_status ***
2021-05-25T19:56:39 DEBUG usb.c: core status: 00030003
2021-05-25T19:56:39 DEBUG common.c:   core status: halted
file /home/pi/Downloads/v0.bin md5 checksum: 7059ebcb3e9bde16901ce938fc6498e, stlink checksum: 0x00b99cac
2021-05-25T19:56:39 INFO common.c: Attempting to write 139392 (0x22080) bytes to stm32 address: 134217728 (0x8000000)
2021-05-25T19:56:39 DEBUG common.c: *** stlink_core_id ***
2021-05-25T19:56:39 DEBUG common.c: core_id = 0x0bc11477
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x0000000c at 0x40022018
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0x00013f00 to 0x40022018
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00000007 at 0x40022004
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0x89abcdef to 0x4002200c
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0x02030405 to 0x4002200c
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00000006 at 0x40022004
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0x8c9daebf to 0x40022010
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0x13141516 to 0x40022010
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00000004 at 0x40022004
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0x0000020c to 0x40022004
2021-05-25T19:56:39 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0x08000000
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-25T19:56:39 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
[... 74 lines cut ...]
2021-05-25T19:56:57 DEBUG common.c: *** stlink_write_mem8 ***
[... cut ...]
2021-05-25T19:56:57 DEBUG common.c:  (16) ***
2021-05-25T19:56:57 DEBUG common.c: data_len = 8 0x8
 80 00 00 00 00 00 00 f0
2021-05-25T19:56:57 DEBUG usb.c: r_idx (16) = 0xf0000000
2021-05-25T19:56:57 INFO common.c: Go to Thumb mode
2021-05-25T19:56:57 DEBUG common.c: *** stlink_write_reg
stlink_fwrite_flash() == -1
2021-05-25T19:56:57 DEBUG common.c: *** stlink_exit_debug_mode ***
2021-05-25T19:56:57 DEBUG common.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2021-05-25T19:56:57 DEBUG common.c: *** stlink_close ***
`

Ant-ON, 2021-05-25:
@wolsty7 Unfortunately, everything ends with an error. I fixed the branch to use 32 bit writes. Can you test branch again?

wolsty7, 2021-05-25:
Success, thank you @Ant-ON 

`
pi@raspberrypi:~/git/stlinktest/stlink $ st-flash --debug --connect-under-reset --freq=8M write /home/pi/Downloads/v0.bin 0x8000000
st-flash 1.6.1-327-g293db13
2021-05-26T09:45:32 DEBUG common.c: *** looking up stlink version
2021-05-26T09:45:32 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-05-26T09:45:32 DEBUG common.c: stlink pid     = 0x374f
2021-05-26T09:45:32 DEBUG common.c: stlink version = 0x3
2021-05-26T09:45:32 DEBUG common.c: jtag version   = 0x7
2021-05-26T09:45:32 DEBUG common.c: swim version   = 0x1
2021-05-26T09:45:32 DEBUG common.c: stlink current mode: mass
2021-05-26T09:45:32 DEBUG usb.c: JTAG/SWD freq set to 8000
2021-05-26T09:45:32 DEBUG common.c: *** stlink_enter_swd_mode ***
2021-05-26T09:45:32 DEBUG common.c: *** stlink_jtag_reset 0 ***
2021-05-26T09:45:32 DEBUG common.c: *** stlink_jtag_reset 1 ***
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x03030003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_soft_reset (halt) ***
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0xa05f0003 to 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0x01000501 to 0xe000edfc
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0x00000008 to 0xe000ed30
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x02000003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x03030003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0xe000ed30
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0x0000001f to 0xe000ed30
2021-05-26T09:45:32 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-05-26T09:45:32 DEBUG common.c: Loading device parameters....
2021-05-26T09:45:32 DEBUG common.c: *** stlink_core_id ***
2021-05-26T09:45:32 DEBUG common.c: core_id = 0x0bc11477
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x410cc601 at 0xe000ed00
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x20086447 at 0x40015800
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x038f00c0 at 0x1ff8007c
2021-05-26T09:45:32 INFO common.c: L0xx Category 5: 20 KiB SRAM, 192 KiB flash in at least 128 byte pages.
2021-05-26T09:45:32 DEBUG common.c: *** stlink_force_debug_mode ***
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0000000000 at 0x40015808
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0x00001800 to 0x40015808
2021-05-26T09:45:32 DEBUG common.c: *** stlink_status ***
2021-05-26T09:45:32 DEBUG usb.c: core status: 00030003
2021-05-26T09:45:32 DEBUG common.c:   core status: halted
file /home/pi/Downloads/v0.bin md5 checksum: 7059ebcb3e9bde16901ce938fc6498e, stlink checksum: 0x00b99cac
2021-05-26T09:45:32 INFO common.c: Attempting to write 139392 (0x22080) bytes to stm32 address: 134217728 (0x8000000)
2021-05-26T09:45:32 DEBUG common.c: *** stlink_core_id ***
2021-05-26T09:45:32 DEBUG common.c: core_id = 0x0bc11477
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x0000000c at 0x40022018
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0x00013f00 to 0x40022018
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x00000007 at 0x40022004
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0x89abcdef to 0x4002200c
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0x02030405 to 0x4002200c
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x00000006 at 0x40022004
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0x8c9daebf to 0x40022010
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0x13141516 to 0x40022010
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x00000004 at 0x40022004
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0x0000020c to 0x40022004
2021-05-26T09:45:32 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0x08000000
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
2021-05-26T09:45:32 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40022018
[... 157 lines cut ...]
[... cut ...]
2021-05-26T09:45:56 DEBUG common.c:  (16) ***
2021-05-26T09:45:56 DEBUG common.c: data_len = 8 0x8
 80 00 02 08 03 00 00 30
2021-05-26T09:45:56 DEBUG usb.c: r_idx (16) = 0x30000003
2021-05-26T09:45:56 INFO common.c: Go to Thumb mode
2021-05-26T09:45:56 DEBUG common.c: *** stlink_write_reg
2021-05-26T09:45:56 DEBUG common.c: *** stlink_exit_debug_mode ***
2021-05-26T09:45:56 DEBUG common.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2021-05-26T09:45:56 DEBUG common.c: *** stlink_close ***
`

Ant-ON, 2021-05-26:
@wolsty7 I'm glad I did it. Could you still check the re-flashing of the STM32L072? In this case, the flash loader should work. Or you may check the flashing of the STM32L151.
Unfortunately I do not have L-series MCU and cannot be sure that I have not broken anything.

Nightwalker-87, 2021-05-26:
Good work @Ant-ON ! :+1: 
Yeah, should there be any chance for verification on the L1 to rule out potential regressions, we should make use of it FMPOV.

wolsty7, 2021-05-27:
no luck on the stm32l151zdt6... i also tried --reset and nothing, this board worked for 1.7.0

`
pi@raspberrypi:~ $ st-flash --debug --connect-under-reset --freq=8M write /home/pi/Downloads/OBC_GEN_CPP.bin 0x8000000
st-flash 1.6.1-327-g293db13
2021-05-27T22:10:42 DEBUG common.c: *** looking up stlink version
2021-05-27T22:10:42 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-05-27T22:10:42 DEBUG common.c: stlink pid     = 0x374f
2021-05-27T22:10:42 DEBUG common.c: stlink version = 0x3
2021-05-27T22:10:42 DEBUG common.c: jtag version   = 0x7
2021-05-27T22:10:42 DEBUG common.c: swim version   = 0x1
2021-05-27T22:10:42 DEBUG common.c: stlink current mode: mass
2021-05-27T22:10:42 DEBUG usb.c: JTAG/SWD freq set to 8000
2021-05-27T22:10:42 DEBUG common.c: *** stlink_enter_swd_mode ***
2021-05-27T22:10:42 DEBUG common.c: *** stlink_jtag_reset 0 ***
2021-05-27T22:10:42 DEBUG common.c: *** stlink_jtag_reset 1 ***
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02030003 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_soft_reset (halt) ***
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0xa05f0003 to 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0x01000501 to 0xe000edfc
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0x00000008 to 0xe000ed30
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02010001 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x02030003 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x00030003 at 0xe000edf0
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x00000008 at 0xe000ed30
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0x0000001f to 0xe000ed30
2021-05-27T22:10:42 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-05-27T22:10:42 DEBUG common.c: Loading device parameters....
2021-05-27T22:10:42 DEBUG common.c: *** stlink_core_id ***
2021-05-27T22:10:42 DEBUG common.c: core_id = 0x2ba01477
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x412fc230 at 0xe000ed00
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x10386436 at 0xe0042000
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0000000000 at 0x1ff800cc
2021-05-27T22:10:42 INFO common.c: L1xx High-density: 48 KiB SRAM, 384 KiB flash in at least 256 byte pages.
2021-05-27T22:10:42 DEBUG common.c: *** stlink_force_debug_mode ***
2021-05-27T22:10:42 WARN usb.c: READDEBUGREG error (0x18)
2021-05-27T22:10:42 DEBUG common.c: *** stlink_status ***
2021-05-27T22:10:42 DEBUG usb.c: core status: 00030003
2021-05-27T22:10:42 DEBUG common.c:   core status: halted
file /home/pi/Downloads/OBC_GEN_CPP.bin md5 checksum: f870c18b875b74af2f49ba8a52bfc5ae, stlink checksum: 0x00a6467b
2021-05-27T22:10:42 INFO common.c: Attempting to write 103376 (0x193d0) bytes to stm32 address: 134217728 (0x8000000)
2021-05-27T22:10:42 DEBUG common.c: *** stlink_core_id ***
2021-05-27T22:10:42 DEBUG common.c: core_id = 0x2ba01477
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x0000000c at 0x40023c18
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0x00013f00 to 0x40023c18
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x00000007 at 0x40023c04
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0x89abcdef to 0x40023c0c
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0x02030405 to 0x40023c0c
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x00000006 at 0x40023c04
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0x8c9daebf to 0x40023c10
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0x13141516 to 0x40023c10
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x00000004 at 0x40023c04
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0x0000020c to 0x40023c04
2021-05-27T22:10:42 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0x08000000
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40023c18
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40023c18
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40023c18
2021-05-27T22:10:42 DEBUG common.c: *** stlink_read_debug32 0x00000009 at 0x40023c18
[... 146 lines cut ...]
2021-05-27T22:10:53 DEBUG common.c: *** stlink_read_debug32 0x0000060b at 0x40023c18
[... cut ...]
2021-05-27T22:10:53 DEBUG common.c:  (16) ***
2021-05-27T22:10:53 DEBUG common.c: data_len = 8 0x8
 80 00 00 00 03 00 00 00
2021-05-27T22:10:53 DEBUG usb.c: r_idx (16) = 0x00000003
2021-05-27T22:10:53 INFO common.c: Go to Thumb mode
2021-05-27T22:10:53 DEBUG common.c: *** stlink_write_reg
stlink_fwrite_flash() == -1
2021-05-27T22:10:53 DEBUG common.c: *** stlink_exit_debug_mode ***
2021-05-27T22:10:53 DEBUG common.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2021-05-27T22:10:53 DEBUG common.c: *** stlink_close ***
`

Nightwalker-87, 2021-06-20:
What is the current state of this ticket?

wolsty7, 2021-06-23:
l0_soft_loader_fix 1.6.1-328-g9acf539 appears to work on both L072 and L151 now

branch l0_soft_loader_fix 1.6.1-328-g9acf539: L072 ok
> st-flash --connect-under-reset --freq=8M write /home/pi/Downloads/v0.bin 0x8000000
> st-flash 1.6.1-328-g9acf539
> 2021-06-23T13:23:46 INFO common.c: L0xx Category 5: 20 KiB SRAM, 192 KiB flash in at least 128 byte pages.
> file /home/pi/Downloads/v0.bin md5 checksum: 7059ebcb3e9bde16901ce938fc6498e, stlink checksum: 0x00b99cac
> 2021-06-23T13:23:46 INFO common.c: Attempting to write 139392 (0x22080) bytes to stm32 address: 134217728 (0x8000000)
> 2021-06-23T13:23:46 INFO common.c: Flash page at addr: 0x08000000 erased
> 2021-06-23T13:23:52 INFO common.c: Flash page at addr: 0x08021f80 erased
> 2021-06-23T13:23:52 INFO common.c: Flash page at addr: 0x08022000 erased
> 2021-06-23T13:23:52 INFO common.c: Finished erasing 1089 pages of 128 (0x80) bytes
> 2021-06-23T13:23:52 INFO common.c: Starting Flash write for L0
> 2021-06-23T13:23:52 INFO flash_loader.c: Successfully loaded flash loader in sram
> 2021-06-23T13:23:52 INFO flash_loader.c: Clear DFSR
> 2021-06-23T13:23:52 ERROR flash_loader.c: write_buffer_to_sram() == -1
> 2021-06-23T13:23:52 WARN common.c: Failed to use flash loader, fallback to soft write
> 2178/2178 halfpages written
> 2021-06-23T13:24:00 INFO common.c: Starting verification of write complete
> 2021-06-23T13:24:01 INFO common.c: Flash written and verified! jolly good!
> 2021-06-23T13:24:01 INFO common.c: Go to Thumb mode

branch l0_soft_loader_fix 1.6.1-328-g9acf539: L151 ok, but says "Starting Flash write for L0"
> st-flash --connect-under-reset --freq=8M write /home/pi/Downloads/OBC_GEN_CPP.bin 0x8000000
> st-flash 1.6.1-328-g9acf539
> 2021-06-23T13:26:34 INFO common.c: L1xx High-density: 48 KiB SRAM, 384 KiB flash in at least 256 byte pages.
> file /home/pi/Downloads/OBC_GEN_CPP.bin md5 checksum: f870c18b875b74af2f49ba8a52bfc5ae, stlink checksum: 0x00a6467b
> 2021-06-23T13:26:34 INFO common.c: Attempting to write 103376 (0x193d0) bytes to stm32 address: 134217728 (0x8000000)
> 2021-06-23T13:26:34 INFO common.c: Flash page at addr: 0x08000000 erased
> 2021-06-23T13:26:34 INFO common.c: Flash page at addr: 0x08000100 erased
> 2021-06-23T13:26:37 INFO common.c: Flash page at addr: 0x08019200 erased
> 2021-06-23T13:26:37 INFO common.c: Flash page at addr: 0x08019300 erased
> 2021-06-23T13:26:37 INFO common.c: Finished erasing 404 pages of 256 (0x100) bytes
> 2021-06-23T13:26:37 INFO common.c: Starting Flash write for L0
> 2021-06-23T13:26:37 INFO flash_loader.c: Successfully loaded flash loader in sram
> 2021-06-23T13:26:37 INFO flash_loader.c: Clear DFSR
> 2021-06-23T13:26:37 ERROR flash_loader.c: write_buffer_to_sram() == -1
> 2021-06-23T13:26:37 WARN common.c: Failed to use flash loader, fallback to soft write
> 807/807 halfpages written
> 2021-06-23T13:26:40 INFO common.c: Starting verification of write complete
> 2021-06-23T13:26:40 INFO common.c: Flash written and verified! jolly good!

branch testing 1.7.0-61-gc6ecdac: Segmentation fault on L072
> st-flash --connect-under-reset --freq=8M write /home/pi/Downloads/v0.bin 0x8000000
> st-flash 1.7.0-61-gc6ecdac
> /etc/stlink/chips: No such file or directory
> Segmentation fault

branch develop 1.7.0-34-g01418b5: L072 fail
> st-flash --connect-under-reset --freq=8M write /home/pi/Downloads/v0.bin 0x8000000
> st-flash 1.7.0-34-g01418b5
> 2021-06-23T13:34:34 INFO common.c: L0xx Category 5: 20 KiB SRAM, 192 KiB flash in at least 128 byte pages.
> file /home/pi/Downloads/v0.bin md5 checksum: 7059ebcb3e9bde16901ce938fc6498e, stlink checksum: 0x00b99cac
> 2021-06-23T13:34:34 INFO common.c: Attempting to write 139392 (0x22080) bytes to stm32 address: 134217728 (0x8000000)
> 2021-06-23T13:34:34 INFO common.c: Flash page at addr: 0x08000000 erased
> 2021-06-23T13:34:34 INFO common.c: Flash page at addr: 0x08000080 erased
> 2021-06-23T13:34:40 INFO common.c: Flash page at addr: 0x08022000 erased
> 2021-06-23T13:34:40 INFO common.c: Finished erasing 1089 pages of 128 (0x80) bytes
> 2021-06-23T13:34:40 INFO common.c: Starting Flash write for L0
> 2021-06-23T13:34:40 INFO common.c: Starting Half page flash write for STM32L core id
> 2021-06-23T13:34:40 INFO flash_loader.c: Successfully loaded flash loader in sram
> 2021-06-23T13:34:40 INFO flash_loader.c: Clear DFSR
> 2021-06-23T13:34:40 ERROR flash_loader.c: Write error
> 2021-06-23T13:34:40 WARN flash_loader.c: Loader state: R2 0x40 R15 0xFFFFFFFE
> 2021-06-23T13:34:40 WARN flash_loader.c: MCU state: DHCSR 0x3000B DFSR 0x8 CFSR 0x0 HFSR 0x0
> 2021-06-23T13:34:40 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
> 2021-06-23T13:34:40 WARN common.c: 
> write_half_pages failed == -1

###it gets stuck here###
> 1088/1089 pages written
> 2021-06-23T13:36:00 ERROR common.c: Flash memory is write protected
> 2021-06-23T13:36:00 INFO common.c: Go to Thumb mode
> stlink_fwrite_flash() == -1

branch develop 1.7.0-34-g01418b5: L151 ok
> st-flash --connect-under-reset --freq=8M write /home/pi/Downloads/OBC_GEN_CPP.bin 0x8000000
> st-flash 1.7.0-34-g01418b5
> 2021-06-23T13:32:55 INFO common.c: L1xx High-density: 48 KiB SRAM, 384 KiB flash in at least 256 byte pages.
> file /home/pi/Downloads/OBC_GEN_CPP.bin md5 checksum: f870c18b875b74af2f49ba8a52bfc5ae, stlink checksum: 0x00a6467b
> 2021-06-23T13:32:55 INFO common.c: Attempting to write 103376 (0x193d0) bytes to stm32 address: 134217728 (0x8000000)
[... cut ...]
> 2021-06-23T13:32:57 INFO common.c: Finished erasing 404 pages of 256 (0x100) bytes
> 2021-06-23T13:32:57 INFO common.c: Starting Flash write for L0
> 2021-06-23T13:32:57 INFO common.c: Starting Half page flash write for STM32L core id
> 2021-06-23T13:32:57 INFO flash_loader.c: Successfully loaded flash loader in sram
> 2021-06-23T13:32:57 INFO flash_loader.c: Clear DFSR
> 807/807 halfpages written
> 2021-06-23T13:33:08 INFO common.c: Starting verification of write complete
> 2021-06-23T13:33:09 INFO common.c: Flash written and verified! jolly good!

wolsty7, 2021-06-24:
aa70b89
branch l0_soft_loader_fix 1.6.1-329-gaa70b89: L072 ok
> st-flash --connect-under-reset --freq=8M write /home/pi/Downloads/v0.bin 0x8000000
> st-flash 1.6.1-329-gaa70b89
> 2021-06-24T12:55:06 INFO common.c: L0xx Category 5: 20 KiB SRAM, 192 KiB flash in at least 128 byte pages.
> file /home/pi/Downloads/v0.bin md5 checksum: 7059ebcb3e9bde16901ce938fc6498e, stlink checksum: 0x00b99cac
> 2021-06-24T12:55:09 INFO common.c: Attempting to write 139392 (0x22080) bytes to stm32 address: 134217728 (0x8000000)
> 2021-06-24T12:55:09 INFO common.c: Flash page at addr: 0x08000000 erased
> 2021-06-24T12:55:09 INFO common.c: Flash page at addr: 0x08000080 erased
> 2021-06-24T12:55:15 INFO common.c: Flash page at addr: 0x08021f80 erased
> 2021-06-24T12:55:15 INFO common.c: Flash page at addr: 0x08022000 erased
> 2021-06-24T12:55:15 INFO common.c: Finished erasing 1089 pages of 128 (0x80) bytes
> 2021-06-24T12:55:15 INFO common.c: Starting Flash write for L0
> 2021-06-24T12:55:15 INFO flash_loader.c: Successfully loaded flash loader in sram
> 2021-06-24T12:55:15 INFO flash_loader.c: Clear DFSR
> 2021-06-24T12:55:15 ERROR flash_loader.c: write_buffer_to_sram() == -1
> 2021-06-24T12:55:15 WARN common.c: Failed to use flash loader, fallback to soft write
> 2178/2178 halfpages written
> 2021-06-24T12:55:23 INFO common.c: Starting verification of write complete
> 2021-06-24T12:55:23 INFO common.c: Flash written and verified! jolly good!

branch l0_soft_loader_fix 1.6.1-329-gaa70b89: L151 ok
> st-flash --connect-under-reset --freq=8M write /home/pi/Downloads/OBC_GEN_CPP.bin 0x8000000
> st-flash 1.6.1-329-gaa70b89
> 2021-06-24T09:59:19 INFO common.c: L1xx High-density: 48 KiB SRAM, 384 KiB flash in at least 256 byte pages.
> file /home/pi/Downloads/OBC_GEN_CPP.bin md5 checksum: f870c18b875b74af2f49ba8a52bfc5ae, stlink checksum: 0x00a6467b
> 2021-06-24T09:59:19 INFO common.c: Attempting to write 103376 (0x193d0) bytes to stm32 address: 134217728 (0x8000000)
> 2021-06-24T09:59:19 INFO common.c: Flash page at addr: 0x08000000 erased
> 2021-06-24T09:59:19 INFO common.c: Flash page at addr: 0x08000100 erased
> 2021-06-24T09:59:21 INFO common.c: Flash page at addr: 0x08019200 erased
> 2021-06-24T09:59:21 INFO common.c: Flash page at addr: 0x08019300 erased
> 2021-06-24T09:59:21 INFO common.c: Finished erasing 404 pages of 256 (0x100) bytes
> 2021-06-24T09:59:21 INFO common.c: Starting Flash write for L0
> 2021-06-24T09:59:21 INFO flash_loader.c: Successfully loaded flash loader in sram
> 2021-06-24T09:59:21 INFO flash_loader.c: Clear DFSR
> 2021-06-24T09:59:21 ERROR flash_loader.c: write_buffer_to_sram() == -1
> 2021-06-24T09:59:21 WARN common.c: Failed to use flash loader, fallback to soft write
> 807/807 halfpages written
> 2021-06-24T09:59:25 INFO common.c: Starting verification of write complete
> 2021-06-24T09:59:25 INFO common.c: Flash written and verified! jolly good!

Nightwalker-87, 2021-07-16:
@wolsty7 Is this fully resolved now?

wolsty7, 2021-07-21:
branch develop ga52e1bc isn't working, is this known @Nightwalker-87 ?:
> st-flash 1.7.0-77-ga52e1bc
> /etc/stlink/chips: No such file or directory
> Segmentation fault

> st-flash 1.7.0-63-ga2a6878
> /etc/stlink/chips: No such file or directory
> Segmentation fault

branch l0_soft_loader_fix 1.6.1-330-g679ab65 still works for both L151 and L072
> st-flash --connect-under-reset --freq=8M write /home/pi/Downloads/OBC_GEN_CPP.bin 0x8000000
> st-flash 1.6.1-330-g679ab65
> 2021-07-21T10:30:04 INFO common.c: L1xx High-density: 48 KiB SRAM, 384 KiB flash in at least 256 byte pages.
> file /home/pi/Downloads/OBC_GEN_CPP.bin md5 checksum: f870c18b875b74af2f49ba8a52bfc5ae, stlink checksum: 0x00a6467b
> 2021-07-21T10:30:04 INFO common.c: Attempting to write 103376 (0x193d0) bytes to stm32 address: 134217728 (0x8000000)
> 2021-07-21T10:30:04 INFO common.c: Flash page at addr: 0x08000000 erased
> 2021-07-21T10:30:04 INFO common.c: Flash page at addr: 0x08000100 erased
> 2021-07-21T10:30:04 INFO common.c: Flash page at addr: 0x08000200 erased
> ###removed for brevity###
> 2021-07-21T10:30:07 INFO common.c: Flash page at addr: 0x08019100 erased
> 2021-07-21T10:30:07 INFO common.c: Flash page at addr: 0x08019200 erased
> 2021-07-21T10:30:07 INFO common.c: Flash page at addr: 0x08019300 erased
> 2021-07-21T10:30:07 INFO common.c: Finished erasing 404 pages of 256 (0x100) bytes
> 2021-07-21T10:30:07 INFO common.c: Starting Flash write for L0
> 2021-07-21T10:30:07 INFO flash_loader.c: Successfully loaded flash loader in sram
> 2021-07-21T10:30:07 INFO flash_loader.c: Clear DFSR
> 2021-07-21T10:30:07 ERROR flash_loader.c: write_buffer_to_sram() == -1
> 2021-07-21T10:30:07 WARN common.c: Failed to use flash loader, fallback to soft write
> 807/807 halfpages written
> 2021-07-21T10:30:10 INFO common.c: Starting verification of write complete
> 2021-07-21T10:30:10 INFO common.c: Flash written and verified! jolly good!

Nightwalker-87, 2021-08-13:
@wolsty7 The segmentation fault may have been fixed in #1166.
However your installation does not seem to be correct: The given directory should not be present and should not be searched for.
Please retry with current _develop_ and actively follow open issues & discussions.

wolsty7, 2021-08-14:
STM32l072CZ is working on develop now

pi@raspberrypi:~/git/stlink $ st-flash --connect-under-reset --freq=8M write /home/pi/Downloads/v0.bin 0x8000000
st-flash 1.7.0-98-ga6d19d7
/etc/stlink/chips: No such file or directory
2021-08-14T15:52:19 INFO common.c: L0xx Cat.5: 20 KiB SRAM, 192 KiB flash in at least 128 byte pages.
file /home/pi/Downloads/v0.bin md5 checksum: 7059ebcb3e9bde16901ce938fc6498e, stlink checksum: 0x00b99cac
2021-08-14T15:52:19 INFO common.c: Attempting to write 139392 (0x22080) bytes to stm32 address: 134217728 (0x8000000)
2021-08-14T15:52:19 INFO common.c: Flash page at addr: 0x08000000 erased
2021-08-14T15:52:19 INFO common.c: Flash page at addr: 0x08000080 erased
...
cut for brevity
...
2021-08-14T15:52:25 INFO common.c: Flash page at addr: 0x08021f80 erased
2021-08-14T15:52:25 INFO common.c: Flash page at addr: 0x08022000 erased
2021-08-14T15:52:25 INFO common.c: Finished erasing 1089 pages of 128 (0x80) bytes
2021-08-14T15:52:25 INFO common.c: Starting Flash write for L0
2021-08-14T15:52:25 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-08-14T15:52:25 INFO flash_loader.c: Clear DFSR
2021-08-14T15:52:25 ERROR flash_loader.c: write_buffer_to_sram() == -1
2021-08-14T15:52:25 WARN common.c: Failed to use flash loader, fallback to soft write
2178/2178 halfpages written
2021-08-14T15:52:33 INFO common.c: Starting verification of write complete
2021-08-14T15:52:34 INFO common.c: Flash written and verified! jolly good!
pi@raspberrypi:~/git/stlink $

Nightwalker-87, 2021-08-14:
@wolsty7 Please leave closing to Github Automation.
Can you identify the commit that fixed the issue for you?

wolsty7, 2021-08-15:
commit 1dd94a16a8090f17d9c595daf1e899d95c0511ab fixed it
commit a52e1bc5489e23f3c1071c6912820efacaa3b22c /etc/stlink/chips: No such file or directory
commit dc5388d94c959b271d51cd98ea973cf4e8febcf0 /usr/local/bin/st-flash: No such file or directory
commit f8b9bac4e581e3fd8d7cb79fd49f346b02fb500e gets stuck at write_half_pages failed == -1
commit a2a687873b908c41c49ab1811d5dd76179b27506 broke it "/etc/stlink/chips: No such file or directory"
commit aa70b89b6e33eac460b9e7d0a73231d8a438ee10 fixed it
commit c6ecdac78ff7c8ec5a88b97db820f79ac2b72b30 segmentation fault
commit a52e1bc5489e23f3c1071c6912820efacaa3b22c broke it again

Nightwalker-87, 2021-08-15:
@wolsty7 Thanks for the detailed coverage. I'll review ASAP.

Nightwalker-87, 2021-08-15:
Fixed by #1166.

closed 2021-08-15 by Nightwalker-87