stlink-org/stlink #1143: STM32L021 mis-identified as STM32L011 and incorrect sram size reported

djix123, 2021-05-19:
st-info --probe mis-reports a STM32L021D4 as a STM32L011 (same chipid: 0x0457) together with incorrect sram size (8192 bytes, when it should be 2048 bytes)

% st-info --probe  
Found 1 stlink programmers
  version:    V2J37S7
  serial:     46004800110000315037504E
  flash:      16384 (pagesize: 128)
  sram:       8192
  chipid:     0x0457
  descr:      L011

Ant-ON, 2021-05-19:
@djix123 This is normal. Everything should work correctly. I will try to correct the target names soon.

djix123, 2021-05-19:
@Ant-ON I created a pull request to change the description. If you'd like a different description I can change etc ...

Also, I didn't update the sram to 2048 since whilst all L021/L011 parts are 2k, there are L010 parts that are 8k / 20k.

closed 2021-05-21 by Nightwalker-87