stlink-org/stlink #1146: [STM32H743/753]: Cannot write to area at 0x0810_0000, but can at 0x0800_0000

rcordeau, 2021-05-27:
- [X ] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [X] Programmer/board type: STLINK V2 
- [X] Operating system an version: Ubuntu 20.04, MacOS 10.15
- [X] **stlink tools version** and/or git commit hash: 1.7.0
- [X] stlink commandline tool name: st-flash
- [X] Target chip (and board, if applicable): STM32H743/753

Commandline output:
Good output @ 0x0800_0000
st-flash --reset write ~/fcc.bin 0x08000000
st-flash 1.7.0
2021-05-27T00:29:24 INFO common.c: H74x/H75x: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
file /Users/rcordeau/Downloads/lightning@0.9.0/lightning_fcc@0.9.0/lightning_fccboot.bin md5 checksum: c951e84957504eb54c2e2e73c5bb77c, stlink checksum: 0x0133d67c
2021-05-27T00:29:24 INFO common.c: Attempting to write 223904 (0x36aa0) bytes to stm32 address: 134217728 (0x8000000)
2021-05-27T00:29:25 INFO common.c: Flash page at addr: 0x08000000 erased
2021-05-27T00:29:25 INFO common.c: Flash page at addr: 0x08020000 erased
2021-05-27T00:29:25 INFO common.c: Finished erasing 2 pages of 131072 (0x20000) bytes
2021-05-27T00:29:25 INFO common.c: Starting Flash write for H7
223904/223904 bytes written
2021-05-27T00:29:30 INFO common.c: Starting verification of write complete
2021-05-27T00:29:32 INFO common.c: Flash written and verified! jolly good!
2021-05-27T00:29:32 WARN common.c: NRST is not connected

Bad output @ 0x0810_0000
st-flash --reset write ~/fcc.bin 0x08100000
st-flash 1.7.0
2021-05-27T00:30:58 INFO common.c: H74x/H75x: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
file /Users/rcordeau/Downloads/lightning@0.9.0/lightning_fcc@0.9.0/lightning_fccboot.bin md5 checksum: c951e84957504eb54c2e2e73c5bb77c, stlink checksum: 0x0133d67c
2021-05-27T00:30:58 INFO common.c: Attempting to write 223904 (0x36aa0) bytes to stm32 address: 135266304 (0x8100000)
2021-05-27T00:30:58 INFO common.c: Flash page at addr: 0x08100000 erased
2021-05-27T00:30:58 INFO common.c: Flash page at addr: 0x08120000 erased
2021-05-27T00:30:58 INFO common.c: Finished erasing 2 pages of 131072 (0x20000) bytes
2021-05-27T00:30:58 INFO common.c: Starting Flash write for H7
223904/223904 bytes written
2021-05-27T00:31:03 INFO common.c: Starting verification of write complete
2021-05-27T00:31:03 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
```
OUTPUT/ERROR of the commandline tool(s)
```
The official STM32Programmer works fine. If I write the same file to 0x080e0000, it half-verifies. The portion starting at 0x08100000 reads back all F's. 

When writing to 0x08000000, the debug output when writing shows (making it a short example):
214400/223904 bytes written2021-05-27T00:18:39 DEBUG common.c: *** stlink_write_mem32 64 bytes to 0x8034580
2021-05-27T00:18:39 DEBUG common.c: *** stlink_read_debug32 0x00010000 at 0x52002010

When writing to 0x08100000, the debug output when writing shows (making it a short example):
214400/223904 bytes written2021-05-27T00:17:06 DEBUG common.c: *** stlink_write_mem32 64 bytes to 0x8134580
2021-05-27T00:17:06 DEBUG common.c: *** stlink_read_debug32 0000000000 at 0x52002010

Ant-ON, 2021-05-27:
@rcordeau When writing to the second bank, the FLASH_SR2 register (`0x52002110`) instead the FLASH_SR register (`0x52002010`) is use. Can you attach complete logs?

rcordeau, 2021-05-27:
[flashlog_08100000.txt](https://github.com/stlink-org/stlink/files/6555448/flashlog_08100000.txt)
Sure thing.

rcordeau, 2021-05-27:
This is another error log when writing to 0x080e0000, which would cross into the 0x08100000 page.
[flashwrite_080e0000.txt](https://github.com/stlink-org/stlink/files/6555475/flashwrite_080e0000.txt)

Ant-ON, 2021-05-28:
@rcordeau Thanks for the logs! I seem to have found the problem. Could you check out https://github.com/Ant-ON/stlink/tree/h7_db_fix branch?

rcordeau, 2021-05-28:
Hi @Ant-ON, yep, this fixed it! Thanks!

PR #1147 (2021-06-03): Fixed clearance of the H7 dual bank flag
Fixed a typo in the procedure of setting parameters of target.

(Closes #1146)
closed 2021-06-03 by Nightwalker-87