stlink-org/stlink #1148: Version 1.7.0 does not find STMicroelectronics ST-LINK/V2 on Debian 10 (Buster)

karstengit, 2021-06-01:
I have now the problem of he error:
`WARN common.c: unknown chip id! 0x989680`
with my last version v1.6.0-31-gf5d0454-dirty

Regarding https://github.com/stlink-org/stlink/issues/568 i installed the Debian package https://github.com/stlink-org/stlink/releases/download/v1.7.0/stlink_1.7.0-1_amd64.deb now.

But this newer version does not find the ST-LINK/V2:

`$ lsusb

...

Bus 008 Device 002: ID 0483:3748 STMicroelectronics ST-LINK/V2

...

$ st-info --probe

Found 0 stlink programmers

$ st-info --version

v1.7.0
`
What can i do now?

Ant-ON, 2021-06-01:
@karstengit Perhaps after installing the package, you need to manually update the udev rules (or just reboot the system)

karstengit, 2021-06-01:
This seems to be a problem of the library libstlink.so.1, because the old version does it find any more too:

```
$ /old/st-flash write file.bin 0x8000000
st-flash 1.6.0-31-gf5d0454-dirty
2021-06-01T11:39:46 WARN usb.c: Couldn't find any ST-Link devices

```

After reinstalling the old library the ST-LINK/V2 is found again.

karstengit, 2021-06-01:
> @karstengit Perhaps after installing the package, you need to manually update the udev rules (or just reboot the system)

No - sorry - after reinstallation of the package and a reboot it is still the same problem!

```
$ st-flash write file.bin 0x8000000
st-flash 1.7.0
2021-06-01T11:54:52 WARN usb.c: Couldn't find any ST-Link devices
$ st-info --probe
Found 0 stlink programmers

```

Ant-ON, 2021-06-01:
Can you try run `sudo st-info --probe`?

karstengit, 2021-06-01:
(Of course) it's already tested as root with the same result.

Nightwalker-87, 2021-06-01:
Verified to work in general for this combination already.
Further I am closing this ticket now, as it does not follow our project guidelines for opening new issues.

closed 2021-06-01 by Nightwalker-87