stlink-org/stlink #1151: Detecting when the ST-Link or Target was disconnected

gustavolaureano, 2021-06-06:
Hi,

Is there a way to detect, after opening the ST-Link, when the target or the ST-Link was disconnected?

I am coding an application that requires periodic polling of the target's RAM, but also requires to detect when the target or the ST-Link was disconnected (or not yet connected) and reconnect (in hotplug) as soon as it finds the target / stlink again.

During some tests on Linux I could not get any error from stlink_read_mem32() or even stlink_status() when removing the target on the fly.. my solution was to always close and open the stlink when starting the next reading, which works without problems on Linux, but keeps giving me random "denied access" errors on Windows, something related to the libusb.

Thanks!

Gustavo

Ant-ON, 2021-06-06:
@gustavolaureano, Hi!

I improved the command status check.
You can try:
https://github.com/Ant-ON/stlink/commits/connect_under_reset_rework

Nightwalker-87, 2021-06-16:
@gustavolaureano Have you had the chance to try in the meanwhile?

gustavolaureano, 2021-06-20:
> @gustavolaureano Have you had the chance to try in the meanwhile?

Hi, unfortunately not yet, I will return to this project in the coming weeks, then I report back here :)

Nightwalker-87, 2021-08-14:
ping @gustavolaureano

Nightwalker-87, 2021-08-22:
@Ant-ON How shall we proceed here? We need somebody to test.

Nightwalker-87, 2021-09-12:
@slyshykO Do you have any idea on this one?

Nightwalker-87, 2022-01-03:
@gustavolaureano Any update on this?

Nightwalker-87, 2022-08-27:
@Ant-ON: Can you contribute your improvements in order to close this issue?

Nightwalker-87, 2022-10-23:
This issue is now closed due to inactivity.
Please also note that any version prior to `v1.7.0` is unsupported according to our [#Security Policy](https://github.com/stlink-org/stlink/security/policy#security-policy).
Should the problem persist, please retry with the latest version in the `develop` branch.
If this is the case, one should then open a new ticket.

closed 2022-10-23 by Nightwalker-87