stlink-org/stlink #1161: [STM32L152RE]: regression, unable to flash since specific revision

dariox86, 2021-07-14:
- [ ] Programmer/board type: [genuine STLINK-V3SET updated to latest version V3J7M3B5S1]
- [ ] Operating system and version: [Parabola GNU/Linux-libre, rolling release operating system updated to the latest version]
- [ ] **stlink tools version** and/or git commit hash: [v1.6.1/git-ee491d9]
- [ ] stlink commandline tool name: [st-flash]
- [ ] Target chip (and board, if applicable): [STM32L152RE]

Since pull request #1124 merged in revision v1.6.1/git-ee491d9, when issuing «st-flash write FIRMWARE.bin 0x8000000» I get an infinite bunch of «WARN common.c: Flash base use default L0 address» cluttering the screen and firmware is not flashed. It works in the previous revision.

Ant-ON, 2021-07-14:
@dariox86 Can you test https://github.com/Ant-ON/stlink/tree/l0_soft_loader_fix branch?
Can you provide full log of flashing operation?

dariox86, 2021-07-15:
Tested with https://github.com/Ant-ON/stlink/tree/l0_soft_loader_fix branch.

Broken since Apr 13, revision ee491d9 with these error messages:

```
$ ./st-flash write FIRMWARE.bin 0x08000000
st-flash 1.7.0
2021-07-15T10:42:26 INFO common.c: L152RE: 80 KiB SRAM, 512 KiB flash in at least 256 byte pages.
file FIRMWARE.bin md5 checksum: bcc198f448b77785a1a49385ab86519, stlink checksum: 0x005c9689
2021-07-15T10:42:26 INFO common.c: Attempting to write 58712 (0xe558) bytes to stm32 address: 134217728 (0x8000000)
2021-07-15T10:42:26 WARN common.c: Flash base use default L0 address2021-07-15T10:42:26 WARN common.c: Flash base use default L0 address2021-07-15T10:42:26 WARN common.c: Flash base use default L0 address2021-07-15T10:42:26

[terminal window filled with an infinite bunch of these until I CTRL+C]
```

Still not working with additional error messages since May 22, revision cab9fc4:

```
$ ./st-flash write FIRMWARE.bin 0x08000000
st-flash 1.7.0
2021-07-15T10:47:03 INFO common.c: L152RE: 80 KiB SRAM, 512 KiB flash in at least 256 byte pages.
2021-07-15T10:47:03 WARN usb.c: READDEBUGREG error (0x18)
file FIRMWARE.bin md5 checksum: bcc198f448b77785a1a49385ab86519, stlink checksum: 0x005c9689
2021-07-15T10:47:03 INFO common.c: Attempting to write 58712 (0xe558) bytes to stm32 address: 134217728 (0x8000000)
2021-07-15T10:47:03 WARN common.c: Flash base use default L0 address2021-07-15T10:47:03 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:47:03 WARN common.c: Flash base use default L0 address2021-07-15T10:47:03 WARN usb.c: WRITEDEBUGREG error (0x19)
2021-07-15T10:47:03 WARN common.c: Flash base use default L0 address2021-07-15T10:47:03 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:47:03 WARN usb.c: WRITEDEBUGREG error (0x19)
2021-07-15T10:47:03 WARN common.c: Flash base use default L0 address2021-07-15T10:47:03 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:47:03 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:47:03 WARN usb.c: WRITEDEBUGREG error (0x19)
2021-07-15T10:47:03 WARN common.c: Flash base use default L0 address2021-07-15T10:47:03 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:47:03 INFO common.c: Flash page at addr: 0x08000000 erased
2021-07-15T10:47:03 WARN common.c: Flash base use default L0 address2021-07-15T10:47:03 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:47:03 WARN common.c: Flash base use default L0 address2021-07-15T10:47:03 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:47:03 WARN common.c: Flash base use default L0 address2021-07-15T10:47:03 WARN usb.c: READDEBUGREG error (0x18)

[infinite messages like the last ones until I CTRL+C]
```

It is working again with a bunch of error messages in May 22, revision cab9fc4:

```
$ ./st-flash write FIRMWARE.bin 0x08000000
st-flash 1.7.0
2021-07-15T10:51:54 INFO common.c: L152RE: 80 KiB SRAM, 512 KiB flash in at least 256 byte pages.
2021-07-15T10:51:54 WARN usb.c: READDEBUGREG error (0x18)
file FIRMWARE.bin md5 checksum: bcc198f448b77785a1a49385ab86519, stlink checksum: 0x005c9689
2021-07-15T10:51:54 INFO common.c: Attempting to write 58712 (0xe558) bytes to stm32 address: 134217728 (0x8000000)
2021-07-15T10:51:54 WARN common.c: Flash base use default L0 address2021-07-15T10:51:54 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:51:54 WARN common.c: Flash base use default L0 address2021-07-15T10:51:54 WARN usb.c: WRITEDEBUGREG error (0x19)
2021-07-15T10:51:54 WARN common.c: Flash base use default L0 address2021-07-15T10:51:54 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:51:54 WARN usb.c: WRITEDEBUGREG error (0x19)
2021-07-15T10:51:54 WARN common.c: Flash base use default L0 address2021-07-15T10:51:54 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:51:54 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:51:54 WARN usb.c: WRITEDEBUGREG error (0x19)
2021-07-15T10:51:54 WARN common.c: Flash base use default L0 address2021-07-15T10:51:54 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:51:54 INFO common.c: Flash page at addr: 0x08000000 erased
2021-07-15T10:51:54 WARN common.c: Flash base use default L0 address2021-07-15T10:51:54 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:51:54 WARN common.c: Flash base use default L0 address2021-07-15T10:51:54 WARN usb.c: WRITEDEBUGREG error (0x19)
2021-07-15T10:51:54 WARN common.c: Flash base use default L0 address2021-07-15T10:51:54 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:51:54 WARN usb.c: WRITEDEBUGREG error (0x19)
2021-07-15T10:51:54 WARN common.c: Flash base use default L0 address2021-07-15T10:51:54 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:51:54 WARN usb.c: READDEBUGREG error (0x18)
2021-07-15T10:51:54 WARN usb.c: WRITEDEBUGREG error (0x19)

[a lot of these repeated all over]

2021-07-15T10:51:55 INFO common.c: Flash page at addr: 0x0800e500 erased
2021-07-15T10:51:55 INFO common.c: Finished erasing 230 pages of 256 (0x100) bytes
[... 64 lines cut ...]
```

Not working with different error messages since June 23, revision aa70b89:

```
$ ./st-flash write FIRMWARE.bin 0x08000000
st-flash 1.7.0
2021-07-15T11:09:14 INFO common.c: L152RE: 80 KiB SRAM, 512 KiB flash in at least 256 byte pages.
file FIRMWARE.bin md5 checksum: bcc198f448b77785a1a49385ab86519, stlink checksum: 0x005c9689
2021-07-15T11:09:14 INFO common.c: Attempting to write 58712 (0xe558) bytes to stm32 address: 134217728 (0x8000000)
[... cut ...]
2021-07-15T11:09:15 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-07-15T11:09:15 INFO flash_loader.c: Clear DFSR
2021-07-15T11:09:15 WARN common.c: Flash base use default L0 address2021-07-15T11:09:15 WARN common.c: Flash base use default L0 address2021-07-15T11:09:15 ERROR flash_loader.c: write_buffer_to_sram() == -1
2021-07-15T11:09:15 WARN common.c: Failed to use flash loader, fallback to soft write
2021-07-15T11:09:15 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1
```

Ant-ON, 2021-07-15:
@dariox86
Thanks for the very detailed review! I have not been able to understand the reasons for this behavior yet. But I fixed one bug that was there. Can you test the https://github.com/Ant-ON/stlink/tree/l152re_fix branch?

dariox86, 2021-07-15:
@Ant-ON, I am glad my report was helpful. It works! Thank you! In aa70b89 I experience the usual issue with «WARN common.c: Flash base use default L0 address». After your commit in 679ab65 it works. It is also much faster! Before that it took maybe a couple of minutes. Now it's a matter of seconds.

For the record:

$ ./st-flash write FIRMWARE.bin 0x08000000
st-flash 1.7.0
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/034, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
2021-07-15T14:13:57 INFO common.c: L152RE: 80 KiB SRAM, 512 KiB flash in at least 256 byte pages.
file FIRMWARE.bin md5 checksum: bcc198f448b77785a1a49385ab86519, stlink checksum: 0x005c9689
2021-07-15T14:13:57 INFO common.c: Attempting to write 58712 (0xe558) bytes to stm32 address: 134217728 (0x8000000)
2021-07-15T14:13:57 INFO common.c: Flash page at addr: 0x08000000 erased
2021-07-15T14:13:57 INFO common.c: Flash page at addr: 0x08000100 erased
2021-07-15T14:13:57 INFO common.c: Flash page at addr: 0x08000200 erased

[...]

2021-07-15T14:13:59 INFO common.c: Flash page at addr: 0x0800e300 erased
2021-07-15T14:13:59 INFO common.c: Flash page at addr: 0x0800e400 erased
2021-07-15T14:13:59 INFO common.c: Flash page at addr: 0x0800e500 erased
2021-07-15T14:13:59 INFO common.c: Finished erasing 230 pages of 256 (0x100) bytes
2021-07-15T14:13:59 INFO common.c: Starting Flash write for L0
2021-07-15T14:13:59 INFO flash_loader.c: Successfully loaded flash loader in sram
2021-07-15T14:13:59 INFO flash_loader.c: Clear DFSR
2021-07-15T14:13:59 INFO flash_loader.c: Clear CFSR
2021-07-15T14:13:59 INFO common.c: Go to Thumb mode
2021-07-15T14:14:00 ERROR flash_loader.c: Flash loader run error
2021-07-15T14:14:00 WARN flash_loader.c: Loader state: R2 0x0 R15 0x0
2021-07-15T14:14:00 WARN flash_loader.c: MCU state: DHCSR 0x90009 DFSR 0x0 CFSR 0x100 HFSR 0x0
2021-07-15T14:14:00 WARN common.c: Failed to use flash loader, fallback to soft write
458/458 halfpages written
2021-07-15T14:14:03 INFO common.c: Starting verification of write complete
2021-07-15T14:14:03 INFO common.c: Flash written and verified! jolly good!
2021-07-15T14:14:03 INFO common.c: Go to Thumb mode

PR #1162 (2021-07-16): Fixed get_stm32l0_flash_base address for STM32L152RE
Determination of the flash base address for the L0/L1 series is centralized in one function. Fixed get flash base address for STM32L152RE.

(Closes #1161)
closed 2021-07-16 by Nightwalker-87