stlink-org/stlink #1163: [stm32f103c8t6]: st-flash Segmentation Fault - Fedora 34

machadothi, 2021-07-18:
Hello everyone!
I'm getting a Segmentation Fault when I perform a st-flash command, running on Fedora 34.
I already tryed to install it by downloading the repository:

  * $ sudo dnf install git make cmake libusbx-devel (libusb-1.0-0-dev Ubuntu)
  * $ git clone https://github.com/stlink-org/stlink
  * $ cd stlink
  * $ make
  * $ cd build/Release
  * $ sudo make install

And also using 'dnf' command:
* $ sudo dnf install stlink

Error output:
$ /usr/local/bin/st-flash  write main.bin 0x8000000
st-flash 1.7.0-77-ga52e1bc
make: *** [../../Makefile.incl:134: flash] Segmentation fault (core dumped)

Ant-ON, 2021-07-21:
@machadothi Can you try run the `/usr/local/bin/st-flash --debug write main.bin 0x8000000` and post log?

machadothi, 2021-07-21:
Sure! Here it is:

**$ /usr/local/bin/st-flash --debug write main.bin 0x8000000**
st-flash 1.7.0-77-ga52e1bc
2021-07-21T20:54:10 DEBUG common.c: *** looking up stlink version
2021-07-21T20:54:10 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-07-21T20:54:10 DEBUG common.c: stlink pid     = 0x3748
2021-07-21T20:54:10 DEBUG common.c: stlink version = 0x2
2021-07-21T20:54:10 DEBUG common.c: jtag version   = 0x1d
2021-07-21T20:54:10 DEBUG common.c: swim version   = 0x7
2021-07-21T20:54:10 DEBUG common.c: stlink current mode: dfu
2021-07-21T20:54:10 DEBUG usb.c: -- exit_dfu_mode
2021-07-21T20:54:10 DEBUG usb.c: JTAG/SWD freq set to 0
2021-07-21T20:54:10 DEBUG common.c: stlink current mode: mass
2021-07-21T20:54:10 DEBUG common.c: *** stlink_enter_swd_mode ***
2021-07-21T20:54:10 DEBUG common.c: Loading device parameters....
2021-07-21T20:54:10 DEBUG common.c: *** stlink_core_id ***
2021-07-21T20:54:10 DEBUG common.c: core_id = 0x1ba01477
2021-07-21T20:54:10 DEBUG common.c: *** stlink_read_debug32 0x411fc231 at 0xe000ed00
2021-07-21T20:54:10 DEBUG common.c: *** stlink_read_debug32 0x20036410 at 0xe0042000
[1]    84904 segmentation fault (core dumped)  /usr/local/bin/st-flash --debug write main.bin 0x8000000

rusefillc, 2021-07-25:
Let me join the party with https://github.com/stlink-org/stlink/issues/1165

Ant-ON, 2021-07-28:
@machadothi This problem occurred after @rewolff move chip information to files. The problem may be solved by using stlink_chipid_get_params_old instead of stlink_chipid_get_params.

rusefillc, 2021-07-28:
@rewolff @machadothi "segmentation fault" is probably not the best way to handle an error condition?

rewolff, 2021-07-28:
Of course! Segmentation fault is the system's way of saying "You screwed up badly and I can't proceed from this situation". 

So instead of just terminating the program, it at least tells you SOMETHING. 

Consider a big hall with (open) PO boxes. So you're doing a quest of some sorts. So your first assignment is go to box 123 and read the instructions. So box 123 has a note that says put this note in box 124 and proceed to box 12345. But when you walk to the end... you find that the boxes only go up to 9999. 

So that's what happened: the computer got told to put (or get) something in a box that doesn't exist, or to continue getting instructions from a box that doesn't exist.

Nightwalker-87, 2021-07-28:
Please test again with the proposed fix in the linked PR.

closed 2021-08-01 by Nightwalker-87