stlink-org/stlink #1164: [feature] Support for G491 and H7A3/H7B3

jjamesson, 2021-07-20:
Dear STLINK maintainers,

We are considering using STM32G491, and STM32H7A3/STM32H7B3 in our designs. 

As of now, it seems that these boards are not supported by ST tools yet. Could we expect support from your toolset for these chips in the near future? It would be very helpful.

With best regards!

Ant-ON, 2021-07-21:
@jjamesson I added support for these chips earlier (https://github.com/stlink-org/stlink/commit/9eb81e1e0adc6a5bf7a1c68cbbaf3fd35ec85be0 and https://github.com/stlink-org/stlink/commit/54000d67acefd86df4bdb0594dee940dd04763fa already in the `develop` branch). I had nothing to test these chips on, but everything should work. Support for the H7 and G4 series has been added previously and has been tested.

jjamesson, 2021-07-21:
Thanks for the reply, @Ant-ON ,

You're right, I checked the `/doc/devices_boards.md` file and saw that these chips weren't supported according to the list, but I didn't try the development branch myself.

Thanks,

Nightwalker-87, 2021-07-28:
@jjamesson The list will be updated as soon as these changes merge into the `master` branch with the next release (at the latest).

commit 26a63c11 (2021-07-28): [doc] Updated list of devices & boards

(Closes #1164)
closed 2021-07-28 by Nightwalker-87